mattanger/ckb-next #93: Mouse jumping to the top left on click and hold

snowcamo, 2017-02-28:
I've recently uninstalled ckb, because too often my mouse would jump to the very top left corner of my screen when I would click and hold. I'm not exactly sure why this problem occurs. Sometimes changing my USB ports helps, sometimes it does not go away. I'm not sure if this is how related or not this is to the actual issue, but a trend I noticed was, if I left my mouse unplugged for a while and then replugged it would function correctly at a higher percentage rate.  I am using Sierra 10.12.2 on a hackintosh. I'm pretty sure the issue is not hackintosh related, because none of my other USB devices have issues. I would be completely willing to reinstall ckb-next and try and do some trouble shooting if I could do anything to help. K65 mouse, K95 keyboard. Thanks!

frickler24, 2017-02-28:
@snowcamo I think you mean a M65 mouse, not K65?

snowcamo, 2017-02-28:
@frickler24 Correct! My bad.

tatokis, 2017-02-28:
Thank you for taking the time to report this issue.
Unfortunately I don't think anyone in our dev team has a Corsair mouse on macOS to test.

Apple made some API changes that broke ckb-next in Sierra, and we are slowly trying to find workarounds.

fleischie, 2017-03-01:
@tatokis I do have a mac and a mouse (also M65) at my exposal.

I can try to re-create the behavior, althought I don't know what to look for, when I encounter this specific behavior: Run `ckb` and it's daemon with debugging enabled? What would I need to look for?

JarrenJ, 2017-03-21:
I use a Scimitar RGB on IMAC running osSierra. I used to get this problem although only when I restart or shut down my computer. A possible solution you could try to do. Is when restarting or shutting down quit ckb-next. Before doing so. Then I plug devices as it is shutting down. And when you turn it on the problem should be fixed. Only have been able to recreate it twice; but, it might help.

light2yellow, 2017-03-25:
@fleischie @JMONEYTX from #120 has M65 and experiences roughly the same. Thus I suppose it's not a device-specific issue. Should be some kind of overflow in the code or something.

tatokis, 2017-03-25:
Any chance it could be related to dropped events?
Would be worth giving https://github.com/mattanger/ckb-next/pull/106 a shot and seeing if it helps.

JMONEYTX, 2017-03-25:
Anyway i can fix it. It makes my computer unusable

tatokis, 2017-03-25:
@JMONEYTX Uninstall (https://github.com/mattanger/ckb-next#uninstalling-1) and reboot?

JMONEYTX, 2017-03-25:
I fixed it. lol just saw urs. So what i did was i saved my config that i liked to my mouse. i quit CQB
then i restarted my computer. Lol

ollieatkinson, 2017-04-04:
I can reproduce the issue with Sierra 10.12.4 and a "Corsair Gaming CH-9303011-EU Sabre". Happy to try provide information where possible.

fleischie, 2017-05-07:
I just reproduced the described behavior. Running ckb on Mac OS (El Capitan as well) going to sleep (in my case) and returning makes a click+hold move the mouse to a specific position on the screen (it's a little off the center of my screen).

Holding click and moving the mouse always moves the cursor back to it's fixed position. This is limited to the Corsair Mouse, as the trackpad does not seem to be affected.

I'll investigate this a little more.

OscarBarrett, 2017-07-17:
This still happens, and is related to https://github.com/mattanger/ckb-next/pull/190 (which follows on from https://github.com/mattanger/ckb-next/pull/106).

Switching users is one way to reproduce, and restarting the daemon should fix it.

AlfaceVerde, 2017-08-05:
I having the same issue with my Corsair Sabre RGB, someone know a way to solve this issue

mehh, 2017-11-29:
Looks like this issue is still happening.  Bummer that the only fix is to uninstall your program.

ollieatkinson, 2017-11-29:
@mehh if you uninstall and reinstall with the daemon does it work? 

At least work long enough to write settings to your mouse memory

OscarBarrett, 2017-12-31:
Both #106 and #190 are in master now but not included in a new release yet. As far as I can see this has been fixed for most cases.

However, switching users still causes the issue. e.g. two users logged in, one running ckb.app. Switching from either one to the other requires the daemon to be restarted for mouse events to work correctly.

ZirconiumX, 2017-12-31:
CKB is not really designed for multi-user use. Sharing access between users is pretty awkward. If there are two CKB GUIs, who should get control of the lighting and events? What if they have conflicting macros?

The daemon is very dumb, and that has advantages and disadvantages.

OscarBarrett, 2018-01-01:
IMO if two users have a LaunchAgent for the GUI, when the active user changes the GUI should be opened under the new user and close the previous instance.

Running master I'm actually getting crashes from the GUI when switching to a secondary user unless I restart the daemon - it fails to launch a second instance. On the 0.2.8 release it doesn't crash but both try to control the lighting and events which results in event issues and lights flickering between profiles.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/7

closed 2018-01-22 by tatokis