mattanger/ckb-next #94: M65 Pro RGB not being detected

CharmsyPone, 2017-02-28:
I've had a long running history of CKB not working with my mouse in Arch Linux. I'm using a fresh install of Antergos, and it has not wanted to to detect the M65 RGB or the M65 Pro RGB. I no longer have the M65 RGB, as it's been replaced with the Pro RGB.

I was hoping ckb-next would solve the issue, but it still has this same issue.

Here's a listing from lsusb (with Corsair being highlighted in bold):

Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 04f2:b374 Chicony Electronics Co., Ltd 
Bus 001 Device 003: ID 0489:e04e Foxconn / Hon Hai 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 003 Device 007: ID 1b1c:1b2e Corsair** 
Bus 003 Device 006: ID 0c45:7605 Microdia 
Bus 003 Device 005: ID 13fe:2012 Kingston Technology Company Inc. 
Bus 003 Device 004: ID 1058:259f Western Digital Technologies, Inc. My Passport Ultra (WD10JMVW)
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Of note: Laptop I'm using is Acer Aspire v3-571G, if this is a hardware related issue.

tatokis, 2017-02-28:
We added support for the M65 Pro RGB in the testing branch. If I recall 
correctly, it doesn't let you set the DPI as high as it can go, but 
it's a minor issue.
https://github.com/mattanger/ckb-next/archive/testing.zip

CharmsyPone, 2017-02-28:
That actually works! I got the rainbow effect and everything. 

However, on the note of DPI, sometimes when changing the DPI settings, I can only scroll through the first two DPI settings unless I change another DPI setting. Initially I thought it was doing it because the second one was lower than the first, but that doesn't seem to be the case. This is an unrelated issue though, so I'll close this.

closed 2017-02-28 by CharmsyPone