mattanger/ckb-next #97: K70 LUX firmware update fails

bulletmark, 2017-03-02:
I installed ckb-next-latest-git v0.2.4.r189.g8627b7e-1 on my Arch PC to update the firmware in my K70 LUX KB but it fails with error:
````
ckb-daemon[23557]: [E] fwupdate (firmware.c:78): Firmware file /tmp/ckb-13648-fwblob.bin doesn't match device (V: 1b1c P: 1b13)
````

Any suggestion?

light2yellow, 2017-03-02:
@bulletmark, we are really sorry about it. I don't have enough knowledge to evaluate on this, but here's the big picture: we have a file containing firmwares downloaded from Corsair website and the file entry that is used for your device mismatches the real device, as I see this.

What is your current firmware version?

Is it K70 LUX RGB or non-RGB?

@tatokis, if I understand this correctly that's what we were discussing on IRC - the guesses.

bulletmark, 2017-03-02:
FW version is currently 2.04, wants to upgrade to 2.05. KB is non-RGB. Here is `lsusb -v` output:
~~~~
Bus 002 Device 005: ID 1b1c:1b36 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b36 
  bcdDevice            2.04
  iManufacturer           1 Corsair
  iProduct                2 Corsair Gaming K70 LUX Keyboard 
  iSerial                 3 11037023AEBA044455DB378DF5001946
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     112
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      25
          Report Descriptor: (length is 25)
            Item(Global): Usage Page, data= [ 0xc2 0xff ] 65474
                            (null)
            Item(Local ): Usage, data= [ 0x04 ] 4
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Local ): Usage, data= [ 0x02 ] 2
                            (null)
            Item(Main  ): Feature, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x04 ] 4
                            (null)
            Item(Main  ): Output, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
~~~~

bulletmark, 2017-03-02:
BTW: The ckb utility merely reports "Update Failed" and I have to go digging through my system journal to uncover that error message. That's a little unfriendly for users. Can't you capture error messages and show them in the UI?

tatokis, 2017-03-02:
There's definitely no firmware version 2.05 for the LUX non RGB, so ckb-next should not be prompting you for a firmware update. Can you please run ```cat /dev/input/ckb1/model``` and post the output? (Assuming the keyboard is the only corsair device on your system.)

Regarding the actual error message, all the gui does is download the file and send a command to the daemon to flash it.
We could probably tell the daemon to return the reason to the GUI, something to keep in mind for the future.

bulletmark, 2017-03-02:
~~~~
pc:/dev/input ls -l
total 0
drwxr-xr-x 2 root root     140 Mar  3 07:45 by-id
drwxr-xr-x 2 root root     180 Mar  3 07:45 by-path
crw-rw---- 1 root input 13, 64 Mar  3 07:25 event0
crw-rw---- 1 root input 13, 65 Mar  3 07:45 event1
crw-rw---- 1 root input 13, 74 Mar  3 07:25 event10
crw-rw---- 1 root input 13, 75 Mar  3 07:25 event11
crw-rw---- 1 root input 13, 76 Mar  3 07:25 event12
crw-rw---- 1 root input 13, 77 Mar  3 07:25 event13
crw-rw---- 1 root input 13, 78 Mar  3 07:25 event14
crw-rw---- 1 root input 13, 79 Mar  3 07:25 event15
crw-rw---- 1 root input 13, 80 Mar  3 07:25 event16
crw-rw---- 1 root input 13, 81 Mar  3 07:25 event17
crw-rw---- 1 root input 13, 82 Mar  3 07:25 event18
crw-rw---- 1 root input 13, 66 Mar  3 07:45 event2
crw-rw---- 1 root input 13, 67 Mar  3 07:25 event3
crw-rw---- 1 root input 13, 68 Mar  3 07:25 event4
crw-rw---- 1 root input 13, 69 Mar  3 07:25 event5
crw-rw---- 1 root input 13, 70 Mar  3 07:25 event6
crw-rw---- 1 root input 13, 71 Mar  3 07:25 event7
crw-rw---- 1 root input 13, 72 Mar  3 07:25 event8
crw-rw---- 1 root input 13, 73 Mar  3 07:25 event9
crw-rw---- 1 root input 13, 63 Mar  3 07:25 mice
crw-rw---- 1 root input 13, 32 Mar  3 07:25 mouse0

pc:/dev/input ls -l by-id
total 0
lrwxrwxrwx 1 root root 10 Mar  3 07:25 usb-046d_0825_614D0540-event-if00 -> ../event18
lrwxrwxrwx 1 root root  9 Mar  3 07:45 usb-Corsair_Corsair_Gaming_K70_LUX_Keyboard_11037023AEBA044455DB378DF5001946-event-kbd -> ../event1
lrwxrwxrwx 1 root root  9 Mar  3 07:45 usb-Corsair_Corsair_Gaming_K70_LUX_Keyboard_11037023AEBA044455DB378DF5001946-if01-event-kbd -> ../event2
lrwxrwxrwx 1 root root  9 Mar  3 07:25 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event0
lrwxrwxrwx 1 root root  9 Mar  3 07:25 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse0
pc:/dev/input 
~~~~
Catting that first Corsair file returns nothing, the second file dumps a few binary chars to my screen.

frickler24, 2017-03-02:
@bulletmark Your client tries to install a firmware for ProductID 0x1b13 (P_K70) into your device with productID 0x1b36 (P_K70_LUX_NRGB). I'll have a look at the code what may be the reason for this.

tatokis, 2017-03-02:
From our discussion in the IRC room, the daemon correctly detects the 
keyboard as "Corsair Gaming K70 LUX Keyboard" so that part works.
I also saw 
https://github.com/mattanger/ckb-next/blob/master/src/ckb/fwupgradedialog.cpp#L14 
, however it does seem to pass that check, so the issue is before that.

frickler24, 2017-03-02:
@bulletmark in your last post with the content of /dev/input I cannot see any ckb#-directory. 
Please run the daemon as root in a terminal and post the output.
/dev/input should look like
```
lutz@Mainfrix ~ $ ll /dev/input
insgesamt 0
0 drwxr-xr-x  2 root root     260 Mär  2 23:16 by-id
0 drwxr-xr-x  2 root root     280 Mär  2 23:16 by-path
0 drwxr-xr-x  2 root root     100 Mär  2 23:16 ckb0
0 drwxr-xr-x  2 root root     180 Mär  2 23:16 ckb1
0 crw-rw----  1 root input 13, 64 Mär  2 06:07 event0
0 crw-rw----  1 root input 13, 65 Mär  2 06:07 event1
0 crw-rw----  1 root input 13, 74 Mär  2 06:07 event10
0 crw-rw----  1 root input 13, 75 Mär  2 06:07 event11
0 crw-rw----  1 root input 13, 76 Mär  2 06:07 event12
0 crw-rw----  1 root input 13, 77 Mär  2 06:07 event13
0 crw-rw----  1 root input 13, 78 Mär  2 06:07 event14
0 crw-rw----  1 root input 13, 79 Mär  2 06:07 event15
0 crw-rw----  1 root input 13, 80 Mär  2 23:16 event16
0 crw-rw----  1 root input 13, 81 Mär  2 23:16 event17
0 crw-rw----  1 root input 13, 82 Mär  2 06:07 event18
0 crw-rw----  1 root input 13, 83 Mär  2 06:07 event19
0 crw-rw----  1 root input 13, 66 Mär  2 06:07 event2
0 crw-rw----  1 root input 13, 84 Mär  2 23:15 event20
0 crw-rw----  1 root input 13, 85 Mär  2 23:15 event21
0 crw-rw----  1 root input 13, 86 Mär  2 06:07 event22
0 crw-rw----  1 root input 13, 87 Mär  2 06:07 event23
0 crw-rw----+ 1 root input 13, 88 Mär  2 06:07 event24
0 crw-rw----  1 root input 13, 89 Mär  2 23:15 event25
0 crw-rw----  1 root input 13, 67 Mär  2 06:07 event3
0 crw-rw----  1 root input 13, 68 Mär  2 06:07 event4
0 crw-rw----  1 root input 13, 69 Mär  2 06:07 event5
0 crw-rw----  1 root input 13, 70 Mär  2 06:07 event6
0 crw-rw----  1 root input 13, 71 Mär  2 06:07 event7
0 crw-rw----  1 root input 13, 72 Mär  2 06:07 event8
0 crw-rw----  1 root input 13, 73 Mär  2 06:07 event9
0 crw-rw-r--+ 1 root input 13,  0 Mär  2 06:07 js0
0 crw-rw----  1 root input 13, 63 Mär  2 06:07 mice
0 crw-rw----  1 root input 13, 32 Mär  2 23:16 mouse0
0 crw-rw----  1 root input 13, 33 Mär  2 06:07 mouse1
0 crw-rw----  1 root input 13, 34 Mär  2 23:15 mouse2
0 crw-rw----  1 root input 13, 35 Mär  2 06:07 mouse3
```
If you find a /dev/input/ckb1 directory, please send us the output of three files:
```lutz@Mainfrix /dev/input/ckb1 $ cd /dev/input/ckb1
lutz@Mainfrix /dev/input/ckb1 $ ll
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Mär  2 23:16 cmd
4,0K -rw-r--r-- 1 root root 47 Mär  2 23:16 features
4,0K -rw-r--r-- 1 root root  5 Mär  2 23:16 fwversion
4,0K -rw-r--r-- 1 root root 32 Mär  2 23:16 model
   0 prw-r--r-- 1 root root  0 Mär  2 23:16 notify0
4,0K -rw-r--r-- 1 root root  5 Mär  2 23:16 pollrate
4,0K -rw-r--r-- 1 root root 33 Mär  2 23:16 serial
lutz@Mainfrix /dev/input/ckb1 $ cat model
Corsair K95 RGB Gaming Keyboard
lutz@Mainfrix /dev/input/ckb1 $ cat fwversion 
0205
lutz@Mainfrix /dev/input/ckb1 $ cat features 
corsair k95 rgb bind notify fwversion fwupdate
```
Especially the output of the model and the features are relevant for finding the correct firmware file.
IMHO our decision, that 0x1b36 is a monochrome device, but we handle it as a RGB, leads to your issue. with the info from your files I can check if I'm right.

bulletmark, 2017-03-02:
@frickler24 we discovered this later in IRC where I did that and we established what @tatokis says above.

frickler24, 2017-03-02:
@tatokis I'm in the same file, but more at line #153 (static QString tableName(const QString& features))

bulletmark, 2017-03-02:
~~~~
pc:/dev/input/ckb1 ls -l
total 20
prw-rw-rw- 1 root root  0 Mar  3 08:27 cmd
-rw-r--r-- 1 root root 56 Mar  3 08:27 features
-rw-r--r-- 1 root root  5 Mar  3 08:27 fwversion
-rw-r--r-- 1 root root 32 Mar  3 08:27 model
prw-r--r-- 1 root root  0 Mar  3 08:27 notify0
-rw-r--r-- 1 root root  5 Mar  3 08:27 pollrate
-rw-r--r-- 1 root root 33 Mar  3 08:27 serial
pc:/dev/input/ckb1 cat fwversion 
0204
pc:/dev/input/ckb1 cat features 
corsair k70 rgb pollrate bind notify fwversion fwupdate
pc:/dev/input/ckb1 cat model
Corsair Gaming K70 LUX Keyboard
~~~~

tatokis, 2017-03-02:
@frickler24 I see exactly what you mean. I suggest making a blacklist for devices that shouldn't be updated due to known bugs, and add in the future add a new class for the K70 NRGB so that it doesn't show up as an RGB device, but still has the right commands.

Alternatively, we could modify the firmware file so that the GUI gets the firmware files based on USB IDs, which should be much better in my opinion, however that could possibly break old GUI builds.

frickler24, 2017-03-02:
@tatokis Your idea to take the productID will be the best. At least we use a short table productID<==>nameOfFirmware, because a lot of versions will take the same fw-file.
In @bulletmark 's case the features file leads directly to a K70RGB, which is P_K70 in usb.h and has productID 0x1b13.

algtmatt, 2017-03-03:
fwiw, I've not been able to update my K70 LUX RGB either. The 2.04 and now 2.05 updates won't apply.

It seems like I'm getting a mismatch between the idProduct and firmware version.

I'll try to dig in a bit this weekend and see if I can get a fix together but wanted to add some additional info to this ticket now.

ckb-daemon[1124]: [E] fwupdate (firmware.c:78): Firmware file /tmp/ckb-3197-fwblob.bin doesn't match device (V: 1b1c P: 1b13)

Bus 002 Device 010: ID 1b1c:1b33 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b33 
  bcdDevice            1.05
  iManufacturer           1 Corsair
  iProduct                2 Corsair Gaming K70 LUX RGB Keyboard 
  iSerial                 3 xx
  bNumConfigurations      1

% ls -ahl /dev/input/ckb2/
total 20K
drwxr-xr-x 2 root root 200 Mar  2 15:21 ./
drwxr-xr-x 7 root root 680 Mar  2 15:21 ../
prw-rw-rw- 1 root root   0 Mar  2 15:37 cmd|
-rw-r--r-- 1 root root  56 Mar  2 15:21 features
-rw-r--r-- 1 root root   5 Mar  2 15:21 fwversion
-rw-r--r-- 1 root root  35 Mar  2 15:21 model
prw-r--r-- 1 root root   0 Mar  2 15:21 notify0|
prw-r--r-- 1 root root   0 Mar  2 15:37 notify1|
-rw-r--r-- 1 root root   5 Mar  2 15:21 pollrate
-rw-r--r-- 1 root root  33 Mar  2 15:21 serial

% cat fwversion
0105

% cat model
Corsair Gaming K70 LUX RGB Keyboar

% cat features
corsair k70 rgb pollrate bind notify fwversion fwupdate

tatokis, 2017-03-03:
@frickler24 I remember now, I was trying to solve this some months ago, 
before ckb-next was formed.
My plan was to create a "lux" and "rfire" feature, so that they can be 
used by the code that upgrades the fw, but I ended up scrapping the 
whole thing.

We could possibly expand the FIRMWARE table to do associations between 
model and dev id, so that we don't break the existing format.

frickler24, 2017-03-03:
@tatokis The problem I see is, that the communication between ckb and ckb-daemon is not clear to this aspect. This leads to your issue mentioned,  @algtmatt.:

ckb-daemon talks to the usb interfaces and handles VendorID:ProductId.
ckb handles models, names, features, but definitely no ProductID.

So the best design change IMHO is the following:
In the init phase of the daemon, where the files feature, model, fwversion, pollrate and serial files are created, we should add a file product-id and save the 0x.... in it. Or we add that info to the version file.

In the ckb functions, which handle fw-search, upgrade and so on, we change the algorithm to determine the device type and read the new info.

With this info we implement a lookup function for a new column in FIRMWARE:
[image]

Alternatively - if we do not want to sign the FIRMWARE file just because a new device is added or an entry has to be changed - we can implement this lookup table in kbfirmware.[ch] also, because for this info the signing is not neccessary (no download from the web as for the firmware-zips).

(@mattanger and @light2yellow also): Please give me some feedback to this . Tomorrow I've a hard day in our company (release day), so maybe on sunday I have some time to think a bit more about it.

light2yellow, 2017-03-04:
@frickler24 regarding FIRMWARE: both solutions seem good to me, both have its rationale. But from my point of view putting id arrays into FIRMWARE is more logical since we already have product names there. 

Also, we probably need to create a markdown document somewhere to gather all products and their idd altogether for an easy lookup when a new issue is opened.

And while we're at it, I just saw that the FIRMWARE has a column that requires version XYZ from ckb. That's an important thing, something to not forget about in the future. And this something is a proper tag. Will it work if the tag is `v0.2.8` for example? I mean, is `v` discarded and the numbers are compared with a dot as a separator? Because this is how Github proposes the tags to be done and how most of the tags were done, so this is how we will do our new tag too. Linking it to #32.

Also, I do not really understand why we didn't have the firmware matching issues earlier, until the file was updated. Is it just a coincidence?

frickler24, 2017-03-05:
@light2yellow Thank you for the hints.
I believe that we have the trouble only with hardware which has inconsistency between the names and its features. eg why has a kb with one color the feature rgb?
Thinking about this, I found the following lines in fwupgradedialog.cpp:
```
static KbId ids[] = {
    // Keyboards
    { 0x1b1c, 0x1b17, "corsair k65 rgb" },
    { 0x1b1c, 0x1b13, "corsair k70 rgb" },
    { 0x1b1c, 0x1b11, "corsair k95 rgb" },
    { 0x1b1c, 0x1b15, "corsair strafe monochrome" },
    { 0x1b1c, 0x1b20, "corsair strafe rgb" },
    // Mice
    { 0x1b1c, 0x1b12, "corsair m65 rgb" },
    { 0x1b1c, 0x1b1e, "corsair scimitar rgb" },
    //SABRE CH-9000111-EU
    { 0x1b1c, 0x1b32, "corsair sabre optical rgb" }
};
```
I'm not yet sure, but I think, this is the mechanism we are looking for. Up to now it is used to check **after** loading a fw-file, wheter it is the correct one. Maybe we can used it for our purpose **before** addressing FIRMWARE. Give me some time to analyze the code...

frickler24, 2017-03-05:
The code knows about the  the structure of the firmware-file at positions 0x102 with length 6 Bytes for: VendorID, ProductID, VersionNumber. Each Info 2 bytes long. 
VersionNumber is the version in hex, eg 0x0205 for v2.05
Interesting is the VersionNumber of FREYA_App_v1.07.123.bin, because only v1.07 is stored.
So we should shorten the Version Number in our FIRMWARE file.
For the Firmware-files I downloaded in the last time, it looks like
```
lutz@Mainfrix ~/Projekte/ckb-next/.firmware $ find . -name \*.bin -exec hd -s 0x102 -n 8 -v {} \;
00000102  1c 1b 36 1b 04 02 50 30                           |..6...P0|
0000010a
00000102  1c 1b 33 1b 05 02 50 30                           |..3...P0|
0000010a
00000102  1c 1b 3a 1b 05 02 50 30                           |..:...P0|
0000010a
00000102  1c 1b 38 1b 05 02 50 30                           |..8...P0|
0000010a
00000102  1c 1b 2d 1b 07 01 50 30                           |..-...P0|
0000010a
```
The code checks the VendorID, ProductID of the Firmware and the entry in the ids[] and the first parts of the features-file against the third part of ids[]. If they match, the firmware is accepted and the VersionNumber is read from the firmware-blob also.

So this is a third (fourth?) redundant part to check for vendor, product, feature, technical capabilities, fw-version.
I think I#ll draw a picture and think about restructuring.

About your point version-number of the client: This info is used to determine, if a new fw-version will be presented to the user. If the fw-version does not fit to the client version, it is not shown.
What happens if we download and install a fw-version manually and afterwards have an older or newer client-version and/or newer fw-version than noted in FIRMWARE - I don't know.

frickler24, 2017-03-08:
I started an implementation to handle these firmware constellations (416232c in one of [my branches](https://github.com/frickler24/ckb-next/tree/issue-97-K70-LUX-firmware-update-fails). Do not merge this branch into testing, because it contains more than the bugfix-code yet. I'll correct this when the fix has been finished.). 
Now I'm some days offline and hope to continue the implementation next week.

tatokis, 2017-04-13:
@frickler24 Any updates on this? If not, could you please make a short description of what is left to be implemented so that someone can help?

Thanks

frickler24, 2017-05-05:
@bulletmark @algtmatt any new findings on that? Is PR #155 working for your devices?

bulletmark, 2017-05-05:
@frickler24 I only originally installed ckb-next to update my firmware and don't have it installed anymore but if you tell me clearly what to install I will test it to prove this issue is addressed or not.

frickler24, 2017-05-06:
@bulletmark If you don't mind, please do the following test:
```
cd /tmp    # or where you like to
git clone https://github.com/frickler24/ckb-next.git --single-branch --branch testing+issue97-K70-LUX-firmware-update-fails --depth 1
cd ckb-next/
yes|quickinstall
ckb --close && sleep 1 && ckb
```
This should bring some output like
```
Asking existing instance to close.
Path  to settings is "/home/xxxxxx/.config/ckb/ckb.conf"
Scanning  "/usr/lib/ckb-animations/ckb-gradient"
...
first look for special firmware returned -1
Reading 3 entries from fwversion file, VID= "0x1b1c" , PID= "0x1b11"
Firmware version read from ckb-daemon in fwversion file is "0205"
...
Downloaded new firmware list. 12 entries found.
```
Please post what you get as debug output.
You may deinstall ckb-next by calling `make clean && cd .. && rm -rf ckb-next` afterwards.
Thanks in advance!

@ all: Is there someone with a K70 Lux and firmware 2.04 or less?

bulletmark, 2017-05-06:
After the `quickinstall` command above the new ckb + gui started itself and I immediately got a popup saying "A new firmware is available for your K70 LUX  (v2.05)." I did not have to explicitly start ckb as your instructions say to.

I presently have v2.04. Given your last question above, do you want me to do something else before I press the update button?

frickler24, 2017-05-07:
@bulletmark please run the client manually (after quickinstall it is started automatically, but I get no debug info). Please answer the question about new FW with no. First I want to check which download it wants to use. 
The `ckb --close && sleep 1 && ckb` stops the client and starts a new one. Please answer "no" again for the FW upgrade and send me the output from ckb client.

bulletmark, 2017-05-07:
Output is:
```
c:~/ckb-next ckb
Path  to settings is "/home/mark/.config/ckb/ckb.conf"
Scanning  "/usr/lib/ckb-animations/ckb-gradient"
Scanning  "/usr/lib/ckb-animations/ckb-heat"
Scanning  "/usr/lib/ckb-animations/ckb-mviz"
Scanning  "/usr/lib/ckb-animations/ckb-pinwheel"
Scanning  "/usr/lib/ckb-animations/ckb-rain"
Scanning  "/usr/lib/ckb-animations/ckb-random"
Scanning  "/usr/lib/ckb-animations/ckb-ripple"
Scanning  "/usr/lib/ckb-animations/ckb-wave"
first look for special firmware returned 1
Reading 3 entries from fwversion file, VID= "0x1b1c" , PID= "0x1b36"
Firmware version read from ckb-daemon in fwversion file is "0204"
Downloaded new firmware list. 12 entries found.
```

frickler24, 2017-05-07:
OK, thank you. 
As I was afraid, unfortunately it does not run as I thought - it selects the wrong file.
Currently I have a discussion with @tatokis about it at #155.

bulletmark, 2017-05-07:
So I should wait for another update from you before I attempt to update to firmware V2.05?

frickler24, 2017-05-08:
There is no newer firmware for your device. The software detected another one for you, but this is wrong.
So you have actual firmware in your device with 2.04.
You can delete the branch "testing+issue97-K70-LUX-firmware-update-fails".
Thanks for testing. I call again when the new softare version is ready for test.

bulletmark, 2017-05-08:
No worries. Note that you don't provide anything to uninstall all the stuff that gets installed into our system. Please consider adding the following `uninstall.sh` script to your repo:
```
#!/bin/sh
echo "Uninstalling ckb .."

if [ $(id -u) != 0 ]; then
    echo "Must use sudo or be root."
    exit 1
fi

echo "Shuting down ckb-daemon .."
systemctl disable ckb-daemon 2>/dev/null
systemctl stop ckb-daemon 2>/dev/null
service ckb-daemon stop 2>/dev/null
rc-service ckb-daemon stop 2>/dev/null
rc-update del ckb-daemon default 2>/dev/null

echo "Removing ckb files .."
rm -vrf \
    /usr/lib/systemd/system/ckb-daemon.service \
    /etc/init.d/ckb-daemon \
    /etc/init/ckb-daemon.conf \
    /usr/bin/ckb \
    /usr/bin/ckb-daemon \
    /usr/share/applications/ckb.desktop \
    /usr/share/icons/hicolor/512x512/apps/ckb.png \
    /usr/lib/ckb-animations \
    /usr/bin/ckb-animations
```

mbostwick, 2017-05-14:
also having issues with firmware...firmware loads from

https://raw.githubusercontent.com/mattanger/ckb-next/master/FIRMWARE

frickler24, 2017-06-03:
@mbostwick @bulletmark please have a  look at the new approach #220

ntom, 2017-06-08:
I also get this update firmware prompt for my k70 - 'update failed' is always reported, no error log anywhere to copy/paste ... i would like to update if possible, maybe fix the buggy issues im still having ?

frickler24, 2017-06-10:
@ntom Did you test the PR #220?

PandorasFox, 2018-01-14:
So, I had this issue with my K70 LUX RGB! I got it fresh with 1.05 and fw updates to 2.05 would fail.

I got a bit bored and downloading the fw zip and extracted it, then waited for CKB to prompt me to start the flash process (after the fw file was dropped into /tmp).

Replacing that file in /tmp with the `K70LUXRGB_APP_V205.bin` from the zip, and _then_ hitting "start", worked perfectly fine, and now my LUX is happily on 2.05.

I'm guessing that some older FW versions don't support flashing straight from a .zip and require the .bin, which is why they fail to update.

ZirconiumX, 2018-01-14:
@PandorasFox Nice theory, but no. What's happening is that the GUI thinks you have a K70 RGB, not a K70 LUX, so it downloads the firmware for the K70 RGB. The daemon then checks the firmware file matches the device and fails because the K70 RGB and K70 LUX are different devices. Your method works because you're replacing the K70 RGB firmware with K70 LUX firmware, which passes the device check and thus gets flashed.

tatokis, 2018-01-21:
Should be fixed in https://github.com/ckb-next/ckb-next/

closed 2018-01-21 by tatokis