mattanger/ckb-next #98: Unable to install (Manjaro Arch)

RobMoore606, 2017-03-02:
I just cannot get this to install using the quickinstall file either by double clicking or copying the contents into terminal. even tried the testing files
I do get an install via pacman and have tried **ckb-git**, **ckb-next-git**, **ckb-git-latest** and **ckb-next-latest-git** but no devices connected
**Bus 007 Device 003: ID 1b1c:1b2e Corsair** 
Bus 007 Device 002: ID 047f:aa01 Plantronics, Inc. 
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 04d9:1203 Holtek Semiconductor, Inc. Keyboard
**Bus 005 Device 002: ID 1b1c:1b17 Corsair**

light2yellow, 2017-03-02:
So what *is* the problem? You cannot install (in the sense "cannot compile/execute/whatever") or it does install, but the device is not detected? Please also describe the device model in English (and make it into the title).

P. S. Regarding AUR packages, I published a deprecation message and submitted a deletion request for `ckb-git` and `ckb-git-latest` today. Use only those described in the README. Should probably link this to #5 for the sake of consistency.

tatokis, 2017-03-02:
1b17 is the normal k65 rgb, and it should be fully supported. 
My guess is that the daemon hasn't stopped, so you are using an older version. 

Uninstall what packages you may have installed, and reboot. 

Once that's done, try the latest master by downloading the source code and executing in the terminal: ./quickinstall 

If that falls, copy any output from the terminal and paste it here.

RobMoore606, 2017-03-02:
Thanks @tatokis that did the trick many thanks to the team working on this app it is better than the original and I look forward to seeing it grow :)

closed 2017-03-02 by RobMoore606