mattanger/ckb-next #99: ckb command line  warnings and segmentation fault in linux

hevanaa, 2017-03-02:
Running ckb on the command line produces various warnings and an error depending on which switch is used. 

1.
```
$ ckb

(ckb:18759): GLib-GObject-WARNING **: g_object_get_valist: object class 'GtkSettings' has no property named 'gtk-application-prefer-dark-theme'

(ckb:18759): GLib-GObject-WARNING **: g_object_get_valist: object class 'GtkSettings' has no property named 'gtk-long-press-time'
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
```

ckb window is correctly opened. But the above messages are not really helpful.

2.
```
$ ckb -h

(ckb:18909): GLib-GObject-WARNING **: g_object_get_valist: object class 'GtkSettings' has no property named 'gtk-application-prefer-dark-theme'

(ckb:18909): GLib-GObject-WARNING **: g_object_get_valist: object class 'GtkSettings' has no property named 'gtk-long-press-time'
Usage: ckb [options]
Open Source Corsair Input Device Driver for Linux and OSX.

Options:
  -v, --version     Displays version information.
  -h, --help        Displays this help.
  -b, --background  Starts in background, without displaying the main window.
  -c, --close       Causes already running instance (if any) to exit.
Segmentation fault (core dumped)
```
The segmentation fault makes it impossible to even try to create a man page with help2man (but I can't say if an auto-generated man page would be any good at this point).

tatokis, 2017-03-02:
The gtk errors are beyond our control, as ckb uses qt. 
The png ones are pretty harmless. 

That segfault, however, should definitely not happen. Can you please run it through gdb and post a full backtrace?

light2yellow, 2017-03-02:
> GLib-GObject-...

**These have nothing to do with ckb-next.** Because it's how Qt implements its backend on GNOME. Update your Qt and if this persists go open an issue on Qt bugtracker. Moreover, #70, #51, ccMSC/ckb#500, ccMSC/ckb#461.

> libpng warning: iCCP: known incorrect sRGB profile

Already fixed on `cmake-bustling`.

> Segmentation fault (core dumped)

Interesting. Worksforme (on the branch above). If you have no idea what to do (like me), read [Step-by-step debugging guide](https://wiki.archlinux.org/index.php/Step-by-step_debugging_guide) and google "gdb debug segfault".

hevanaa, 2017-03-02:
Here is a backtrace:
[ckb_h_backtrace.txt](https://github.com/mattanger/ckb-next/files/815081/ckb_h_backtrace.txt)

QT version is 5.7.1 (as provided by Fedora 25).

light2yellow, 2017-03-02:
Here's what I get (on `testing`):
```
Starting program: /usr/bin/ckb -h
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe8b09700 (LWP 26732)]
[New Thread 0x7fffe3fff700 (LWP 26733)]
[New Thread 0x7fffdad78700 (LWP 26735)]
[New Thread 0x7fffda577700 (LWP 26736)]
[New Thread 0x7fffd9d76700 (LWP 26737)]
[New Thread 0x7fffd8c96700 (LWP 26738)]
[New Thread 0x7fffcbfff700 (LWP 26739)]
Usage: /usr/bin/ckb [options]
Open Source Corsair Input Device Driver for Linux and OSX.

Options:
  -v, --version     Displays version information.
  -h, --help        Displays this help.
  -b, --background  Starts in background, without displaying the main window.
  -c, --close       Causes already running instance (if any) to exit.
[Thread 0x7fffe3fff700 (LWP 26733) exited]
[Thread 0x7fffcbfff700 (LWP 26739) exited]
[Thread 0x7fffd8c96700 (LWP 26738) exited]
[Thread 0x7fffda577700 (LWP 26736) exited]
[Thread 0x7fffdad78700 (LWP 26735) exited]
[Thread 0x7fffe8b09700 (LWP 26732) exited]
[Thread 0x7ffff7e41280 (LWP 26725) exited]
[Inferior 1 (process 26725) exited normally]
(gdb) 
```
nice and clean. Ubuntu 16.10.

This is really strange. Again, it looks more like Qt issue than a ckb-next problem. Maybe somebody can tell us more.

light2yellow, 2017-03-02:
Couldn't you downgrade Qt by any chance? I'm using 5.6.1 and @tatokis uses 5.5.1. I will try this on Arch Linux as well in a few minutes.

hevanaa, 2017-03-02:
I tried in Fedora 24 with Qt 5.6.1 and there it works fine. 

```
$ ckb
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
```

```
$ ckb -h 
Usage: ckb [options]
Open Source Corsair Input Device Driver for Linux and OSX.

Options:
  -v, --version     Displays version information.
  -h, --help        Displays this help.
  -b, --background  Starts in background, without displaying the main window.
  -c, --close       Causes already running instance (if any) to exit.
```

light2yellow, 2017-03-03:
Ha! Cool. So you've found a bug in Qt.

Here's Qt 5.8.0 on Arch Linux (gdb 7.12.1):
```
(gdb) r -h
Starting program: /usr/bin/ckb -h
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffe61e8700 (LWP 13623)]
Usage: /usr/bin/ckb [options]
Open Source Corsair Input Device Driver for Linux and OSX.

Options:
  -v, --version     Displays version information.
  -h, --help        Displays this help.
  -b, --background  Starts in background, without displaying the main window.
  -c, --close       Causes already running instance (if any) to exit.
[Thread 0x7ffff7f15240 (LWP 13619) exited]
[Inferior 1 (process 13619) exited normally]
(gdb) 
```

hevanaa, 2017-03-03:
Fedora 26 (release in June) will have Qt 5.8. So maybe time fixes the problem.

light2yellow, 2017-03-03:
Yep, sooner or later even Debian will update the packages. Still, that's unfortunate.

If you have time and wish you can report this to Qt. I cannot do this myself because I don't have a repro. I tried launching Fedora live ISO in Virtualbox and it said I cannot install packages, and the "Install Fedora" app did not even start. So there's no easy way for me to maintain a report (in case they would like to know more details).

light2yellow, 2017-03-03:
Ironically, today I got a segfault as well. But at the boot time instead. The kb was completely dark and not responding until physically replugged in. After replugging it was successfully recognized by ckb-daemon. It was built from `testing`. Ubuntu 16.10, Qt 5.6.1.

N.B. I do not use ckb generally. This one was installed yesterday for testing purposes regarding this issue and I just didn't uninstall it. So the problem could've been there much earlier.

[_usr_bin_ckb.1000.crash.txt](https://github.com/mattanger/ckb-next/files/817218/_usr_bin_ckb.1000.crash.txt)

light2yellow, 2017-03-09:
From what I saw in the logs and googled afterwards, my bug is a GTK/Qt/Xorg bug revealed by other projects more than 9 years ago (just google some of the errors right before ckb is killed), so it's not related to the @hevanaa's issue.

> Already fixed on `cmake-bustling`.
>> libpng warning: iCCP: known incorrect sRGB profile

Now the fix is in the `testing` too.

And since the issue is not related to ckb-next directly and it got somewhat resolved I'm closing this with a notice that we should add this as a known upstream issue and possibly inform a user about the problems with Qt 5.7 (however this also might be just Fedora's package, so I'd wait if someone reports this on the other distro too).

closed 2017-03-09 by light2yellow