mattanger/ckb-next #100: M65 Pro RGB Firmware Update Fails

viggy96, 2017-03-03:
Exactly the title. CKB client prompts for firmware update, however, upon clicking through to install it, the installation fails. I really would like for this to work, as I have pretty much given up on Windows. 

P.S. The mouse still works fine.

frickler24, 2017-03-03:
@viggy96 please give me some more info:
- what version do you have in the M65 at the moment (you can find it in the logfiles or in the version file in /dev/input/ckb# or similar on Mac)
- what version and from what branch are you running ckb-next?
- If you start your ckb in a terminal: Is there any error message corresponding to fw-ugrade? You need at least 7301ee8 from branch master, because in this commit the new FIRMWARE file ans signature is committed.

viggy96, 2017-03-03:
- Its got version 2.02, (or 0202 as stated by /dev/input/ckb1/fwversion) 
- I am running the testing version of ckb-next, a2d4d7e
- I get this error in the terminal when running the upgrade:
`fwupdate (firmware.c:78): Firmware file /tmp/ckb-725-fwblob.bin doesn't match device (V: 1b1c P: 1b12)
`

frickler24, 2017-03-03:
@viggy96 OK, you have the same error as #97 .
So I close this one here and we need to fix it anyhow.

closed 2017-03-03 by frickler24