mattanger/ckb-next #101: Strafe Connecting at Boot issue

RaymondSchnyder, 2017-03-03:
Hey,

I have a weird issue with my Strafe Non RGB from Corsair.
While boot, i have to slay on the keys on the keyboard like a speed addict, to provocate an usb hid error, so that the keyboard works. I already added the quirks parameter into grub and made the "0x4" change. but nothing helps.
I use archlinux and on every other distribution, it works out of the box, but not on arch.

Is there any logs or sysfiles i can look at to resolve this? 
Thank you

frickler24, 2017-03-03:
@RaymondSchnyder in the kernel log files (arch linux uses systemd? If so, use journalctrl) search for ckb-daemon. Around messages from ckb-daemon you can find usb infos from the kernel.
They may bring you some more info. Up to now I did not have such an effect as you describe.

RaymondSchnyder, 2017-03-04:
This is from a successfull boot with the "workaround" i found at boot (beat the keys on the keyboard like a insane person)

-- Logs begin at Fri 2016-12-30 13:44:03 CET, end at Sat 2017-03-04 10:17:36 CET. --
Mär 04 10:17:20 arch-anywhere systemd[1]: Started Corsair Keyboard Daemon.
Mär 04 10:17:20 arch-anywhere ckb-daemon[336]:     ckb: Corsair RGB driver beta-v0.2.7
Mär 04 10:17:20 arch-anywhere ckb-daemon[336]: [I] Root controller ready at /dev/input/ckb0
Mär 04 10:17:20 arch-anywhere ckb-daemon[336]: [I] Connecting Corsair STRAFE Gaming Keyboard at /dev/input/ckb1
Mär 04 10:17:20 arch-anywhere ckb-daemon[336]: [I] Starting input thread for /dev/input/ckb1
Mär 04 10:17:21 arch-anywhere ckb-daemon[336]: [E] loadrgb_kb (led_keyboard.c:224): Bad input header
Mär 04 10:17:21 arch-anywhere ckb-daemon[336]: [E] loadrgb_kb (led_keyboard.c:228): color = 0, i = 1, mode = 0
Mär 04 10:17:21 arch-anywhere ckb-daemon[336]: Input(Antwort): 0e ff 01 3c ff ff ff ff
Mär 04 10:17:21 arch-anywhere ckb-daemon[336]: Output (Frage): 0e ff 01 3c
Mär 04 10:17:26 arch-anywhere ckb-daemon[336]: [E] os_usbrecv (via led_keyboard.c:233): Connection timed out
Mär 04 10:17:26 arch-anywhere ckb-daemon[336]: [W] _start_dev (device.c:45): Unable to load hardware profile
Mär 04 10:17:26 arch-anywhere ckb-daemon[336]: [I] Setup finished for /dev/input/ckb1

light2yellow, 2017-03-04:
Merged into #96, references #95.

closed 2017-03-04 by light2yellow