mattanger/ckb-next #105: k65 LUX RGB Not detected.

startupfoundry, 2017-03-04:
Hello, I've built the most recent version of ckb-next but I am unable to detect my k65 LUX RGB. I'm on OS X 10.12.3.

[image]

frickler24, 2017-03-05:
@startupfoundry if you have got the current master branch from ckb-next, you should have ckb-next 0.2.7, not ckb 0.2.4

vagrant10, 2017-03-06:
Hi, I'm also thinking of getting a k65 Lux RGB, however just to check if cbk-next will work (functional including RGB), as it's not listed in the supported list. 

Background - Sierra 10.12.2 and my country doesnt sell K65 RGB anymore. 

Thanks everyone!

light2yellow, 2017-03-06:
@vagrant10 see #2. If you don't have a device nd can't provide any information, just press the "Subscribe" button and wait. We have no ETAs, so there's that.

light2yellow, 2017-03-06:
@startupfoundry excuse me, I was distracted by @vagrant10.
And yes, @frickler24 is right, you are using the old ckb, current master branch looks like that:

[image]

Try at least it and if no success, try `testing`.

Then report back.

auibrian, 2017-03-07:
@startupfoundry Make sure you are running the ckb-daemon, I have a k65 RGB and it works fine (barring the issues listed on github here)

startupfoundry, 2017-03-09:
I got it working. Thanks for your comments. I literally just built it from source again (the same way as I did previously) and it worked.

closed 2017-03-09 by startupfoundry

mik115, 2017-03-13:
hi @auibrian,
i've the same problem of @startupfoundry but i've installed the latest version of ckb-next and i have a k65 rgb rapidfire.

[image]

any suggestions?

light2yellow, 2017-03-13:
@mik115 reboot, replug the kb.

mik115, 2017-03-13:
@light2yellow already done multiple times, nope.

light2yellow, 2017-03-13:
@mik115 then execute this script https://github.com/mattanger/ckb-next/blob/cmake/macos/uninstall.sh to be sure that every instance (and even the old one) is dead and recompile the ckb-next (one last time, I promise) from `testing` (for example, it should work :) ). 

In case the kb is still not showing, execute:
```sh
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
sudo /usr/bin/ckb-daemon
```
and provide the output.

You can also try the second option first, if you feel confident about what you've already tried.

mik115, 2017-03-13:
Now it works! thx!

light2yellow, 2017-03-13:
@mik115 

[image]

mik115, 2017-03-13:
**You truly are the wisest wizard cat** 🤣  🤣