mattanger/ckb-next #107: Strafe sidelightToggled issue

jp94, 2017-03-09:
On ckb start, sidelight is on regardless of user settings. This also overrides 0% brightness setting.
If sidelight is toggled on while brightness is at 0%, interacting with GUI to toggle on/off sidelight will not work until brightness is adjusted above 0%.

I've tested on master and testing branch. This issue existed since /ckb branch, but I didn't bother to bring it up until now.

tatokis, 2017-04-03:
CC @light2yellow 
Any ideas?

light2yellow, 2017-04-14:
> On ckb start, sidelight is on regardless of user settings.

I can confirm this.

> This also overrides 0% brightness setting. If sidelight is toggled on while brightness is at 0%, interacting with GUI to toggle on/off sidelight will not work until brightness is adjusted above 0%.

I can confirm this too.

Thank you for reporting this.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/8

closed 2018-01-22 by tatokis