mattanger/ckb-next #108: .xinit compose key setting overridden

hoclun-rigsep, 2017-03-10:
Thanks for keeping this alive. 

For a long time I've had a .xinit file with the following contents:
`setxkbmap -option "compose:caps"`
I noticed that since installing ckb a few days a go sometimes I go to use the compose key and it is back to being a capslock key and I have to manually source .xinit again to switch back. Any idea why this would happen/any cheap fixes for it? Thanks; I'm not an expert with the X input system etc.

light2yellow, 2017-03-10:
I don't understand. So when you press the Compose key it sometimes acts as a Compose key but you want it always act as a Caps Lock?

light2yellow, 2017-03-10:
And how does ckb ruin it?

hoclun-rigsep, 2017-03-10:
Allow me to rephrase. As seen in my xinit script, I have remapped caps lock to function as compose. However since installing ckb a few days ago and using my k70 rgb, I notice that the caps lock key sometimes suddenly reverts, i.e. it stops composing and starts toggling caps again.

I will attempt to see if anything more particular triggers this effect and report back here.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

light2yellow, 2017-03-10:
I see now, thanks for the re-explanation. This is an interesting issue.

hoclun-rigsep, 2017-03-10:
Yeah, so I use KDE5 on Linux Mint. My caps lock key has been working as compose key, as expected, all day, but just a minute ago it stopped being a compose key and started toggling caps. I can think of nothing in particular which I did in the last few minutes that I hadn't already been doing earlier in the session.

hoclun-rigsep, 2017-03-10:
Also: 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                             
    17752 jim       20   0  905992  60960  45244 S   9.3  0.7 133:27.24 ckb

Is that kind of CPU usage normal for ckb? I didn't have it doing anything.

tatokis, 2017-03-10:
Not going to lie, I have never encountered this, and I use the compose key daily.
However, I do not have it bound using .xinit, but use the built in keyboard settings in unity-control-center.

Maybe KDE has something similar you can try?

light2yellow, 2017-03-11:
@hoclun-rigsep please open another issue or find a similar one regarding another problem.

hoclun-rigsep, 2017-03-11:
@tatokis, than you for the suggestion. I checked and it turns out I already had KDE's "Position of Compose key" set for Caps Lock.

light2yellow, 2017-03-29:
@hoclun-rigsep is it solved?

hoclun-rigsep, 2017-04-03:
I'm not sure what I changed but I haven't had this issue since discussing it here. Sorry that isn't more helpful feedback.

closed 2017-04-03 by hoclun-rigsep