mattanger/ckb-next #109: ckb CPU usage

hoclun-rigsep, 2017-03-11:
System: Linux Mint 18 "Sarah" with KDE5.
Keyboard: Corsair k70 LUX RGB (1404001FAEBD15015627CF1DF5001940)
Firmware version: 2.05 (as reported by ckb

I've noticed that `top' reports ckb usage of between 10% and 35%. I haven't programmed ckb to do anything special, just change the color of the LEDs from red to white and back again. 

Is this expected behavior? I will help gather information on this if you give me instructions. 

Thank you.

light2yellow, 2017-03-11:
I doubt this is an expected behavior, nor have I any knowledge on how to profile a long-running application to see where the most processor time is spent.

Might be related to #42.

skogsmaskin, 2017-03-11:
Same issue here. It seems to be related to the "Show animated" option. When I uncheck those, CPU usage falls drastically (from about 35% to about 7%). It's still using 7-8% though, even minimized to tray.

tatokis, 2017-03-11:
Unfortunately it takes a lot of CPU time to refresh the Qt GUI in order to show the animations onscreen.

I don't think there's anything that can be done for this, other than disabling "show animated" as mentioned by @skogsmaskin.
When minimizing the GUI to tray, CPU usage should drop due to the fact that no updates are being sent to the GUI.

The remaining 7% CPU usage is due to the animations running, and/or ckb-next just sending usb packets at 30 'frames' per second to the keyboard.

You can minimize this by lowering the framerate the keyboard is updated at from settings -> more settings.

hoclun-rigsep, 2017-03-12:
Thanks for the feedback; I will try it.

closed 2017-04-03 by tatokis