mattanger/ckb-next #110: Issue template

light2yellow, 2017-03-11:
Below is my vision on how an issue template should look like.

----

Switch on the "Preview" mode for better experience.

# Prerequisites

* __Read__ the [`README.md`](https://github.com/mattanger/ckb-next/blob/master/README.md). __Skim__ through the other `.md` documents. Do this right now if you didn't. The chances are it contains [at least] a mention of your problem or [at most] a solution to it. __Avoid__ opening a new issue in case of the latter. Instead, find an existing issue related to the particular problem/solution and describe your case there. Don't worry, we get the notifications for closed issues as well. It will be reopened if needed.
* __Search__ before submitting an issue. Do this right now if you didn't. The chances are there is [at least] a mention of your problem or [at most] a solution to it.
* __Use Markdown__ to write your messages. [Markdown Cheatsheet](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet). Special attention to triple backticks. Use them if you copy-paste more than _one_ line.
* __Remove all unused sections (including this one) after filling everything in below.__

# Type of the issue

Choose only one, either a "Feature request" or a "Bug report".

## Feature request

#### What is the current behaviour of the programme?
...

#### What would you like it to do instead?
...

#### Any other notes & comments?

(In case of a new device support upload the output of `ckb-dev-detect` script from the root of the source tree if you are on Linux, otherwise upload a dump like in [this](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447) comment)
...

## Bug report

### Gathering information

#### OS
__Linux__: (replace these parentheses with your Linux distribution name and version (in case it is versioned) and `uname -r` output, for example: _"Arch Linux 4.6.0-1-surfacepro3"_ or _"Ubuntu 16.10 4.8.0-41-generic"_)

__macOS__: (replace these parentheses with your macOS version, for example: _"macOS 10.12.3"_ or _"Hackintosh 10.12.3"_)

#### ckb-next info
__From__: (replace these parentheses with the information describing how did you get this programme, for example: _"`ckb-next-git` package in AUR"_ or _"built manually using quickinstall script"_ or _"pkg for macOS"_)

__Version__: (replace these parentheses with ckb-next's version which can be found in the "Settings" tab in the GUI, for example: _"ckb-next beta-v0.2.7"_)

__Branch__: (replace these parentheses with a branch in case ckb-next was compiled manually, otherwise pass, for example: _"master"_ or _"testing"_)

__Logs & crash reports__: (replace these parentheses with the uploads of any useful information regarding the bug, these could be an OS crash report, debugger info, a journal log etc.)

### Problem description

#### What is the current behaviour of the programme?
...
#### What is the expected behaviour of the programme?
...
#### Any other notes & comments?
...

---

Now what are _your_ observations on the above? Propose your changes and/or directly edit this message.

light2yellow, 2017-03-11:
Rationale behind creation of a template can be found in #75.

fleischie, 2017-03-13:
Although I feel this issue template is thorough, it also might be too extensive for the larger part of our audience. I fear that most people will remove all the text and still enter the most vague description of the problem.

I wonder, whether we can engage the more casual users with having checkboxes for the prerequisites. Perhaps with a disclaimer of retain the right to close issues uncommented, if we find the issue not suitable enough.

Also I would suggest we use international english (i.e. "behavior of the program" instead of "behaviour of the programme").

light2yellow, 2017-03-13:
@fleischie rewritten.

light2yellow, 2017-03-14:
Some of these ideas [from here](https://dev.to/danmademe/github-issues-helping-without-committing) may be incorporated into the above, one day.

frickler24, 2017-04-13:
The idea is ok IMHO and will save our time. Two questions: 
1. How is this template working - do I get it as form when I click on "new issue"? 
2. And is it possible to have a short menu to select different forms? If yes, I think we could have two form: One for feature request, one for bug-report/Question/other-issue.
Anyway: Let's start with it.

light2yellow, 2017-04-30:
@frickler24 
It works [like that](https://github.com/blog/2111-issue-and-pull-request-templates), e. g. when user creates a new issue he gets it pre-filled with this template, that's all. Pretty straightforward, even too much. I'd be glad to see a more interactive process like choosing the template when creating an issue, but there's no such feature right now, so there's that.

light2yellow, 2017-05-02:
The template above is obsolete now, track the changes on #133.

PR #133 (2017-06-29): Issue template
* Shorten developers' time spent on the issue
* Ease issue sorting and cathegorizing
* Make users do their homework
* Tell users precisely what developers want

Closes #110
See also: #75
commit 7a40d7b3 (2017-06-29): Add an issue template

* Shorten developers' time spent on the issue
* Ease issue sorting and cathegorizing
* Make users make their homework
* Tell users precisely what developers want

Closes #110
See also: #75
closed 2017-06-29 by light2yellow