mattanger/ckb-next #113: Disappearing Profiles

hifighetto, 2017-03-17:
Quick specs: 
OSX 10.12.3 
k65-RGB
ckb-next beta-v0.27

If my mac crashes, or if i change the USB port(s) it is in, it will lose all the profiles I have created. I thought it might be a USB issue, so i tried another port and it continued to happen. 
I do not dual boot my computer, so it is on the one OS. 
This last time I was saving the profiles to the hardware in hopes that it wouldn't happen again.

Is there a way to manually backup the settings/preferences so that if they are overwritten I can manually restore them? 

Thanks

light2yellow, 2017-03-22:
Please try the `testing` branch and tell us if the issue persists.

samrocks9, 2017-03-29:
Hey, I was also having this issue but instead of profiles disappearing, modes do.

light2yellow, 2017-03-29:
Try #90. I thought it got merged, but occasionally it haven't yet.

samrocks9, 2017-03-29:
occasionally?

light2yellow, 2017-03-29:
It should contain "eventually" instead, and also "hasn't been", sorry.

tatokis, 2017-04-25:
Closing as duplicate of #83.

closed 2017-04-25 by tatokis