mattanger/ckb-next #114: ckb is already running, except it's not

AbsyntheSyne, 2017-03-21:
Just installed ckb-next on a clean Debian Stretch install. Downloaded the listed dependencies, got the "ckb is already running" thing. It's not running, but I ran "gdb ckb" and got some possibly useful info

Temporary breakpoint 1 at 0x27480
Starting program: /usr/bin/ckb 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Temporary breakpoint 1, 0x000055555557b480 in main ()

light2yellow, 2017-03-22:
If ckb works as expected (does it?), this can be safely ignored.

AbsyntheSyne, 2017-03-22:
It had something to do with my custom kernel. I don't know what, but the userspace input driver was enabled.

light2yellow, 2017-03-25:
Well, I just "reproduced" this. It happens when you reinstall ckb without rebooting, because ckb can leave without ckb-daemon. Just do "Settings" --> "Quit" and launch ckb once again. It will be launched as usual.

closed 2017-04-10 by tatokis