mattanger/ckb-next #116: Freeze on click of colour selector

Tapemaster21, 2017-03-22:
Hello, I'm on Fedora WS 25. When I attempt to highlight 1 or more keys and open the colour selector button below and left of the keyboard picture, the program freezes and is unresponsive until I force quit it. 

I ran it from terminal to catch problems. Hopefully they help. 

`[user@computer ~]$ ckb

(ckb:5167): GLib-GObject-WARNING **: g_object_get_valist: object class 'GtkSettings' has no property named 'gtk-application-prefer-dark-theme'

(ckb:5167): GLib-GObject-WARNING **: g_object_get_valist: object class 'GtkSettings' has no property named 'gtk-long-press-time'
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile

(ckb:5167): GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'

(ckb:5167): GLib-GObject-WARNING **: cannot add class private field to invalid type '<invalid>'

(ckb:5167): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(ckb:5167): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:5167): GLib-GObject-WARNING **: cannot register existing type 'GtkBuildable'

(ckb:5167): GLib-GObject-CRITICAL **: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(ckb:5167): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed

(ckb:5167): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:5167): GLib-GObject-CRITICAL **: g_type_register_static: assertion 'parent_type > 0' failed

(ckb:5167): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
`

tatokis, 2017-03-22:
Your Qt has a known bug, it has nothing to do with us. https://github.com/mattanger/ckb-next#linux

light2yellow, 2017-03-22:
F25 ships with 5.7.1, #70, #51, ccMSC/ckb#500, ccMSC/ckb#461

Tapemaster21, 2017-03-23:
Ah, sorry I'm stupid and didn't read. 

I downloaded the qt5ct and it seems to allow me to change colors when I launch through terminal but it still is broken when launching from the super menu. Is there a way I can go about fixing that?

light2yellow, 2017-03-23:
You can try different combinations from an Arch Linux thread from #70 nebo #51. Pesky Qt!

light2yellow, 2017-03-25:
Anyway, merged into all the above issues as a duplicate. If you have tried all possible solutions from all the issues (including a long Arch Linux thread), please open a new one specifying what exactly was tried and didn't work out.

closed 2017-03-25 by light2yellow