mattanger/ckb-next #117: The animation Is working, but it stays the same color (red).

FkBoyLeader, 2017-03-24:
I change rainbow to breathing/trippy but it doesn't change color it's only showing me the color red on the keyboard. The animation is working, it's just not displaying a different color other than red, and yes my keyboard has RGB. Any ideas? Thanks in advance

JarrenJ, 2017-03-25:
Have you had the rainbow or any other animation/color working? Or has it always been a static color such as red?

FkBoyLeader, 2017-03-25:
Yes I used to be able to customize it, had it rainbow and different colors. These keyboards are such a headache, I knew I should've gone with a Razer keyboard, their Sypnase software is second to none. Going try to install a windows partition on my Mac and see if it works there. Thanks

JarrenJ, 2017-03-25:
I agree about the Razor Synapse, I use it for my FireFly mousepad. But, I would love to try and get Ckb-Next working for you. What specific keyboard are you using; and what branch did you build it from? 

PS-- at it took me long to respond. I just got off my flight back home.

light2yellow, 2017-03-25:
That's a strange issue, did you try reinstalling?

tatokis, 2017-03-25:
Check the blend mode on your animations.

tatokis, 2017-04-13:
Closing due to inactivity.

closed 2017-04-13 by tatokis