mattanger/ckb-next #119: Script could not be loaded, please check your animiation directory

gregf, 2017-03-25:
With both ckb-next and ckb-next-latest in aur I get this error now on all animations.

light2yellow, 2017-03-25:
It means that they're missing or corrupted.

If you have some files under `/opt/ckb/ckb-animations`:

* provide `ls -l /opt/ckb/ckb-animations`
* remove the package with `pacman -Rns` and see if they are still there
* if they're still there, it's bad, if they have gone, reinstall the package from AUR and test again

light2yellow, 2017-03-25:
Forget about the above. I just pushed some updates for `ckb-next-latest-git`, please install/update it. It should all work now.

light2yellow, 2017-03-25:
Also, run ckb from the terminal to make sure it greps all the deps. It is working for me.

gregf, 2017-03-25:
Working now, thank you.

closed 2017-03-25 by gregf