mattanger/ckb-next #121: Macro not recording propely.

AleKrabbe, 2017-03-26:
I have a Corsair Scimitar, I was trying to associate the 7 key of the mouse numpad with the action 'alt + 7', however when I hit record nothing that I type on the keyboard is recognized only mouse buttons, also if I try to hit the stop button nothing happens, it only stops if I exit out of ckb-next or click insanely fast multiple times on the stop until it finally stop recording. How would I got about associating commands like "MODIFIER + KEYPRESS" to any of the mouse buttons? What I'm really trying to achieve here is to be able to rate songs playing on Rhythmbox with the mouse numpad, so I have a custom keyboard shortcut setup to "alt + 7" that runs the follow action "rhythmbox-client --set-rating 5", if "alt + 7" is pressed it works, so I would like to execute "alt + 7" with the mouse button.

LivingWithHippos, 2017-03-26:
Yeah I can't record them properly too. But you can do it manually: go to macro and instead of pressing start to record you can just write your macro in the "Macro key Actions" windows on its left. Write them like this: +key,+another key, ... For example Cut, Copy and Paste are +lctrl,+x +lctrl,+c +lctrl,+v
Remember to press apply after every macro you write or they won't get memorized.

AleKrabbe, 2017-03-26:
Thanks for the reply, +pad7 doesn't seem to work. I tried only setting it to +pad7 to see if the number 7 would be typed and it isn't. How would you go about discovering the names of the keys that macro recognizes?

AleKrabbe, 2017-03-26:
I decided to keep this issue opened since macro still not recording properly.

frickler24, 2017-04-01:
@AleKrabbe As I understand your first post, you click "start" and none of the keys you type afterwards are recorded? Or just some special keys like them at the numpad?
And when you click on start, ist there any changing in the gui (just stop, clear and cancel may be active then). And when clicking on stop, are the other buttons reactivated?

LivingWithHippos, 2017-04-01:
Yeah it happens to me too. No keys/mouse clicks are recorded (of any kind). When I click start the button get deactivated and the others (stop, clear and cancel) become active. Clicking on stop does nothing on the gui, since it doesn't record anything I can't know if it stops recording (but I'd think not), I have to close it from the task manager.

frickler24, 2017-04-02:
@LivingWithHippos you also have a Scimitar?

LivingWithHippos, 2017-04-02:
Yeeeah it's an awesome mouse, especially now that I can use macro on Linux

frickler24, 2017-04-07:
@LivingWithHippos or @AleKrabbe if you start ckb manually, there should be some debug output when clicking "start" or "stop". It should look like the following (with different pathes):
```
**When clicking start:**
MacroReader::run() started with 2 and "/dev/input/ckb1/notify2" and QPlainTextEdit(0x16de1a0, name="pteMacroBox") and QPlainTextEdit(0x16ce9f0, name="pteMacroText")

**And when clicking Stop:**
MacroReader::run() ends.
```
Do you get them?

LivingWithHippos, 2017-04-08:
`MacroReader::run() started with 2 and "/dev/input/ckb1/notify2" and QPlainTextEdit(0x1146e10, name = "pteMacroBox") and QPlainTextEdit(0x114a180, name = "pteMacroText")   
`

No the stop or clear event doesn't get triggered. I think that when recording the click event is captured and consumed without letting it reach the listener for the click on the button. I mean in Java at the end of a listener method you could pass true if the event was considered "consumed" and so it wasn't passed down to other methods, or "false" and so it was passed down the chain. Maybe it's the same thing here?

frickler24, 2017-04-09:
@LivingWithHippos Thank you for your analysis!
You are completely right with Edit 2: The main problem is, that my code does not recognize if it is handling a mouse or a keyboard (it was originally designed for KB only). So clicking somewhere when changing mouse-Keys is recorded as normal input. With my M65, after starting the recording and clicking or using the wheel, all actions are shown in the macro Key Actions panel:
[image]
So first I'll have a look at your first hint: If the priority lays on the panel first and then on my process, clicking on "stop", "clear" or "cancel" would terminate the process instead of sending the click event to the listener.
The fastest way is what you mentioned: commenting out line 31. But if we do so, we have to set the focus into the Macro Text pane directly after clicking "start" (or we must click into the pane as the first manual action), otherwise the textual presentation of what you type in the macro is not shown anymore. So I moved the setFocus() into the MacroReaderThread::run(). With this, all characters are shown unless you change the focus manually.

What I do not understand is: Why don't you get the click events into your notify2-pipe?
What happens if you click "start", then do a cat /dev/input/ckb1/notify2 and click around?

For me, it is understandable that a mouse macro does not accept data from the keyboard. This can not be done yet, because it opens with the second notify track only the device for which the macro is recorded (so you do not see clicks when recording a KB-macro and no keys when recording a mouse-macro). For this I see two possible solutions:

- We extend the implementation so that all channels are opened for all connected devices
- or we work with a workaround: First record the macro as a keyboard macro and copy the contents of the macro key text panes and create the mouse macro manually, as described in your first post.

I sent a PR #140 as a first fix.

LivingWithHippos, 2017-04-09:
@frickler24 cat on that file correctly report what I'm pressing, my bad, it also appear on the macroBox, I don't know why I said it didn't at first, maybe I was trying just the keyboard since all of my macros don't need the mouse. I'd be happy If I could record a keyboard only macro for my mouse

frickler24, 2017-04-10:
@LivingWithHippos @AleKrabbe and others:
When I started the implementation of the new behavior, the following question came up: Do we really want to record mouse clicks? Since we do not record the movements of the mouse relatively or absolutely, it is quite senseless - right?
I will first change the implementation so that always the keystrokes are used in macro recording. Should there be a need for mouse-click-recording, please give me a hint.

LivingWithHippos, 2017-04-10:
@frickler24 yeah I'm all for keyboard-only macro

frickler24, 2017-04-10:
Anyone interested in testing? Please have a look at #147 and give me feedback please.
It is e2f154c in [my repo](https://github.com/frickler24/ckb-next/tree/issue121-Macro-not-recording-properly).

Anonrate, 2017-09-16:
If you must record the macro, pressing <kbd>ALT + O</kbd> is the accelerator key (Underline Shortcut) for the Stop button.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/10

closed 2018-01-22 by tatokis

PiFlUn, 2018-12-29:
Is there a list of variable names for the (special) keys? Like left alt := +lalt.