mattanger/ckb-next #124: GUI Gets Cut Off

samrocks9, 2017-03-27:
Pretty self-explanatory. I am using macOS Sierra and the Strafe RGB. https://i.imgur.com/ZxrbsDh.jpg

light2yellow, 2017-03-27:
Yep, this happens on Linux as well.

[image]

tatokis, 2017-03-31:
Looks like a Qt scaling issue, not sure what can be done about it.
[image]

frickler24, 2017-04-01:
@samrocks9 and @light2yellow Do you have scaling factors active? On my mint with 100% scale factor it looks fine as it does at @tatokis screenshot

tatokis, 2017-04-01:
@frickler24  I am positive @light2yellow has scaling enabled.

light2yellow, 2017-04-01:
@frickler24, @tatokis yes, you are right. I use 4k monitor and an NVIDIA video card, so I installed a proprietary NVIDIA driver and now `xdpyinfo` reports 184x182 dpi. I also use KDE Plasma and SDDM, they both have out-of-box HiDPI support.

Looking at other Qt applications I use ( Viber, Telegram), they "solved" this by providing an interface size slider. 

[image]

Either it was done at the period of time when Qt's HiDPI support was poor, or it's still poor (at least by default), but why then Plasma and SDDM, both heavily relying on Qt, have such a great HiDPI support?

samrocks9, 2017-04-01:
I don't think I have any scaling on. I'm just using an iMac without a second monitor.

frickler24, 2017-04-02:
hmmm, seems that we have two different bugs in the ui definition:
@light2yellow you can see, that the keyboard-buttons are scaled down.
But in the picture from @samrocks9 we have just the buttons moved downwards.
When I added the macro tab, I had several problems bringing all UI elements in the tab. Resizing is ignored by qt.
Is there anybody with high experience in qt UI definitons?

tatokis, 2017-04-02:
@frickler24 , @light2yellow has pointed this out. 
https://github.com/telegramdesktop/tdesktop/blob/dev/Telegram/SourceFiles/application.cpp#L634

Perharps we can do something similar?

light2yellow, 2017-04-02:
@frickler24 well, I experience both :)
If I switch to the Macro tab (which I should've done before but instead made a screenshot of the Program), then it all becomes a total mess with unreachable buttons below.

[image]

Regarding Telegram, the way they implement macros from the link posted above is unclear to me, they play with the preprocessor [here](https://github.com/telegramdesktop/tdesktop/blob/master/Telegram/SourceFiles/settings.h#L40) to create c* function calls.

Anyway, it looks more like there's a hardcoded height somewhere, where it shouldn't be.

tatokis, 2017-04-02:
@light2yellow This should be it https://github.com/mattanger/ckb-next/blob/master/src/ckb/kbbindwidget.ui#L206-L210

tatokis, 2017-04-02:
also, vsize is fixed. https://github.com/mattanger/ckb-next/blob/master/src/ckb/kbbindwidget.ui#L195

mst3kharris, 2017-04-08:
Hello, I've just installed this driver for my Corsair M65 mouse.  It appears to be working beautifully; however, the GUI for the settings is too tall for my screen.  I'm running an old-ish Dell laptop and the screen resolution is fixed at 1366x768.  I'm able to change the size of the GUI horizontally, but not vertically, which is to be expected if vsize has a fixed value.  Attached is a screenshot of my workspace, where you can see the bottom of the window cuts off the buttons for saving settings, applying them, et cetera.
[image]

light2yellow, 2017-04-08:
@mst3kharris thanks for reporting back, I'll try to do something else to it.

light2yellow, 2017-04-11:
@mst3kharris did you actually try the PR mentioned above and get such results or are you posting the issue about current GUI in `master` or `testing` just because this thread mentions GUI?

light2yellow, 2017-05-01:
Regardless, I've just tested our `master` on 1366x768 and it indeed doesn't even fit into the screen. I'll take a look at it.

light2yellow, 2017-05-01:
Sorry, I'm not able to fix this. It requires a redesign and I neither know all the widgets and frames, nor I know how to write GUI applications in Qt.

You can find my pathetic attempts under `dehardcoded-gui` branch, but I recommend starting from scratch.

samrocks9, 2017-12-10:
I have reinstalled ckb after months of going without it, but am still having this issue. It is preventing me from making macros.

ZirconiumX, 2017-12-11:
Short of rewriting the GUI, no, not really.

samrocks9, 2017-12-11:
Hmm. It seems like an important thing to be fixed, considering it cuts off
crucial parts of the program. I have no way of setting macros and stuff.
Why can't the window just be scrolled?

ZirconiumX, 2017-12-12:
The GUI was written by somebody who knew Qt well enough to create a very functional interface with it. But nobody on the current development team knows much Qt at all, so we don't know what to do.

light2yellow, 2017-12-12:
There's a PR that provides scrolling, but its scrolling is deficient because it adds additional widget or something. Then, @tatokis did some job and implemented a proper scroll, but never published it. I request @tatokis's response for comment on their work (I am satisfied with it).

fdagpigj, 2017-12-24:
If nobody knows Qt well enough to fix the GUI, why not add a command-line interface?

fleischie, 2017-12-24:
@fdagpigj we actually had at some point, maybe even do.... You can eve  try to talk to the server directly and store the settings in a local file or something.

Anyway this is not really the point of this issue. So thanks for your concern and happy holidays. 🎉

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/11

closed 2018-01-22 by tatokis