mattanger/ckb-next #129: Strafe RGB Not being detected

embluk, 2017-03-29:
I have install cbk-next using the build.md instructions but when running cbk it does not detect my strafe RGB keyboard, I am running Elementary OS Loki.

I tried unpluggin my keyboard and plugging it in again but still no luck. If I build the application in QT it outputs: /home/user/Downloads/ckb-next-master/bin/ckb-animations/ckb-mviz does not exist.

However the app does run if I just run cbk fron the bin folder.

light2yellow, 2017-03-29:
You may have forgotten to start/enable the daemon service.

What prevented you from using the generic way from the main readme?

embluk, 2017-03-29:
I have got it to work the generic way now, it was due to some dependency issues. However, my problem still persists. cbk builds and is created within the bin directory, my guess is that I just click/ run the ckb-daemon and then run the ckb app right after? Because that is what I've been doing and still no strafe rgb keyboard detected.

light2yellow, 2017-03-30:
The `quickinstall` script asks you whether you want to do XYZ, which should be answered `Y` if you want this to "just work". After that the daemon starts automatically as well as the GUI.

Please, provide a thorough description of what you've done and in what order for me to help you. Strafe RGB works well.

embluk, 2017-03-31:
Going to start from scratch again and provide step by step instructions in what I am doing...

1. Download the latest master branch

2. Unpack master branch

3. Install the dependencies: `sudo apt-get install build-essential libudev-dev qt5-default zlib1g-dev libappindicator-dev `

4. Make sure `quickinstall` has the correct permissions.

5. Double click quickinstall, everything builds and the bin directory is created but no GUI pop up asking me what I want to do, nothing happens after that.

6. Try running ckb from the bin directory with the daemon first and then the ckb app. The app opens after running it from the bin directory but no keyboard detected.

That is the process I follow, really not sure why it won't work because I remember using the old ckb github on Ubuntu and it worked for me on the same keyboard. The Strafe RGB I have does have the latest firmware as I installed that from Windows the other day. Is it something to do with me using Elementary?

tatokis, 2017-03-31:
Instead of double clicking quickinstall, try running it from a terminal like so:
```./quickinstall```
It if all goes well, you should see something like this:
[image]
Answer Y to all the questions, and it should take care of everything.

If you see something else, please paste the output here.

embluk, 2017-03-31:
There we go, problem solved.

@tatokis I just followed your post and everything now works correctly, Thanks!

I guess it was just because I did not run the quickinstall using ./quickinstall

Thanks!

closed 2017-04-03 by light2yellow