mattanger/ckb-next #130: Looking for a fairly stable branch with fixes

samrocks9, 2017-04-01:
Alright, I am on macOS Sierra using the Strafe RGB. I am currently using the master branch of ckb-next. I was wondering if anyone could recommend a fairly stable branch which is up to date and fixes the [Cmd + click issue](https://github.com/mattanger/ckb-next/issues/29) on Sierra and the issue with [profiles/modes disappearing](https://github.com/mattanger/ckb-next/issues/113). If anyone could spare me some of their time, I would really appreciate an explanation about how I would go about moving over to a different branch. I am sure others would also find a step by step explanation useful. Thanks a ton!

light2yellow, 2017-04-01:
https://github.com/mattanger/ckb-next/issues/29#issuecomment-289412693

samrocks9, 2017-04-01:
Yeah I saw that comment but was unsure how to proceed.

light2yellow, 2017-05-03:
As on how to proceed: every issue has a "Subscribe" button. You'll receive the notifications in the chosen threads. Subscribe to them and you won't miss the moment when the issues are updated. You can also "Watch" the project to receive all notifications.

As you might see, right now none of the issues are resolved - one does not survive a logout/reboot and the other one is under investigation and yet to be reproduced, although contains a workaround by another user.

As on how to install from a different branch: just the same way as with any other branch - README for the win. If you want to incorporate multiple fixes from different branches, you'll have to learn git basics.

closed 2017-05-03 by light2yellow

samrocks9, 2017-05-05:
Hey, I read through #29 again, uninstalled CKB, rebooted, installed the testing branch, and cmd + click was working. Then I rebooted and cmd + click was not working anymore. Is there something I need to do to fix this issue? From what I understand, it is an issue with CKB loading the wrong script on reboot?