mattanger/ckb-next #134: [OS X] quickinstall script doesnt include ckb-daemon?

syropian, 2017-04-03:
After cloning master into my home folder, I simply cd'd in and ran the quickinstall script. It built the app, and I booted it up but I noticed it wasn't detecting my keyboard. I checked Activity Monitor and noticed `ckb.app` was running but not `ckb-daemon`, which explained my keyboard not being detected. I figured I could just manually start the daemon but a peek inside `ckb.app/Contents/Resources` revealed that the daemon wasn't even present. Is this where it's supposed to be, or does it live somewhere else?  If there's any logs or files I can include to help let me know.

light2yellow, 2017-04-03:
Yes, please. Send us everything that was produced by `./quickinstall` (including your decisions (Y/n)). Just copy-paste from the terminal.

syropian, 2017-04-03:
the quickinstall script never prompted me for any y/n answers. It spit out a ton of warnings, mostly about unused parameters and such. However this one stuck out:
```
1 warning generated.
Undefined symbols for architecture x86_64:
  "_libiconv", referenced from:
      _u16enc in profile.o
      _u16dec in profile.o
  "_libiconv_open", referenced from:
      _u16enc in profile.o
      _u16dec in profile.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [../../ckb.app/Contents/Resources/ckb-daemon] Error 1
make[1]: *** [debug] Error 2
make: *** [sub-src-ckb-daemon-make_first] Error 2
make: *** Waiting for unfinished jobs....
```

If you really want me to paste the entire output i can but its about a mile long.

light2yellow, 2017-04-03:
Do `xcode-select --install` and try again.

syropian, 2017-04-03:
@light2yellow I had the cli tools installed already, but I wasn't fully up to date. Updated them, did a restart and ran quickinstall again but unfortunately same error.

light2yellow, 2017-04-03:
Alright. Then do `./quickinstall > out.txt` and upload the text file.

light2yellow, 2017-04-03:
Also, do you have XCode installed?

syropian, 2017-04-03:
I do have XCode installed but its also not fully up to date. Let me get that all updated and I'll try again. If that fails I'll upload the output text.

syropian, 2017-04-03:
@light2yellow Hm, still no dice. I updated XCode and also updated my OS to 10.12.4. Here's the output - [out.txt](https://github.com/mattanger/ckb-next/files/891430/out.txt)

syropian, 2017-04-03:
If I were to guess whatever is trying to use libiconv likely just has the wrong path. `which iconv` points me to `/usr/bin`. I'm not sure how to check where the installer is specifically looking though.

light2yellow, 2017-04-03:
@syropian I think I know what's the problem, working on it.

light2yellow, 2017-04-03:
Please, try compiling from `testing` now. I pushed a commit, so you'll need to `pull` or redownload a single branch (like here https://github.com/mattanger/ckb-next/issues/29#issuecomment-289412693, but make sure you're not inside a git repo if you decide to download).

This should fix all issues.

syropian, 2017-04-03:
Did you push to `testing`? Github says last commit was Jan 22nd

light2yellow, 2017-04-03:
Yes, exactly :)

syropian, 2017-04-03:
Oh! Thought you meant you _just_ pushed, my bad! 😅

syropian, 2017-04-03:
That seems to work, thanks! Will those changes make to master any time soon?

light2yellow, 2017-04-03:
Yes, I'll merge them right now. Just wanted to be sure nothing will break.

light2yellow, 2017-04-03:
Done. Now you can `master` too.

This was an old issue that was fixed long time ago but didn't make into any of the branches because I was the only one experiencing these problems.

closed 2017-04-03 by light2yellow