mattanger/ckb-next #135: Corsair Gaming CH-9303011-EU Sabre - No Longer Detected

ollieatkinson, 2017-04-04:
I have installed from source (master) and ckb can no longer detect my mouse! 

As far as I can tell the device *should* be supported and I can see it listed in `usb.h` - but for some reason it is not recognising it. 

I am unsure what exact commit or version my old install had, so unfortunately cannot roll back or compare the two. Any help is appreciated.

Source: https://github.com/mattanger/ckb-next/commit/4b105ee78419bffee925807ffd1a1523b45c1850

Version: ckb-next beta-v0.2.7
Mouse: Sabre 0x1b2f
macOS: Sierra 10.12.4

Here is the output from `ckb-dev-detect`

```
ckb-dev-detect - Tue Apr  4 13:11:41 BST 2017

Kernel: Darwin Falcon9 16.5.0 Darwin Kernel Version 16.5.0: Fri Mar  3 16:52:33 PST 2017; root:xnu-3789.51.2~3/RELEASE_X86_64 x86_64

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.12.4
BuildVersion:	16E195

USB Devices:
                Corsair Gaming Sabre RGB Mouse:

                  Product ID: 0x1b2f
                  Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
                  Version: 2.03
                  Serial Number: 0D00B015AEBD1840562C571FF5001946
                  Speed: Up to 12 Mb/sec
                  Manufacturer: Corsair
                  Location ID: 0x14232000 / 12
                  Current Available (mA): 500
                  Current Required (mA): 300
                  Extra Operating Current (mA): 0

```

tatokis, 2017-04-04:
Try the testing branch
https://github.com/mattanger/ckb-next/blob/testing/src/ckb-daemon/usb.h#L59

ollieatkinson, 2017-04-04:
Thanks for the quick response @tatokis .

I can verify that the `testing` branch has the same issue, even after a restart.

ollieatkinson, 2017-04-04:
Seems its solved by re-installing the software and ensuring launchd is enabled.

closed 2017-04-04 by ollieatkinson