mattanger/ckb-next #138: K70 RGB Rapidfire

Sycocylin, 2017-04-08:
Hi, I need help. My keyboard was working fine after a restart but I switched back to windows for a bit and now my keyboard is unrecognized and no imputs work exept for in the BIOS.

tatokis, 2017-04-09:
OS? Does ckb say anything? Did you upgrade the firmware while on windows?

Xcrysdale, 2017-04-27:
Hey sorry to hijack this thread but I've been experiencing the same issue using the Corsair Strafe Mechanical Keyboard, whenever I restart Ubuntu it is no longer detected and I need to reinstall the driver and restart in order to fix the issue. It could possibly be something to do with my Ubuntu install but if you have any insight please help me haha.  To clarify I've also tried switching the ports, re running the software, and replugging back in to no avail.

tatokis, 2017-04-28:
Does it say not detected or is the daemon not running?
try `sudo service ckb-daemon restart`
and if it throws an error, send back the output of `sudo service ckb-daemon status`

Xcrysdale, 2017-04-29:
Alright I ran the sudo service check and it's running for sure. I'm coming to the realization that I didn't correctly configure my dual boot and having both Windows 10 and Ubuntu running off the same drive means I'm unable to access the OS hard drive because Windows Hibernates it after Logging out. To me it seems like the issue is the software tries to Install itself but can't access the directories (neither can I) so that would explain why it occasionally works after restarting. I'll be looking into how to stop Windows from locking me out of my drive but if you had any insight or next step that would also be greatly appreciated. Thanks! 

Edit: I'm an idiot I just need to completely shut down Windows and not quick restart it :P 

Logs If You're Interested:

   Loaded: loaded (/usr/lib/systemd/system/ckb-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2017-04-29 12:00:46 PDT; 30s ago
 Main PID: 4208 (ckb-daemon)
   CGroup: /system.slice/ckb-daemon.service
           └─4208 /usr/bin/ckb-daemon

Apr 29 12:00:46 xcrysdale-MS-7917 systemd[1]: Started Corsair Keyboard Daemon.
Apr 29 12:00:46 xcrysdale-MS-7917 ckb-daemon[4208]:     ckb: Corsair RGB driver beta-v0.2.6
Apr 29 12:00:46 xcrysdale-MS-7917 ckb-daemon[4208]: [I] Root controller ready at /dev/input/ckb0
Apr 29 12:00:46 xcrysdale-MS-7917 ckb-daemon[4208]: [I] Connecting  at /dev/input/ckb1
Apr 29 12:00:46 xcrysdale-MS-7917 ckb-daemon[4208]: [W] os_setupusb (usb_linux.c:293): Unable to read endpoint count from udev, assuming 4...
Apr 29 12:00:46 xcrysdale-MS-7917 ckb-daemon[4208]: [E] os_setupusb (usb_linux.c:296): Failed to claim interfaces: No such file or directory
Apr 29 12:00:46 xcrysdale-MS-7917 ckb-daemon[4208]: [I] Disconnecting /dev/input/ckb1
Apr 29 12:00:46 xcrysdale-MS-7917 ckb-daemon[4208]: [I] Removed device path /dev/input/ckb1
~

frickler24, 2017-04-30:
@Xcrysdale This does not look good:
```
os_setupusb (usb_linux.c:293): Unable to read endpoint count from udev, assuming 4...
Apr 29 12:00:46 xcrysdale-MS-7917 ckb-daemon[4208]: [E] os_setupusb (usb_linux.c:296): Failed to claim interfaces: No such file or directory
```
The message means that there is a problem to identify the amount of endpoints available from your device.

Could you please post the content of the following files:
`/dev/input/ckb0/` all files except pipes if exist
`/dev/input/ckb1/` all files except cmd (also a pipe)
`journalctrl | grep ckb`

tatokis, 2017-04-30:
It makes sense that it couldn't read any endpoints, because the actual 
path it tried to access does not exist.
This can be seen later on at
Apr 29 12:00:46 xcrysdale-MS-7917 ckb-daemon[4208]: [E] os_setupusb 
(usb_linux.c:296): Failed to claim interfaces: No such file or directory

frickler24, 2017-04-30:
@tatokis The perror message means, that the ioctl could not find the endpoint:
```
        if(ioctl(kb->handle - 1, USBDEVFS_CLAIMINTERFACE, &i)) {
            ckb_err("Failed to claim interface %d: %s\n", i, strerror(errno));
```
@Xcrysdale So we may also need the output of ckb-dev-detect

tatokis, 2017-04-30:
Remind me again, why do we assume endpoints?
If we assume them and they don't really exist, then there's no way we 
can do anything with them later on.

frickler24, 2017-04-30:
That's a really good question. The original code greps the number of endoints via a library function which returns a string with the number of EPs. The function in ckb-daemon assumed always a valid number (2, 3 or 4), but one of our incidents has shown " " as retval.
So I implemented as first reaction for some cases the handling of 0 EPs to assume 3 or 4, based on the information what device it is:
```
547 const char* ep_str = udev_device_get_sysattr_value(dev, "bNumInterfaces");
548 kb->epcount = 0;
549 if(ep_str)
550 sscanf(ep_str, "%d", &kb->epcount);
551 if(kb->epcount == 0){
552 // This shouldn’t happen, but if it does, assume EP count based on what the device is supposed to
have
553 kb->epcount = (HAS_FEATURES(kb, FEAT_RGB) ? 4 : 3);
554 ckb_warn("Unable to read endpoint count from udev, assuming %d...nn", kb->
epcount);
555 }
556 if(usbclaim(kb)){
557 ckb_err("Failed to claim interfaces: %snn", strerror(errno));
558 return -1;
559 }
```

Xcrysdale, 2017-04-30:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/966994/ckb-dev-detect-report.gz)

Here's the file you said you may need, one issue I had been experiencing which could have been effecting the install as well would be the fact that windows Hibernated the OS drive when I restarted instead of hard-shutdown but I've resolved that, and the Ckb is still spotty in detecting my Keyboard.

Xcrysdale, 2017-04-30:
It looks like from the report the Keyboard doesn't exist, but I've had it working before so I'm a bit confused at the moment. Next time I have it working I'll run the reporter and upload the file but for the moment it won't budge.

Xcrysdale, 2017-04-30:
The folder at /dev/input/ckb0 contains the following files:

Connected.txt (Blank text file)
pid.txt (contains the pid of "3784")
version.txt (contains "beta-v0.2.7")

Unfortunately I don't have a ckb1 folder. 

Also turns out its journalctl | grep ckb not journalctrl | grep ckb haha had me confuse for a second there.

Outputs of running "journalctl | grep ckb"

```
Apr 30 13:35:35 xcrysdale-MS-7917 ckb-daemon[827]:     ckb: Corsair RGB driver beta-v0.2.7
Apr 30 13:35:35 xcrysdale-MS-7917 ckb-daemon[827]: [I] Root controller ready at /dev/input/ckb0
Apr 30 13:35:38 xcrysdale-MS-7917 gnome-session[1407]: Scanning  "/usr/bin/ckb-animations/ckb-gradient"
Apr 30 13:35:38 xcrysdale-MS-7917 gnome-session[1407]: Scanning  "/usr/bin/ckb-animations/ckb-heat"
Apr 30 13:35:38 xcrysdale-MS-7917 gnome-session[1407]: Scanning  "/usr/bin/ckb-animations/ckb-mviz"
Apr 30 13:35:38 xcrysdale-MS-7917 gnome-session[1407]: Scanning  "/usr/bin/ckb-animations/ckb-pinwheel"
Apr 30 13:35:38 xcrysdale-MS-7917 gnome-session[1407]: Scanning  "/usr/bin/ckb-animations/ckb-rain"
Apr 30 13:35:38 xcrysdale-MS-7917 gnome-session[1407]: Scanning  "/usr/bin/ckb-animations/ckb-random"
Apr 30 13:35:38 xcrysdale-MS-7917 gnome-session[1407]: Scanning  "/usr/bin/ckb-animations/ckb-ripple"
Apr 30 13:35:38 xcrysdale-MS-7917 gnome-session[1407]: Scanning  "/usr/bin/ckb-animations/ckb-wave"
Apr 30 13:36:11 xcrysdale-MS-7917 ckb-daemon[827]: [I] Connecting  at /dev/input/ckb1
Apr 30 13:36:11 xcrysdale-MS-7917 ckb-daemon[827]: [W] os_setupusb (usb_linux.c:293): Unable to read endpoint count from udev, assuming 4...
Apr 30 13:36:11 xcrysdale-MS-7917 ckb-daemon[827]: [E] os_setupusb (usb_linux.c:296): Failed to claim interfaces: No such file or directory
Apr 30 13:36:11 xcrysdale-MS-7917 ckb-daemon[827]: [I] Disconnecting /dev/input/ckb1
Apr 30 13:36:11 xcrysdale-MS-7917 ckb-daemon[827]: [I] Removed device path /dev/input/ckb1
Apr 30 13:37:30 xcrysdale-MS-7917 sudo[3664]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/bin/echo
Apr 30 13:37:30 xcrysdale-MS-7917 sudo[3672]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/bin/mkdir -p /usr/bin
Apr 30 13:37:30 xcrysdale-MS-7917 sudo[3674]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/usr/bin/install bin/ckb /usr/bin/ckb
Apr 30 13:37:30 xcrysdale-MS-7917 sudo[3676]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/usr/bin/install bin/ckb-daemon /usr/bin/ckb-daemon
Apr 30 13:37:30 xcrysdale-MS-7917 sudo[3678]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/bin/mkdir -p /usr/bin/ckb-animations
Apr 30 13:37:30 xcrysdale-MS-7917 sudo[3680]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/usr/bin/install bin/ckb-animations/ckb-gradient bin/ckb-animations/ckb-heat bin/ckb-animations/ckb-mviz bin/ckb-animations/ckb-pinwheel bin/ckb-animations/ckb-rain bin/ckb-animations/ckb-random bin/ckb-animations/ckb-ripple bin/ckb-animations/ckb-wave /usr/bin/ckb-animations
Apr 30 13:37:30 xcrysdale-MS-7917 sudo[3682]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/usr/bin/xdg-icon-resource install --novendor --size 512 usr/ckb.png
Apr 30 13:37:30 xcrysdale-MS-7917 sudo[3709]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/usr/bin/xdg-desktop-menu install --novendor usr/ckb.desktop
Apr 30 13:37:30 xcrysdale-MS-7917 gnome-session[1407]: (gnome-software:1720): As-WARNING **: failed to rescan: Failed to parse /usr/share/applications/ckb.desktop file: cannot process file of type application/x-desktop
Apr 30 13:37:32 xcrysdale-MS-7917 sudo[3745]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/bin/systemctl stop ckb-daemon
Apr 30 13:37:32 xcrysdale-MS-7917 ckb-daemon[827]: [I] Caught signal 15
Apr 30 13:37:32 xcrysdale-MS-7917 ckb-daemon[827]: [I] Closing root controller
Apr 30 13:37:32 xcrysdale-MS-7917 ckb-daemon[827]: [I] Removed device path /dev/input/ckb0
Apr 30 13:37:32 xcrysdale-MS-7917 sudo[3749]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/bin/mkdir -p /usr/lib/systemd/system
Apr 30 13:37:32 xcrysdale-MS-7917 sudo[3751]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/usr/bin/install -m 0644 service/systemd/ckb-daemon.service /usr/lib/systemd/system
Apr 30 13:37:32 xcrysdale-MS-7917 sudo[3753]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/bin/systemctl daemon-reload
Apr 30 13:37:32 xcrysdale-MS-7917 sudo[3767]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/bin/systemctl enable ckb-daemon
Apr 30 13:37:32 xcrysdale-MS-7917 sudo[3781]: xcrysdale : TTY=pts/17 ; PWD=/home/xcrysdale/Desktop/StrafeKeyboard/ckb-next-master ; USER=root ; COMMAND=/bin/systemctl start ckb-daemon
Apr 30 13:37:32 xcrysdale-MS-7917 ckb-daemon[3784]:     ckb: Corsair RGB driver beta-v0.2.7
[... cut ...]
Apr 30 13:37:32 xcrysdale-MS-7917 ckb-daemon[3784]: [W] os_setupusb (usb_linux.c:293): Unable to read endpoint count from udev, assuming 4...
Apr 30 13:37:32 xcrysdale-MS-7917 ckb-daemon[3784]: [E] os_setupusb (usb_linux.c:296): Failed to claim interfaces: No such file or directory
Apr 30 13:37:32 xcrysdale-MS-7917 ckb-daemon[3784]: [I] Disconnecting /dev/input/ckb1
Apr 30 13:37:32 xcrysdale-MS-7917 ckb-daemon[3784]: [I] Removed device path /dev/input/ckb1
```

Sorry for the horrid formatting I'm currently typing using the mouse :)

light2yellow, 2017-04-30:
The report talks about `1b15`, but this is Strafe non-RGB's number:
```c
#define P_STRAFE_NRGB       0x1b15
```
Err, what?

Xcrysdale, 2017-04-30:
Just got it working again, anything you want me to run?

Xcrysdale, 2017-04-30:
Some more logs with working Keyboard:
Apr 30 14:33:03 xcrysdale-MS-7917 ckb-daemon[848]:     ckb: Corsair RGB driver beta-v0.2.7
Apr 30 14:33:03 xcrysdale-MS-7917 ckb-daemon[848]: [I] Root controller ready at /dev/input/ckb0
Apr 30 14:33:03 xcrysdale-MS-7917 ckb-daemon[848]: [I] Connecting Corsair STRAFE Gaming Keyboard at /dev/input/ckb1
Apr 30 14:33:03 xcrysdale-MS-7917 ckb-daemon[848]: [I] Starting input thread for /dev/input/ckb1
Apr 30 14:33:03 xcrysdale-MS-7917 kernel: input: ckb1: Corsair STRAFE Gaming Keyboard as /devices/virtual/input/input33
Apr 30 14:33:03 xcrysdale-MS-7917 kernel: input: ckb1: Corsair STRAFE Gaming Keyboard as /devices/virtual/input/input34
Apr 30 14:33:05 xcrysdale-MS-7917 ckb-daemon[848]: [I] Setup finished for /dev/input/ckb1

light2yellow, 2017-04-30:
@Xcrysdale 
Oh, okay. So, you do have Strafe. Please, next time open another issue, the title of this thread is (was) "K70 RGB Rapidfire" and not "Strafe non-RGB".

> input33
> input34

This doesn't look healthy.
Did you try using another USB port?

@Sycocylin 
Sorry, you'll have to open a new thread in case you still experience any problems. Try providing as much information as possible - OS, journal logs, verbal description etc. Don't use BIOS mode. Try installing `testing`. Check whether `ckb-daemon` is running. Try other things from the appropriate README sections.

Xcrysdale, 2017-05-01:
Yeah I've tried a new USB port and now it's working. But I've tried this method before and it doesn't last so I'll keep you updated. And the ckb1 folder was just created if you would like the contents of that. It's still registering the keyboard as Input33 and input34, should I attempt a reboot and see if the setting reset or the keyboard fails? And sorry for hijacking the thread as I said before I intended my post to be a quick question but as of now there doesn't seem to be an end in sight.

light2yellow, 2017-05-01:
You definitely should. And after the reboot show us `systemctl status ckb-daemon`. I hope you're not using BIOS mode.

Regardless of the future pace of the discussion, I'm closing this as it's working right now (although poorly).

Xcrysdale, 2017-05-02:
Alright I know this is closed but; after reboot all setting were lost and the keyboard is not detected. Since it seems like I'm taking this in circles I'll stop taking up your time and do some more research. Apologies for all of this I'm completely new to Ubuntu and Linux as a whole so I need to take the time to look into whether or not I am using BIOS mode.

Xcrysdale, 2017-05-02:
Thanks for all your guys help, I really appreciate it :)

tatokis, 2017-05-02:
@Xcrysdale If lsusb can't see your keyboard, it's definitely not a software issue, and there's nothing we can do about it.
Sounds like your usb hub isn't providing enough power.

And no, _**never EVER**_ turn BIOS mode on with the daemon running.

Also, make sure your keyboard is running the latest firmware version.

light2yellow, 2017-05-02:
@Xcrysdale 
The keyboard is in BIOS mode if the rightmost LED in the top-right corner (it has an "arrow down" symbol) constantly blinks. This LED is near Caps Lock and Num Lock LEDs. In case it does, you will have to disable it. There's no switch on Strafe for this and I don't remember the exact key combination to hold down, but I recall that I found it on the Corsair forum by googling "Strafe RGB BIOS mode". This is nowhere to be found in the official manuals, no idea why. I suppose non-RGB Strafe is just the same.

Also, I see this line in your log:
```
Apr 29 12:00:46 xcrysdale-MS-7917 ckb-daemon[4208]: ckb: Corsair RGB driver beta-v0.2.6
```
This is a very old version of ckb, you should reinstall current `master` at the very least, but I recommend to install `testing` directly.

After you figured it out and the keyboard is definitely not in BIOS mode, let's try everything once again:

* before you reboot, make sure the daemon won't load automatically next time: `systemctl disable ckb-daemon`
* reboot
* launch `ckb-dev-detect` and upload the output
* launch the daemon manually: `sudo ckb-daemon`
* wait for a bit to allow the daemon set up the communication (30 sec - 1 min)
* post the output generated by the daemon
* in case everything goes well, you can `start` and `enable` the daemon just like you've `disable`d it above
* in case you feel like everything is not going well, save the information gathered and try another USB port

The `ckb-dev-detect` failing to detect the keyboard mentioned above means that either our script is flawed or (which is more likely) you have hardware-related problems.

tatokis, 2017-05-02:
@light2yellow on the contrary, it is unlikely that the dev detect script is at fault, since even if `lsusb -vd '1b1c:` failed to find it, then it would've shown up in dmesg and the hid-generic driver would've tried to talk to it.

Since there are no mentions of the hid driver, it is much more likely that the computer can't properly talk to the keyboard.
Instead, we get `[   32.745995] usb 3-4: New USB device found, idVendor=1b1c, idProduct=1b15` followed by nothing but silence.

Output from mine for comparison
```
[    2.704809] usb 8-1: New USB device found, idVendor=1b1c, idProduct=1b13
[    2.704816] usb 8-1: Product: Corsair K70 RGB Gaming Keyboard 
[    2.704818] usb 8-1: Manufacturer: Corsair
[    2.731486] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-1/8-1:1.0/0003:1B1C:1B13.0001/input/input3
[    2.731552] hid-generic 0003:1B1C:1B13.0001: input,hidraw0: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-1/input0
[    2.732032] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-1/8-1:1.1/0003:1B1C:1B13.0002/input/input4
[    2.732122] hid-generic 0003:1B1C:1B13.0002: input,hiddev0,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-1/input1
[    7.780364] input: ckb1: Corsair K70 RGB Gaming Keyboard as /devices/virtual/input/input18
[    7.781420] input: ckb1: Corsair K70 RGB Gaming Keyboard as /devices/virtual/input/input19
```

Xcrysdale, 2017-05-03:
Alright here's the ckb-dev-detect output file: 
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/974900/ckb-dev-detect-report.gz)

Then, I ran the `sudo ckb-daemon` manually in terminal and it gave the same error log as before, see below.

`[sudo] password for xcrysdale: 
    ckb: Corsair RGB driver beta-v0.2.7
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair STRAFE Gaming Keyboard at /dev/input/ckb1
[W] os_setupusb (usb_linux.c:293): Unable to read endpoint count from udev, assuming 4...
[E] os_setupusb (usb_linux.c:296): Failed to claim interfaces: No such file or directory
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1`

Finally I switched USB ports to no avail, still no detection.

Xcrysdale, 2017-05-03:
I did notice when poking through the Detect file that it did see my keyboard, which is an improvement but still no endpoints it seems. I had it working yesterday out of the blue, it seems completely random to me.

tatokis, 2017-05-04:
Does your keyboard actually work?
Seeing something like 
```
  iManufacturer           1 (error)
  iProduct                2 (error)
  iSerial                 3 (error)
```
could very well indicate hardware failure

tatokis, 2017-05-04:
If you do `dmesg | grep -i usb`, do you see any errors?

light2yellow, 2017-05-04:
The function in `lsusb` producing `(error)`: https://github.com/gregkh/usbutils/blob/master/usbmisc.c#L168
```c
static char *get_dev_string_ascii(libusb_device_handle *dev, size_t size,
                                  u_int8_t id)
{
	char *buf = malloc(size);
	int ret = libusb_get_string_descriptor_ascii(dev, id,
	                                             (unsigned char *) buf,
	                                             size);

	if (ret < 0) {
		free(buf);
		return strdup("(error)");
	}

	return buf;
}

```

Reading through `libusb` mailing lists that contain `libusb_get_string_descriptor_ascii() error reports` I suggest you to install `libusb` and run `xusb -d 1b1c:1b15` where these two numbers mean vendor ID and product ID (these are yours if I copied them right). Mailing lists suggest installing `libusbx`, but that was in 2012 and as far as Google tells me, `libusb` and `libusbx` are now merged into one project under `libusb`.

This is definitely not an issue with `ckb-daemon` anymore - either the hubs are problematic or the keyboard is somewhat broken, because this is the function of a very reliable library that is tested and debugged.

Regarding the output being different - I recall there was a bug in our script and @tatokis was fixing it and it must be he didn't cherry-pick the fix onto one of the branches. He can comment this better.

tatokis, 2017-05-04:
Nope, the same commit is in both branches.

light2yellow, 2017-05-04:
Ah, I should've made a diff before posting :) For both scripts and reports.
Anyway, the difference in this one line of `dmesg`
```
[   21.479252] usb 3-2: Product: Corsair STRAFE Gaming Keyboard
```
being present in one case and missing in the other doesn't let to make any conclusions. I guess if @Xcrysdale made super-clean tests, e. g. rebooting every time, in-before manually stopping the daemon / uninstalling one version before installing the other and not overriding etc. etc. we would eventually witness equalized outputs.

The main idea is that device is still half-detected by `lsusb` and in chain by `libusb`, regardless of `ckb-daemon` version being deployed.

N. B. @frickler24 was right, we should look into `libusb` one day if it's such a beast that even kernel's `usbutils` use it.

light2yellow, 2017-05-04:
And to actually execute the command above with `xusb`, you'll need to:
```sh
git clone https://github.com/libusb/libusb.git
cd libusb
./autogen.sh
make
cd examples
xusb -d 1b1c:1b15
```

tatokis, 2017-05-04:
`./xusb -d 1b1c:1b15`
also, `[   21.479252] usb 3-2: Product: Corsair STRAFE Gaming Keyboard` means that the kernel went a bit further into recognising the keyboard, but it still can't do anything with it.

frickler24, 2017-05-05:
@Xcrysdale Maybe you get some hints what is going wrong if you analyze the usb communication.
I've seen a lot of strange constellations when I've evaluated the different switching operations at the K95RGB's sampling switch (#48).
As a reference I have filed the communication of my K95RGB [here](https://frickler24.github.io/ckb-next/pres-usb-logfiles.html) with the appropriate comments.

frickler24, 2017-05-05:
@tatokis you wrote 

> Remind me again, why do we assume endpoints? If we assume them and they don't really exist, then there's no way we can do anything with them later on.

Please have a look at #89, therein is a better handling of unknown EP-connections implemented.
I'm not sure why this was not merged into testing.
@Xcrysdale like to test this one?

Xcrysdale, 2017-05-06:
Alright well I think it may be a hardware failure as you guys have been alluding to as literally the day I posted on this thread my GPU flat out died, and upon thoroughly cleaning the computer and changing the USB inputs for the keyboard it works first try. Very odd, but it works so you can mark this thread as resolved, thanks all for your help, you guys are truly god sends. Does the project take donations? If so i'd love to contribute for all the time you've spent helping me. Thanks again all of you guys.

TL;DR 
It seems like it was a hardware problem, flaky ports on my motherboard unfortunately, works fairly well now.

@tatokis 
Just for fun i ran `dmesg | grep -i usb` as you requested and it seems to detect the keyboard. No errors as of now.

@light2yellow 
~~I attempted to install Libusb but excuse me if this is a silly mistake, but when I attempt to compile it using any method possible it gives me an error stating:
`xcrysdale@xcrysdale-MS-7917:~/libusb$ sudo ./autogen.sh
./bootstrap.sh: 6: ./bootstrap.sh: autoreconf: not found`
If you have any advice feel free to let me know I'm fairly new to this :).~~
EDIT: I'm a doofu who did not have autoconf installed, apologies. On the other hand the install fails with the quit error of 1, you can see the logs below if you would like.
`xcrysdale@xcrysdale-MS-7917:~/libusb$ ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:37: installing './compile'
configure.ac:29: installing './install-sh'
configure.ac:29: installing './missing'
Makefile.am: installing './INSTALL'
examples/Makefile.am: installing './depcomp'
libusb/Makefile.am:5: error: Libtool library used but 'LIBTOOL' is undefined
libusb/Makefile.am:5:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libusb/Makefile.am:5:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libusb/Makefile.am:5:   If 'LT_INIT' is in 'configure.ac', make sure
libusb/Makefile.am:5:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1`

From the error it looks like another directory error. Any insight?

Once again thanks all for your help, you've been amazing! :)

fleischie, 2017-05-06:
Closing, because seems resolved.

@Xcrysdale Something about 'LT_INIT' not being defined correctly. Please refer to the installation or troubleshooting guides of `libusb`. I think debugging this issue in this thread is a little out of scope. But good luck, though. 👍

closed 2017-05-06 by fleischie

light2yellow, 2017-05-06:
@Xcrysdale 
for this to work you need to install the complete set, e.g. `autogen`, `autoconf`, `automake` etc. Because I don't know how `autotools` work, I suggest you re-clone the clean repo after the dependencies are installed and try once again.

light2yellow, 2017-05-06:
@Xcrysdale 
you'll also need to use `sudo` for `xusb`.

Here's my output (STRAFE RGB):
```
sudo ./xusb -d 1b1c:1b20
[sudo] password for l2y: 
Using libusb v1.0.21.11197

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000001] [000024e9] libusb: debug [libusb_init] created default context
[ 0.000015] [000024e9] libusb: debug [libusb_init] libusb v1.0.21.11197
[ 0.000033] [000024e9] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000039] [000024e9] libusb: debug [op_init] bulk continuation flag supported
[ 0.000041] [000024e9] libusb: debug [op_init] zero length packet flag supported
[ 0.000050] [000024e9] libusb: debug [op_init] sysfs can relate devices
[ 0.000052] [000024e9] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000214] [000024f7] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.001318] [000024e9] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.001323] [000024e9] libusb: debug [linux_get_device_address] scan usb3
[ 0.001345] [000024e9] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.001349] [000024e9] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.001351] [000024e9] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.001407] [000024e9] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 0.001411] [000024e9] libusb: debug [linux_get_device_address] scan 3-2
[ 0.001427] [000024e9] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.001429] [000024e9] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.001432] [000024e9] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.001453] [000024e9] libusb: debug [linux_get_parent_info] Dev 0x1f70eb0 (3-2) has parent 0x1f76af0 (usb3) port 2
[ 0.001483] [000024e9] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.001485] [000024e9] libusb: debug [linux_get_device_address] scan 3-3
[ 0.001499] [000024e9] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.001501] [000024e9] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.001504] [000024e9] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.001518] [000024e9] libusb: debug [linux_get_parent_info] Dev 0x1f815b0 (3-3) has parent 0x1f76af0 (usb3) port 3
[ 0.001543] [000024e9] libusb: debug [linux_get_device_address] getting address for device: 3-4 detached: 0
[ 0.001546] [000024e9] libusb: debug [linux_get_device_address] scan 3-4
[ 0.001560] [000024e9] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.001562] [000024e9] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.001564] [000024e9] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.001578] [000024e9] libusb: debug [linux_get_parent_info] Dev 0x1f816b0 (3-4) has parent 0x1f76af0 (usb3) port 4
[ 0.001602] [000024e9] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.001604] [000024e9] libusb: debug [linux_get_device_address] scan usb4
[ 0.001617] [000024e9] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.001619] [000024e9] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[ 0.001622] [000024e9] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
[ 0.001656] [000024e9] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.001658] [000024e9] libusb: debug [linux_get_device_address] scan usb1
[ 0.001671] [000024e9] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.001673] [000024e9] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.001676] [000024e9] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.001711] [000024e9] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.001713] [000024e9] libusb: debug [linux_get_device_address] scan 1-1
[ 0.001726] [000024e9] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.001728] [000024e9] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.001731] [000024e9] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.001745] [000024e9] libusb: debug [linux_get_parent_info] Dev 0x1f70a00 (1-1) has parent 0x1f76bb0 (usb1) port 1
[ 0.001770] [000024e9] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.001772] [000024e9] libusb: debug [linux_get_device_address] scan usb2
[ 0.001785] [000024e9] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.001787] [000024e9] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.001790] [000024e9] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.001824] [000024e9] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.001826] [000024e9] libusb: debug [linux_get_device_address] scan 2-1
[ 0.001839] [000024e9] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.001841] [000024e9] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[... cut ...]
[ 0.003742] [000024e9] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.003745] [000024e9] libusb: debug [usbi_remove_pollfd] remove fd 6
[ 0.003752] [000024e9] libusb: debug [usbi_remove_pollfd] remove fd 8
[ 0.003768] [000024f7] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

Looking at it I have some ideas about macOS - instead of bustling with half-working tools targeting 10.9 when it's 2017 outside, we could just advise to run `xusb` to get the endpoints info for new devices.