mattanger/ckb-next #139: animations require both ckb-next-daemon and ckb to be running

ToadJamb, 2017-04-09:
I was just curious to see what would happen to the settings if I saved a profile to my new strafe keyboard without ckb running at all. The good news is that it saved the base colors and ckb doesn't have to be running at all (I mean neither `ckb` nor `ckb-daemon` are running) for me to have the basic colors I want, which is fantastic!

The not-so-good (I won't go so far as to say 'bad') news is that it appears that I have to have both `ckb-daemon` and `ckb` running for the animations to work. I was expecting to need the daemon, but I'm wondering if I missed something that would allow me to quit ckb itself. Am I doing something wrong or is that expected behavior?

tatokis, 2017-04-09:
Yes, the animations are all done with ckb, the daemon just acts as a bridge between ckb and the keyboard. It has its own commands (see DAEMON.md).

You can only save static profiles to the keyboard.

closed 2017-04-09 by tatokis

ToadJamb, 2017-04-09:
got it. thank you! really appreciate the work on this.