mattanger/ckb-next #143: Music visualizer isn't working

pseudoArchaea, 2017-04-10:
Using the latest git build - the music visualizer isn't working. I can see in pavucontrol that the fallback device does have sound coming out of it(I can hear that after all), but the keys are all white.

I tried using the default animation settings and I tried messing with them too to no avail.

Is this not supposed to be working yet or am I doing something incorrectly?

tatokis, 2017-04-13:
The mviz plugin is a bit of a mess at the moment, it seems that the key mapping is off. (Patches are welcome.)

First set a solid background colour, add the mviz plugin, open up pavucontrol and make sure it is recording from the monitor device.
This should be enough to get something to show.

You can then open the settings menu for the animation, and open the "Fade color" prompt.
Try setting a colour on the far right.

pseudoArchaea, 2017-04-13:
Actually now I can't see the mviz plugin at all. I can see it's somehow related to the pulse libraries, which I don't recall removing.

  pacman -Q | grep pulse
lib32-libpulse 10.0-1
libcanberra-pulse 0.30-8
libpulse 10.0-2
pulseaudio 10.0-2
pulseaudio-alsa 2-3
pulseaudio-gconf 10.0-2

tatokis, 2017-04-13:
Run `ckb` in a terminal and see if it finds it there, if not, try reinstalling.

It just depends on libpulse and libpulse-simple, so be sure to have both installed, otherwise it won't get compiled.

light2yellow, 2017-04-13:
@tatokis libpulse-simple is the part of libpulse on Arch (just for the information):
```
$ pkgfile libpulse-simple.so
extra/libpulse
community/libpulse-nosystemd
multilib/lib32-libpulse
```

pseudoArchaea, 2017-04-15:
Recompiled against the new git version, now the plugin is there again.

>First set a solid background colour, add the mviz plugin, open up pavucontrol and make sure it is recording from the monitor device.
>This should be enough to get something to show.

Yup, that did it. Had to set it in pavucontrol.

The mapping does seem off, for some reason the 3 keys on the far right on the numpad act as if they were way closer to the left.

Also the way the visualizer works right now(from left to right) makes right half of the keyboard almost always untouched by the visualizer. Would prefer it to be from bottom to top.

Great work so far!

light2yellow, 2017-05-10:
> the 3 keys on the far right on the numpad act as if they were way closer to the left

Could you elaborate on what are those keys? Every key has a mark/tag/picture on it after all :)

What keyboard do you use?

tatokis, 2017-05-10:
The - + and enter keys all the way to the right.
I can reproduce this too

nonchip, 2017-10-08:
also I'd like to note there's always one key "stuck" at 100%.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/13

closed 2018-01-22 by tatokis