mattanger/ckb-next #146: Install fails with "colormap.cpp:(.text+0x148): undefined reference to `__cxa_throw_bad_array_new_length'" Ubuntu 16.04

transvanosis, 2017-04-10:
Trying to install on Ubuntu 16.04 and the install fails with error 1: colormap.cpp:(.text+0x148): undefined reference to `__cxa_throw_bad_array_new_length'
collect2: error: ld returned 1 exit status
Makefile.Release:464: recipe for target '../../bin/ckb' failed

Error 2: Makefile:75: recipe for target 'sub-src-ckb-make_first' failed
make: *** [sub-src-ckb-make_first] Error 2

Please help?
I am also running Anaconda for Python 3, if that makes any difference??

light2yellow, 2017-04-10:
Hi. Make sure all the deps have been installed.

transvanosis, 2017-04-10:
I've installed all the listed requirements: Qt5, libudev, zlib, gcc, g++, and glibc - sudo apt-get install build-essential libudev-dev qt5-default zlib1g-dev libappindicator-dev

Just went through each one again now and gcc and g++ weren't installed I added them both and now getting the error:

Makefile.Release:458: recipe for target '../../bin/ckb' failed
make[2]: Leaving directory '/home/grant/Desktop/ckb-newdev/src/ckb'
Makefile:38: recipe for target 'release' failed
make[1]: Leaving directory '/home/grant/Desktop/ckb-newdev/src/ckb'
Makefile:74: recipe for target 'sub-src-ckb-make_first' failed
In file included from main.c:1:0:
../ckb/ckb-anim.h: In function ‘ckb_getline’:
../ckb/ckb-anim.h:241:5: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
     fgets(line, sizeof(line), stdin);
     ^
release/colormap.o: In function `ColorMap::alloc(int)':
colormap.cpp:(.text+0x148): undefined reference to `__cxa_throw_bad_array_new_length'
collect2: error: ld returned 1 exit status
make[2]: *** [../../bin/ckb] Error 1
make[1]: *** [release] Error 2
make: *** [sub-src-ckb-make_first] Error 2

light2yellow, 2017-04-11:
Output of `qmake -query QMAKE_MKSPECS`?

transvanosis, 2017-04-11:
gcc --version: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
g++ --version: g++ (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609

Output of qmake -query:
QT_SYSROOT:
QT_INSTALL_PREFIX:/home/grant/anaconda
QT_INSTALL_ARCHDATA:/home/grant/anaconda
QT_INSTALL_DATA:/home/grant/anaconda
QT_INSTALL_DOCS:/home/grant/anaconda/doc
QT_INSTALL_HEADERS:/home/grant/anaconda/include/qt
QT_INSTALL_LIBS:/home/grant/anaconda/lib
QT_INSTALL_LIBEXECS:/home/grant/anaconda/libexec
QT_INSTALL_BINS:/home/grant/anaconda/bin
QT_INSTALL_TESTS:/home/grant/anaconda/tests
QT_INSTALL_PLUGINS:/home/grant/anaconda/plugins
QT_INSTALL_IMPORTS:/home/grant/anaconda/imports
QT_INSTALL_QML:/home/grant/anaconda/qml
QT_INSTALL_TRANSLATIONS:/home/grant/anaconda/translations
QT_INSTALL_CONFIGURATION:/home/grant/anaconda
QT_INSTALL_EXAMPLES:/home/grant/anaconda/examples
QT_INSTALL_DEMOS:/home/grant/anaconda/examples
QT_HOST_PREFIX:/home/grant/anaconda
QT_HOST_DATA:/home/grant/anaconda
QT_HOST_BINS:/home/grant/anaconda/bin
QT_HOST_LIBS:/home/grant/anaconda/lib
QMAKE_SPEC:linux-g++
QMAKE_XSPEC:linux-g++
QMAKE_VERSION:3.0
QT_VERSION:5.6.2

light2yellow, 2017-04-11:
But I asked for `which gcc`, not `gcc --version`, it should give you the path.

Looks like you have Qt (and probably compilers too) from Anaconda's installation. Not sure I can help you. Not a big fan of virtualization so I don't know much about it. Maybe the issue is that compilers are not exported, can be just anything. I recommend you try compiling any other qmake-relied program (for example Tiled (map editor, on GitHub, the project structure is very similar to ckb-next)) and see if the issue persists. I'm also not very aware of qmake. We might be missing something essential in our config, though. I also recommend you asking somewhere where Anaconda's​ community lives.

transvanosis, 2017-04-11:
which gcc: /usr/bin/gcc
which g++: /usr/bin/g++

transvanosis, 2017-04-11:
Perhaps if I move the anaconda installation and try without it?

light2yellow, 2017-04-11:
You won't solve anything by just moving. You can try __re__moving it and hope for the best (I mean, if it cleans up all the broken symlinks after itself), installing normal packages normally afterwards, but I'm not sure this will help, because the damage is done. Nevertheless, try it.

transvanosis, 2017-04-11:
Thank you for pointing me in the right direction. 
I simply renamed the Anaconda directory from anaconda to ana-conda and ran ./quickinstall again and CKB has now installed.

Obviously the Anaconda installation was throwing a wrong path somewhere and causing the install fail errors.

light2yellow, 2017-04-11:
Glad you've figured it out in the end.

However, this is not really a solution. It's either you not following Anaconda's regulations regarding environment maintenance or us not using a particular qmake feature.

closed 2017-04-11 by light2yellow