mattanger/ckb-next #148: /dev/input/ckbX-Files with size 0

frickler24, 2017-04-11:
since yesterday I have sometime the following issue:
I start using ckb (testing branch) with my K95RGB. M65 is unplugged at startup, because I normally use a G700S and use the M65 for debugging only.
Now I plug the M65 but it is not detected by ckb. Yesterday in the branch for Issue #121 I inserted a warning message if I cannot read anything from the features-File in the corresponding /dev/input/ckbX/-directory.
Today I have thiss issue again. The two dirs for K95 and M65 look as follows:
```
lutz@Mainfrix ~/Projekte/ckb-next $ ll /dev/input/ckb*
/dev/input/ckb0:
insgesamt 12K
4,0K -rw-r--r-- 1 root root 159 Apr 11 11:54 connected
4,0K -rw-r--r-- 1 root root   5 Apr 11 10:55 pid
4,0K -rw-r--r-- 1 root root  12 Apr 11 10:55 version

/dev/input/ckb1:
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Apr 11 11:26 cmd
4,0K -rw-r--r-- 1 root root 56 Apr 11 10:55 features
4,0K -rw-r--r-- 1 root root  5 Apr 11 10:55 fwversion
4,0K -rw-r--r-- 1 root root 32 Apr 11 10:55 model
   0 prw-r--r-- 1 root root  0 Apr 11 11:25 notify0
4,0K -rw-r--r-- 1 root root  5 Apr 11 10:55 pollrate
4,0K -rw-r--r-- 1 root root 33 Apr 11 10:55 serial

/dev/input/ckb2:
insgesamt 0
0 prw-rw-rw- 1 root root 0 Apr 11 11:54 cmd
0 -rw-r--r-- 1 root root 0 Apr 11 11:54 features
0 -rw-r--r-- 1 root root 0 Apr 11 11:54 fwversion
0 -rw-r--r-- 1 root root 0 Apr 11 11:54 model
0 prw-r--r-- 1 root root 0 Apr 11 11:54 notify0
0 -rw-r--r-- 1 root root 0 Apr 11 11:54 pollrate
0 -rw-r--r-- 1 root root 0 Apr 11 11:54 serial
```
Nice - huh? Now I can Unplug and replug the mouse again and again - all time the same effect.
Now I unplugged both and the deamon cleared both directories.
When I plug in the m65, I get the following structure:
```
ll /dev/input/ckb*                                                                        
/dev/input/ckb0:                                                                          
insgesamt 12K                                                                             
4,0K -rw-r--r-- 1 root root 156 Apr 11 11:59 connected                                    
4,0K -rw-r--r-- 1 root root   5 Apr 11 10:55 pid                                          
4,0K -rw-r--r-- 1 root root  12 Apr 11 10:55 version                                      
                                                                                          
/dev/input/ckb1:                                                                          
insgesamt 0                                                                               
0 prw-rw-rw- 1 root root 0 Apr 11 11:59 cmd                                               
0 -rw-r--r-- 1 root root 0 Apr 11 11:59 features                                          
0 -rw-r--r-- 1 root root 0 Apr 11 11:59 fwversion                                         
0 -rw-r--r-- 1 root root 0 Apr 11 11:59 model                                             
0 prw-r--r-- 1 root root 0 Apr 11 11:59 notify0                                           
0 -rw-r--r-- 1 root root 0 Apr 11 11:59 pollrate                                          
0 -rw-r--r-- 1 root root 0 Apr 11 11:59 serial                                            
                                                                                          
/dev/input/ckb2:                                                                          
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Apr 11 11:58 cmd
4,0K -rw-r--r-- 1 root root 47 Apr 11 11:58 features
4,0K -rw-r--r-- 1 root root  5 Apr 11 11:58 fwversion
4,0K -rw-r--r-- 1 root root 29 Apr 11 11:58 model
   0 prw-r--r-- 1 root root  0 Apr 11 11:58 notify0
4,0K -rw-r--r-- 1 root root  5 Apr 11 11:58 pollrate
4,0K -rw-r--r-- 1 root root 33 Apr 11 11:58 serial
```
And the deamon noticed, that he hast two identical M65 detected:
```
lutz@Mainfrix /dev/input $ cat ckb0/connected 
/dev/input/ckb1 0D02303DAE3D8C69539981F7F5001941 Corsair M65 RGB Gaming Mouse
/dev/input/ckb2 0D02303DAE3D8C69539981F7F5001941 Corsair M65 RGB Gaming Mouse
```
This is reproducible.
Now plugging in K95 again:
```
lutz@Mainfrix /dev/input $ cat ckb0/connected 
/dev/input/ckb1 0D02303DAE3D8C69539981F7F5001941 Corsair M65 RGB Gaming Mouse
/dev/input/ckb2 0D02303DAE3D8C69539981F7F5001941 Corsair M65 RGB Gaming Mouse
/dev/input/ckb3 0F022014AE3B9406537275B6F5001945 Corsair K95 RGB Gaming Keyboard
/dev/input/ckb4 0F022014AE3B9406537275B6F5001945 Corsair K95 RGB Gaming Keyboard
```
this creates four dirs:
```
lutz@Mainfrix /dev/input $ ll ckb*
ckb0:
insgesamt 12K
4,0K -rw-r--r-- 1 root root 318 Apr 11 12:03 connected
4,0K -rw-r--r-- 1 root root   5 Apr 11 10:55 pid
4,0K -rw-r--r-- 1 root root  12 Apr 11 10:55 version

ckb1:
insgesamt 0
0 prw-rw-rw- 1 root root 0 Apr 11 12:02 cmd
0 -rw-r--r-- 1 root root 0 Apr 11 12:02 features
0 -rw-r--r-- 1 root root 0 Apr 11 12:02 fwversion
0 -rw-r--r-- 1 root root 0 Apr 11 12:02 model
0 prw-r--r-- 1 root root 0 Apr 11 12:02 notify0
0 -rw-r--r-- 1 root root 0 Apr 11 12:02 pollrate
0 -rw-r--r-- 1 root root 0 Apr 11 12:02 serial

ckb2:
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Apr 11 12:02 cmd
4,0K -rw-r--r-- 1 root root 47 Apr 11 12:02 features
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:02 fwversion
4,0K -rw-r--r-- 1 root root 29 Apr 11 12:02 model
   0 prw-r--r-- 1 root root  0 Apr 11 12:02 notify0
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:02 pollrate
4,0K -rw-r--r-- 1 root root 33 Apr 11 12:02 serial

ckb3:
insgesamt 0
0 prw-rw-rw- 1 root root 0 Apr 11 12:03 cmd
0 -rw-r--r-- 1 root root 0 Apr 11 12:03 features
0 -rw-r--r-- 1 root root 0 Apr 11 12:03 fwversion
0 -rw-r--r-- 1 root root 0 Apr 11 12:03 model
0 prw-r--r-- 1 root root 0 Apr 11 12:03 notify0
0 -rw-r--r-- 1 root root 0 Apr 11 12:03 pollrate
0 -rw-r--r-- 1 root root 0 Apr 11 12:03 serial

ckb4:
insgesamt 0
0 prw-rw-rw- 1 root root 0 Apr 11 12:03 cmd
0 -rw-r--r-- 1 root root 0 Apr 11 12:03 features
0 -rw-r--r-- 1 root root 0 Apr 11 12:03 fwversion
0 -rw-r--r-- 1 root root 0 Apr 11 12:03 model
0 prw-r--r-- 1 root root 0 Apr 11 12:03 notify0
0 -rw-r--r-- 1 root root 0 Apr 11 12:03 pollrate
0 -rw-r--r-- 1 root root 0 Apr 11 12:03 serial
```
Interesting.
Reboot did the job yesterday...

light2yellow, 2017-04-11:
Could this be by any chance a forgotten `killall` and/or `systemctl stop`?

frickler24, 2017-04-11:
Good idea, but I had no reinstallation of the daemon, just debugging the client.
pgrep -l ckb returns only one daemon (with 1106 a "young" processID).
systemctrl restart daemon runs normally and produces a new process, but the issue remains.

light2yellow, 2017-04-11:
So, the issue (two descriptors) is __always__ reproducible? What is the minimal repro? Variants:

1) Plug in the mouse...
2) Plug in the kb, unplug the kb, plug in the mouse...
3) Plug in the kb, plug in the mouse...

...and two descriptors appear

4) other combination?

I am not able to repro this because of no mouse, but this can clear it up for you a bit to debug.

frickler24, 2017-04-11:
After reboot (I did a cold start) I get the following structure:
```
lutz@Mainfrix ~/Projekte/ckb-next $ ll /dev/input/ckb*
/dev/input/ckb0:
insgesamt 12K
4,0K -rw-r--r-- 1 root root 159 Apr 11 12:29 connected
4,0K -rw-r--r-- 1 root root   5 Apr 11 12:29 pid
4,0K -rw-r--r-- 1 root root  12 Apr 11 12:29 version

/dev/input/ckb1:
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Apr 11 12:31 cmd
4,0K -rw-r--r-- 1 root root 64 Apr 11 12:29 features
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:29 fwversion
4,0K -rw-r--r-- 1 root root 29 Apr 11 12:29 model
   0 prw-r--r-- 1 root root  0 Apr 11 12:29 notify0
   0 prw-r--r-- 1 root root  0 Apr 11 12:29 notify1
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:29 pollrate
4,0K -rw-r--r-- 1 root root 33 Apr 11 12:29 serial

/dev/input/ckb2:
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Apr 11 12:29 cmd
4,0K -rw-r--r-- 1 root root 56 Apr 11 12:29 features
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:29 fwversion
4,0K -rw-r--r-- 1 root root 32 Apr 11 12:29 model
   0 prw-r--r-- 1 root root  0 Apr 11 12:29 notify0
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:29 pollrate
4,0K -rw-r--r-- 1 root root 33 Apr 11 12:29 serial

/dev/input/ckb3:
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Apr 11 12:29 cmd
4,0K -rw-r--r-- 1 root root 64 Apr 11 12:28 features
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:28 fwversion
4,0K -rw-r--r-- 1 root root 29 Apr 11 12:28 model
   0 prw-r--r-- 1 root root  0 Apr 11 12:28 notify0
   0 prw-r--r-- 1 root root  0 Apr 11 12:28 notify1
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:28 pollrate
4,0K -rw-r--r-- 1 root root 33 Apr 11 12:28 serial

and

lutz@Mainfrix ~/Projekte/ckb-next $ cat /dev/input/ckb0/connected 
/dev/input/ckb1 0D02303DAE3D8C69539981F7F5001941 Corsair M65 RGB Gaming Mouse
/dev/input/ckb2 0F022014AE3B9406537275B6F5001945 Corsair K95 RGB Gaming Keyboard
```
KB is detected twice (but not mentioned in the "connected"-file), mouse once. Macro keys and lightning are running at the mouse but not at the KB (which is ok, because of the duplicated cmd and notify-files.
I'll do a recompile of testing branch without any changes and report here.

frickler24, 2017-04-11:
After recompile of "testing" It is the same effect: KB twice, mouse once.
So I'll do some debugging later.
```
lutz@Mainfrix ~/Projekte/ckb-next $ !ll
ll /dev/input/ckb*
/dev/input/ckb0:
insgesamt 12K
4,0K -rw-r--r-- 1 root root 159 Apr 11 12:49 connected
4,0K -rw-r--r-- 1 root root   6 Apr 11 12:49 pid
4,0K -rw-r--r-- 1 root root  12 Apr 11 12:49 version

/dev/input/ckb1:
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Apr 11 12:49 cmd
4,0K -rw-r--r-- 1 root root 64 Apr 11 12:49 features
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:49 fwversion
4,0K -rw-r--r-- 1 root root 29 Apr 11 12:49 model
   0 prw-r--r-- 1 root root  0 Apr 11 12:49 notify0
   0 prw-r--r-- 1 root root  0 Apr 11 12:49 notify1
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:49 pollrate
4,0K -rw-r--r-- 1 root root 33 Apr 11 12:49 serial

/dev/input/ckb2:
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Apr 11 12:49 cmd
4,0K -rw-r--r-- 1 root root 56 Apr 11 12:49 features
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:49 fwversion
4,0K -rw-r--r-- 1 root root 32 Apr 11 12:49 model
   0 prw-r--r-- 1 root root  0 Apr 11 12:49 notify0
   0 prw-r--r-- 1 root root  0 Apr 11 12:49 notify1
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:49 pollrate
4,0K -rw-r--r-- 1 root root 33 Apr 11 12:49 serial

/dev/input/ckb3:
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Apr 11 12:29 cmd
4,0K -rw-r--r-- 1 root root 64 Apr 11 12:28 features
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:28 fwversion
4,0K -rw-r--r-- 1 root root 29 Apr 11 12:28 model
   0 prw-r--r-- 1 root root  0 Apr 11 12:28 notify0
   0 prw-r--r-- 1 root root  0 Apr 11 12:28 notify1
4,0K -rw-r--r-- 1 root root  5 Apr 11 12:28 pollrate
4,0K -rw-r--r-- 1 root root 33 Apr 11 12:28 serial
lutz@Mainfrix ~/Projekte/ckb-next $ 
lutz@Mainfrix ~/Projekte/ckb-next $ !ca
cat /dev/input/ckb0/connected 
/dev/input/ckb1 0D02303DAE3D8C69539981F7F5001941 Corsair M65 RGB Gaming Mouse
/dev/input/ckb2 0F022014AE3B9406537275B6F5001945 Corsair K95 RGB Gaming Keyboard
lutz@Mainfrix ~/Projekte/ckb-next $ 
```

frickler24, 2017-04-11:
Now I changed the USB plug for the mouse - the issue has gone.
But I will do still some investigation.
So please let this issue open.

frickler24, 2017-04-12:
AAAAARRRGGGG: No more space left on device!
```
Mainfrix ckb2 # ll ../ckb1
insgesamt 20K
   0 prw-rw-rw- 1 root root  0 Apr 12 20:26 cmd
4,0K -rw-r--r-- 1 root root 64 Apr 12 20:28 features
4,0K -rw-r--r-- 1 root root  5 Apr 12 20:28 fwversion
4,0K -rw-r--r-- 1 root root 29 Apr 12 20:28 model
   0 prw-r--r-- 1 root root  0 Apr 12 20:26 notify0
4,0K -rw-r--r-- 1 root root  5 Apr 12 20:28 pollrate
4,0K -rw-r--r-- 1 root root 33 Apr 12 20:28 serial
Mainfrix ckb2 # ll
insgesamt 0
0 prw-rw-rw- 1 root root 0 Apr 12 20:28 cmd
0 -rw-rw-rw- 1 root root 0 Apr 12 20:30 model
0 prw-r--r-- 1 root root 0 Apr 12 20:28 notify0
Mainfrix ckb2 # echo huhu >> model 
-bash: echo: Schreibfehler: Auf dem Gerät ist kein Speicherplatz mehr verfügbar.
Mainfrix ckb2 # df .
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
udev            8,4G    8,4G     0  100% /dev
```
bootchart is filling up my udev Device:
.bootchart/log/proc_ps.log has 7.5GB

And why isn't the udevice growing anymore? The disks have space enough!
I close this issue, sorry!

closed 2017-04-12 by frickler24