mattanger/ckb-next #149: New handling for debug and status information

frickler24, 2017-04-11:
This is follower of #147 as @light2yellow recommended

I did some experiments with qDebug.
Since Qt 5.6 they implemented a verbosity-flag. Now you can
- decide which function you use (qDebug, qWarning, qCritical, qFatal)
- give a verbosity information with it:
```
qDebug().setVerbosity(7) << "This is a very critical issue and should be displayed in any case!";
qDebug().setVerbosity(0) << "If a user wants to read this stuff, he may select this.";
```
Standard Qt verbosity is 2.
In the next days I will post a skeleton how I think we may use this.
Like [this article](http://stackoverflow.com/questions/4954140/how-to-redirect-qdebug-qwarning-qcritical-etc-output#4954188) describes some aspects of it.
Or has someone experience with it?

frickler24, 2017-04-12:
Following the article mentioned above, I played a bit with QDebug class.
Standard function is to decide between 5 modes:
- qFatal() is used if you cannot do anything more than exiting. This method needs a special handler and I think we should **not** use it, because of handling notify-channels etc.
- qCritical() is for really critical situations (eg unknown usb events, closed pipes, no more memory, SIGSEGV and so on). I Think we have some error-handling in the code (some handling may be anhanced :-)) to use this mode.
- qWarning() is less than critical and may be used if we could not load new firmware etc.
- qInfo() is less again and might be used for messages like our standard message "Downloaded new firmware list. 12 entries found."
- qDebug() should be used for debugging only. Here we can produce lots of raw data for our purposes.

The handling of these modes is easy and brings a second option: More info about the source of the message.
If we run "make" a Qt-Parameter QT_NO_DEBUG is set - if we run "make debug" it is not set.
We can use this to generate shorter or longer debug info. As a first implementation I added filename, lineno and function-name in the long part.
Then I added a CLI parameter -d / --debuglevel \<num\>
To be compatible with the value range of the verbosity in Qt in general, I created 8 debug-modes (0..7):
```
/// \brief debugLevel controls the amount of debug output.
///     0 = no debug output, just hard errors (Fatal) are displayed in a short form
///     1 = no debug output, just hard errors (Fatal) are displayed in a long form
///     2 = Error-handling and inconsistencies (Fatal + Critical = default value) in short form
///     3 = as 2 but in long form
///     4 = as 2 + Warnings (in short form)
///     5 = as 4 but in long form
///     6 = All info, but no debug data (eg logging from usb) in short form
///     7 = full debug output with complete logging runtime data (eg logging from usb) in long form
```
By adding one test-message per mode to main (), you get the following amount of information - depending on the debug level:
```
lutz@Mainfrix ~/Projekte/ckb-next/src/ckb $ ../../bin/ckb -d0
DebugLevel is set to 0

lutz@Mainfrix ~/Projekte/ckb-next/src/ckb $ ../../bin/ckb -d1
DebugLevel is set to 1

lutz@Mainfrix ~/Projekte/ckb-next/src/ckb $ ../../bin/ckb -d2
DebugLevel is set to 2
Critical: Critical messages are shown

lutz@Mainfrix ~/Projekte/ckb-next/src/ckb $ ../../bin/ckb -d3
DebugLevel is set to 3
Critical: Critical messages are shown (main.cpp:319, int main(int, char**))

lutz@Mainfrix ~/Projekte/ckb-next/src/ckb $ ../../bin/ckb -d4
DebugLevel is set to 4
Warning: Warnings are shown
Critical: Critical messages are shown

lutz@Mainfrix ~/Projekte/ckb-next/src/ckb $ ../../bin/ckb -d5
DebugLevel is set to 5
Warning: Warnings are shown (main.cpp:318, int main(int, char**))
Critical: Critical messages are shown (main.cpp:319, int main(int, char**))

lutz@Mainfrix ~/Projekte/ckb-next/src/ckb $ ../../bin/ckb -d6
DebugLevel is set to 6
Info: Info messages are shown
Warning: Warnings are shown
Critical: Critical messages are shown

lutz@Mainfrix ~/Projekte/ckb-next/src/ckb $ ../../bin/ckb -d7
DebugLevel is set to 7
Debug: Debug messages are shown (main.cpp:316, int main(int, char**))
Info: Info messages are shown (main.cpp:317, int main(int, char**))
Warning: Warnings are shown (main.cpp:318, int main(int, char**))
Critical: Critical messages are shown (main.cpp:319, int main(int, char**))
Debug: Scanning  "/usr/lib/ckb-animations/ckb-gradient" (animscript.cpp:81, bool AnimScript::load())
Debug: Scanning  "/usr/lib/ckb-animations/ckb-heat" (animscript.cpp:81, bool AnimScript::load())
Debug: Scanning  "/usr/lib/ckb-animations/ckb-mviz" (animscript.cpp:81, bool AnimScript::load())
Debug: Scanning  "/usr/lib/ckb-animations/ckb-pinwheel" (animscript.cpp:81, bool AnimScript::load())
Debug: Scanning  "/usr/lib/ckb-animations/ckb-rain" (animscript.cpp:81, bool AnimScript::load())
Debug: Scanning  "/usr/lib/ckb-animations/ckb-random" (animscript.cpp:81, bool AnimScript::load())
Debug: Scanning  "/usr/lib/ckb-animations/ckb-ripple" (animscript.cpp:81, bool AnimScript::load())
Debug: Scanning  "/usr/lib/ckb-animations/ckb-wave" (animscript.cpp:81, bool AnimScript::load())
Debug: Starting  "/usr/lib/ckb-animations/ckb-gradient" (animscript.cpp:315, void AnimScript::begin(quint64))
Debug: Downloaded new firmware list. 12 entries found. (kbfirmware.cpp:141, void KbFirmware::processDownload(QNetworkReply*))
```
Any comments on this?

fleischie, 2017-04-12:
I like the debug-levels but think we don't need 7 levels. I usually have either a `-v`, a `-vv`, and a `-vvv` option to setup the required verbosity. Together with a `-q` option we may have 5 levels:
1. `-q` - no output whatsoever.
2. (no flag) - "normal" output (i.e. like it's now)
3. `-v` - Level 3 as mentioned above.
4. `-vv` - Level 5/6 as mentioned above.
5. `-vvv` - Level 7.

This approach feels more acquainted to me, is just my opinion, though so I'm open for discussion. 😄

frickler24, 2017-04-12:
OK @fleischie, sometimes less is more!
(in shell scripting I learned 30 years ago: more is less than less :-))

light2yellow, 2017-04-14:
I honestly prefer the 7-level debug, however, @fleischie's way is also a good approach, except one thing:
> (no flag) - "normal" output (i.e. like it's now)

It shouldn't be like it's now by default because most of time you don't want to see "foo entries found", this is really a very minor message. Instead, there should exist a message with higher priority (warning, for example), telling the user that no entries were found.

frickler24, 2017-05-05:
@tatokis your opinion? 7 or 5 levels?

tatokis, 2017-05-05:
I'd vote for 3, I can never imagine myself doing ckb -vvvvvvv. That's just ridiculous

light2yellow, 2017-05-14:
@tatokis 
But @frickler24 was on about `-d[0-7]`, not `-vvvvvvvv`. I also find it ridiculous, but `-dX` seems good to me.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/14

closed 2018-01-22 by tatokis