mattanger/ckb-next #150: "No Devices Connected" Please Help

DesperateKidNeedsHelp, 2017-04-12:
I tried many different solutions for CKB to work so I can change my keyboard's ugly color. Please can anyone help me?

JarrenJ, 2017-04-12:
Could you be more specific what exactly is the problem... what isn't connecting? Did you trying restarting your computer? What version are you using. Are you running MacOS or Linux? Please add more details so we can pinpoint your problem.

--**thx**

DesperateKidNeedsHelp, 2017-04-15:
I tried restarting my computer and unplugging the cords. I am running MacOS Sierra. My keyboard works just fine but I just want this color to be changed and maybe an animation to go along. I have the K70 RGB which is one of the keyboards that is supported by CKB. @JarrenJ

tatokis, 2017-04-15:
How did you install ckb?

DesperateKidNeedsHelp, 2017-04-15:
I downloaded it as binary @tatokis

DesperateKidNeedsHelp, 2017-04-15:
Oops..

tatokis, 2017-04-15:
The binaries are ridiculously updated, you have to compile from source. 
There's plenty of info in the readme on how to do it.

DesperateKidNeedsHelp, 2017-04-15:
So I have to download XCode and Qt5 to it? @tatokis

light2yellow, 2017-04-17:
Yes, and while you're waiting for the software to download, find 10 minutes to read the README.

DesperateKidNeedsHelp, 2017-04-19:
Ok, everything is downloaded, but.. What's up with the scripts how do I use them??? I noticed that you can do script stuff in Qt5. Please help these are my last steps.. @tatokis

frickler24, 2017-04-19:
@DesperateKidNeedsHelp Your question is answered in the READE.md. PLease follow @light2yellow's hint and **read it**.

DesperateKidNeedsHelp, 2017-04-19:
Ok. @frickler24

DesperateKidNeedsHelp, 2017-04-19:
Well I read everything, but I already read it before.

tatokis, 2017-04-19:
https://github.com/mattanger/ckb-next#building-from-source
`The easiest way to build the driver is with the quickinstall script, which is present in the ckb-master folder. Double-click on quickinstall and it will compile the app for you, then ask if you'd like to install it system-wide.`

light2yellow, 2017-04-20:
https://stackoverflow.com/questions/6466945/fastest-way-to-download-a-github-project
Second result for "how to download from github" request.

DesperateKidNeedsHelp, 2017-04-20:
Ok, I remember downloading this not knowing what I was doing. I just don't know how to make the code actually.. work. @light2yellow

light2yellow, 2017-04-20:
To make the code actually work, you first read the link from my last message, then you read the README and then execute just 3 commands - cloning, cd'ing and quickinstall'ing.

closed 2017-04-20 by light2yellow