mattanger/ckb-next #153: L

unloco101, 2017-04-13:
I have a scimitar rgb which is not working on Manjaro/Arch linux.  My lsusb shows:
`Bus 001 Device 002: ID 1b1c:1b3e Corsair `

this usbid is different than the one in the listed troubleshooting section.
0x1B1C:0x1B1E

is it possible that this is causing my `problem?`

unloco101, 2017-04-13:
Holy crap i got it to work! First of all, thank you for your work and proper code comments.
For anyone else out there who are looking for a quick fix with the same mouse.  I edited /src/ckb/daemon/usb.h and replaced the usb id with mine.  Used "find" to locate all the old id's, followed the code comments on where else to edit, then rebuild.

unloco101, 2017-04-13:
Thanks for the update. I used ckb-next-latest-git on AUR, which i assumed was the testing branch. I honestly am not familiar with the workflow / managing repos, gits, and branches.  Again thank you all for your work!

tatokis, 2017-04-13:
@unloco101 I wasn't aware of that, thanks :)
CC @light2yellow 

Isn't the "latest" package supposed to have the code from testing? Could you please have a look?

light2yellow, 2017-04-13:
It is and it does. Line 15: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ckb-next-latest-git#n15

closed 2017-04-13 by light2yellow