mattanger/ckb-next #154: doesn't work with karabiner

lsentkiewicz, 2017-04-14:
I am using [Karabiner](https://pqrs.org/osx/karabiner/) to remap some shortcuts.
For example:
CMD + TAB <=> CTRL + TAB

When I use both ckb and Karabiner, the changes from Karabiner are ignored.
Is there any workaround for this?

light2yellow, 2017-04-14:
What's the version of macOS?

lsentkiewicz, 2017-04-14:
El Capitan
v 10.11.6

light2yellow, 2017-04-14:
Not going to lie, I'm very happy to see a 10.11 user :)
This is not related to this issue, but would you be so kind to test #106 and tell us (in that thread) whether modifier keys still work for you? This is really important.

frickler24, 2017-04-15:
@lsentkiewicz I'm not a macOS user. so Ican't test it - but maybe this could be an idea:
Does Kabiner have a CLI? If yes, you can map via Binding / Program a cli call to Karabiner when prsseing CMD+TAB. I once saw this approach in an issue in ccMSC.

light2yellow, 2017-04-30:
Also, why won't you try ckb's macro interface?

lsentkiewicz, 2017-05-01:
@frickler24 no cli
See tekezo/Karabiner#830 I think CKB disables Karabiner

@light2yellow 
Is it possible to map CTRL+TAB to a macro?

light2yellow, 2017-05-01:
Well, first of all, you should switch on Karabiner-Elements as it's the only Karabiner being maintained now. Secondly, AFAIK there's no way to control the kb with both ckb's daemon and Karabiner's daemon. Thirdly, there's a document, `DAEMON.md`, that describes how to create  macros (either through CLI or through GUI). I never bothered myself with macros but I hope the document covers it. Forthly, there is a opened PR where people are working on macro enhancement, it should contain the info on how to record them, too.

frickler24, 2017-05-02:
@lsentkiewicz Unfortunately, modifier keys are not handled at the moment in the way you ask for.
So if you try to select "lctrl" and "tab" in the GUI and add a macro to it, the macro is executed if you type lctrl OR tab and not if you type both of them. But you may create a macro for a function key with the resulting key element you want.
Modifier keys will be handled in a later version, but currently a lot of other tasks have higher priority. First the delay-PR should be merged, before we change macro handling again. This is marked for 0.2.8.

fleischie, 2017-05-05:
@light2yellow Karabiner is for Mac OS up to El Capitan (10.11). Karabiner-Elements is for Sierra (10.12), so at least the premise is valid. Also I can reproduce the described behavior.

One workaround could be to set the keys via `ckb-next`, so Ctrl becomes Cmd and vice versa, as here:
[image]

Otherwise upgrade to Sierra and open a whole different can of worms. (I'll just work through the linked issue on Karabiner and extend my comment).

----

It seems the daemon intercepts the signals from the keyboard before Karabiner access them as part of their application. (I suspect it's because of the daemon, which runs "more" "low-level" than Karabiner.

Anyway. Using `ckb-next` actually seems to conflict with Karabiner. I actually am not able to tell you more.

----

What's the next steps? Do we label this `wontfix` as it seems to be an obscure combination of both `ckb-next` + Karabiner + OS version (who knows, whether this works/how it behaves on different OSs?). Do we suggest a cheap "workaround" by rebinding the keys via `ckb`? (Which I don't find a workaround, as it doesn't at all target the issue at hand.)

Yeah, that's just my two cents. Any thoughts on this?

light2yellow, 2017-05-05:
@fleischie 
Karabiner-Elements are 10.11.+.
The changes in the underlying implementation are significant yet I didn't investigate, nevertheless it's worth for the OP to try the new version.

light2yellow, 2017-05-05:
Regarding thoughts on this: I guess this is clearly and `enhancement` - we already implement some kind of macro operations and remappings, adding modifier keys would just be a rigorization of ckb's functionality.

closed 2018-01-21 by tatokis