mattanger/ckb-next #158: Caps Lock won't unlock

Cryocubed, 2017-04-18:
Hitting Caps Lock on Lux K70 RGB turns on caps lock. The keyboard light on the K70 and on the mac keyboard both turn on. Pressing caps lock further won't turn caps lock off however.

Cryocubed, 2017-04-18:
Related to #151

light2yellow, 2017-04-18:
Do you experience freezes like in #23?

Cryocubed, 2017-04-18:
Doesn't seem to be. I've experienced no crashes. Similarly, changing the caps lock mode didn't solve the issue.

light2yellow, 2017-04-20:
Hmm. Okay. So the Caps Lock issue doesn't happen to everyone.

Anyway, we have issues with modifiers state not being recorded or vanished or changed when not expected.

CC
@auibrian

light2yellow, 2017-04-20:
Maybe we're able to add a good amount of debugging messages for @Cryocubed to post the logs? (And have @frickler24's work on debug levels merged)

Cryocubed, 2017-04-20:
I can post whatever, just let me know what you need/how to get it.

Cryocubed, 2017-04-24:
Unsure of what changed, but the problem has fixed itself.

closed 2017-04-24 by Cryocubed

Donzap, 2017-11-11:
I have the exact same problem, when I hit Caps on my K65 RGB it write cap letters, but i press it again and it didn't unlock, if I want to unlock caps I must press caps on laptop's keyboard (I use it on a macbook pro).

Any idea or clue on what you did?