mattanger/ckb-next #160: Cannot change key colors

ToyReid, 2017-04-21:
So I just installed Ubuntu for the first time on my computer and figured I'd try to remake my keyboard's profile I normally use.

After installing ckb-next, whenever I try to select a key and then hit the change color button, the program freezes and spits this out:

[image]

I'm able to navigate and use the rest of the GUI fine, and can add animations to keys. It's just changing colors that seems to be giving me a problem.

Any help's much appreciated.

tatokis, 2017-04-21:
Excellent! Ubuntu 17.04 shipped with broken Qt 5.7.

We've only seen this on Arch so far, and I personally still run Ubuntu 16.04, so I can't tell you exactly what to do.
This thread ( https://bbs.archlinux.org/viewtopic.php?pid=1658017#p1658017 ) might be useful, and also our README has some info (under GLib errors) https://github.com/mattanger/ckb-next/blob/master/README.md#troubleshooting

If you do figure it out, please report back with how you solved it, so we can add it to the readme.
Thanks :)

tatokis, 2017-04-21:
Perhaps you can try installing qt5ct from https://launchpad.net/~nilarimogard/+archive/ubuntu/webupd8

Whisprin, 2017-04-26:
Installing qt5ct alone did not help me. I have to execute `QT_QPA_PLATFORMTHEME=qt5ct ckb` in order to get the color dialog to open.

Arch Linux with ckb-next-git (master branch) and QT 5.8

closed 2017-04-30 by light2yellow

embluk, 2017-05-01:
On the readme where it says to remove QT configuration files I could not find where they were so I followed Whisprin and installed qt5ct and then ran QT_QPA_PLATFORMTHEME=qt5ct ckb
Still does not work for me, I ran qt5ct and I get:
Please remove the QT_STYLE_OVERRIDE environment variable
The QT_QPA_PLATFORMTHEME environment variable is not set correctly

light2yellow, 2017-05-01:
Readme also provides links to other issues. Some of them contain an Arch Linux thread that contains the solution. This is the most comprehensive source of things to try. 
Different setups require different algorithms.

All issues that have Qt-GTK problems should be reported to your distro's bug tracker, not here.

A passage about Qt config files will be removed as it distracts more than helps. By the way, it talks about style overrides. I suppose it's covered by the Arch thread.

embluk, 2017-05-02:
Sorry but I've tried everything in https://bbs.archlinux.org/viewtopic.php?id=214147

No luck at all. I mean I should be able to install Qt and CKB on one of the most used distros being Linux Mint without any issues. I'm not using any custom setup at all, I feel this is more of an issue with the way ckb is setup and not really Qt-GTK...

This problem with the colour dialog crashing ckb is quite an issue if it does not work on the latest Ubuntu or Linux Mint.

I'm not really sure what to do now as I tried everything mentioned :(

embluk, 2017-05-02:
Ok, I seem to have fixed my problem by uninstalling Qt 5.8 and installing Qt 5.5 and rebuilding CKB from source using `./quickinstall` with Qt 5.5. CKB worked 100% with Qt 5.5.

Then because I wanted Qt 5.8, I uninstalled Qt 5.5 and installed Qt 5.8 and CKB seems to run fine and no crashes what so ever when I change the colour, it even uses the 5.8 colour picker rather than the one that comes with Qt 5.5. 

My guess is that this is working because CKB was built with Qt 5.5 where it does not have the GTK issue and after I installed Qt 5.8 I never rebuilt CKB. I bet if I updated CKB from source it would be built with 5.8 and again my issue would come back.

So if there is ever an update which I need, I will probably just have to keep building CKB with Qt 5.5 from now, unless I can solve the issue with Qt 5.8...

light2yellow, 2017-05-02:
@embluk Fortunately I got an access to a Linux Mint box, will test this issue until the end of the week.

frickler24, 2017-05-02:
@embluk I run liinux mint with several versions of ckb since a year (Mint 17.{1,2,3}, Mint 18, Mint 18.1), but with KDE, (18.0 short time with xfce) --> no problem. 
It must be a problem with gtk installation or Qt-gtk.

embluk, 2017-05-03:
> @embluk I run liinux mint with several versions of ckb since a year (Mint 17.{1,2,3}, Mint 18, Mint 18.1), but with KDE, (18.0 short time with xfce) --> no problem.
It must be a problem with gtk installation or Qt-gtk.

Yeah its something to do with Qt >= 5.7.
Because I've been trying out different Linux distros over the last week for me to use full time as a daily OS, I was on elementary OS before and CKB worked fine right out of the box, no GTK issues at all.

Then I switched over to Linux Mint because elementary wasn't working for me elsewhere... I did notice on Linux Mint when running the `./quickinstall` after it finishes and asks you if you would like to start CKB and I enter 'y' it does not come up at all, I have to go to the menu and open it from there. Also not sure if this is suppose to happen but upon reboot (I selected CKB to be system wide and run at startup) CKB does not show up in the task bar on mint, but CKB does work because my keyboard changes to the colour I set before and if I run CKB from the menu again it opens and shows in the tray again.

tatokis, 2017-05-03:
"and I enter 'y' it does not come up at all"
Send us the output  of quickinstall.

embluk, 2017-05-03:
> "and I enter 'y' it does not come up at all"
Send us the output  of quickinstall.

I would but if I run the quick install again with Qt 5.8 installed, I think my issue would come back again because CKB would be built with Qt 5.8 and not 5.5. I should of saved the output when I did it, but I was more focused on trying to get CKB to be fully working so I did not think much of the 'Would you like to run CKB now' not working :/

I suppose I could create a virtual machine with Linux Mint installed and get the output that way?

frickler24, 2017-05-03:
Interesting. In the last weeks I sometimes had the same effect. When compiling all stuff via quickinstall, the last step (starting ckb in backgound) had no effect. I'll have a closer look at it when I finished the travis setup. Starting it by hand works.

light2yellow, 2017-05-03:
> There is a syntax error, it tries to start /usr/ckb
Indeed, sorry about that - missed one occurrence when the directory layout was changed. I suggest you push a fix ASAP.

Regarding Mint - can I replicate this on Cinnamon? Long time ago, when I had Arch with XFCE, I was able to replicate this. From what I remember, Cinnamon should be gtk2 or gtk3 based?

tatokis, 2017-05-03:
@light2yellow testing or master? I lost track
Also, wrt Mint, what is the issue exactly? I also got confused. If it is tray icons, then all we can do is make sure libappindicator support is compiled. Otherwise the Qt tray icon takes over.

frickler24, 2017-05-03:
@light2yellow Waht I need to know is the path under darwin:
`[[ "$OSTYPE" == "darwin"* ]] && PREFIX="/Applications"`
is it /Applications or /Applications/bin ?

light2yellow, 2017-05-03:
@tatokis `testing`. Do we have a problem with `master` too? :D 

Also, in addition to ckb and ckb-daemon not starting, `testing` needs a slightly different approach in the end to be compatible with macOS part.

@frickler24 it is `/Applications`.

frickler24, 2017-05-03:
@light2yellow Please have a look on it.
And yes, the same bug is an master.

tatokis, 2017-05-03:
@frickler24 I don't think we changed the paths in master though.

frickler24, 2017-05-03:
@tatokis Sorry, you are right. I had some testing with my master branch, so I got the newer quickinstall.
Master is OK.

commit f2f31f38 (2017-05-03): Fix #160 for installation on linux

The path in the variable PREFIX is too short for linux,
so for linux its usage must be PREFIX/bin.