mattanger/ckb-next #161: "No devices connected" error, no lighting

stellarsarah, 2017-04-22:
I just got a K70 RGB Rapidfire. First thing I did was plug into a windows computer and used CUE to update firmware (CUE says it's running 2.05). I also fiddled around a bit with some of the presets available in CUE and saved one to device memory, in case that's relevant. I then installed ckb on my mac and plugged in the keyboard. 

So far, still unable to get the RGB lights to work at all on either of my macs, both running 10.9.5. Ckb just says "No devices connected". Keyboard input registers fine--even the music control buttons are communicating with iTunes with no hassle. Just zero lighting. 

I've tried a bunch of different combinations of toggling the poll rate switch, restarting computers, uninstalling and reinstalling ckb, and switching up which USB ports I use. I don't have a hub handy so I haven't tried that. And while troubleshooting, I saw the linked forum post about USB connection requirements and have since been plugging in the connector with the two arrows icon before the one with the keyboard icon each time. Still nothing.

Sorry for any extraneous info. Let me know if you have any suggestions or want any further info. Thank you for reading.

tatokis, 2017-04-22:
Compile from source

stellarsarah, 2017-04-22:
It's working now, thanks.

closed 2017-04-22 by stellarsarah

hypendex, 2017-07-15:
What does compile from source mean?

light2yellow, 2017-07-15:
@hypendex 

It means that instead of using a macOS pkg (and suppose from the word "iTunes" @stellarsarah is/was using macOS) you follow the README instructions.

But you'd better google it anyway.

Also, if you have a recent version of Qt (like 5.9) then you'll get compilation errors that we don't know how to fix right now.