mattanger/ckb-next #162: Support for new keyboard K63

tonyfreelance, 2017-04-25:
Hi guys,

Firstly thanks for this awesome driver. Recently I just got a new K63 which seems not supported yet. I would highly appreciate if anyone can support this one. I'm using it on MacOS 10.12.4

Thanks :)

tatokis, 2017-04-25:
Please run the ckb-dev-detect script located in the root of the repository.

tonyfreelance, 2017-04-25:
I did run it. And after it said: "This script will collect information about the Corsair devices in your system.Please make sure they are plugged in and press Enter.", I hit Enter and my iTerm screen disappeared so quicky that I didn't notice what happened..

tatokis, 2017-04-25:
Did it generate a report file?
Try opening a terminal manually, and then run
`cd path/to/ckb-next/source
./ckb-dev-detect`

That should tell you what went wrong

tonyfreelance, 2017-04-25:
Hey, it works now!

Here is what I got after following your instruction:
`ckb-dev-detect - Tue Apr 25 21:00:40 +07 2017

Kernel: Darwin tony 16.5.0 Darwin Kernel Version 16.5.0: Fri Mar  3 16:52:33 PST 2017; root:xnu-3789.51.2~3/RELEASE_X86_64 x86_64

OS Version:
ProductName:    Mac OS X
ProductVersion: 10.12.4
BuildVersion:   16E195

USB Devices:
        Corsair Gaming K63 Keyboard:

          Product ID: 0x1b40
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 1.00
          Serial Number: 02028039AEC9810456AD84DCF5001947
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14100000 / 3
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0`

light2yellow, 2017-05-03:
Linking this to #2.

light2yellow, 2017-05-14:
@Segflow 
None.

You can start implementing this if you would like to, it seems to be a subset of other keyboard models, so it needs only choosing from and reducing the amount of existing features for other, more feature-complete keyboards.

tatokis, 2017-05-27:
As a quick solution that might just work, try editing usb.h and adding your keyboard as K65_NRGB.

Try the following patch and see what happens.
```
diff --git a/src/ckb-daemon/usb.h b/src/ckb-daemon/usb.h
index d2a577b..3f37d59 100644
--- a/src/ckb-daemon/usb.h
+++ b/src/ckb-daemon/usb.h
@@ -40,8 +40,8 @@
 
 #define P_K65           0x1b17
 #define P_K65_STR       "1b17"
-#define P_K65_NRGB      0x1b07
-#define P_K65_NRGB_STR  "1b07"
+#define P_K65_NRGB      0x1b40
+#define P_K65_NRGB_STR  "1b40"
 #define P_K65_LUX       0x1b37
 #define P_K65_LUX_STR   "1b37"
 #define P_K65_RFIRE     0x1b39
```

tatokis, 2017-05-27:
CC @Segflow

Segflow, 2017-05-27:
@tatokis 
Same.

tatokis, 2017-05-27:
@Segflow what happens if you stop ckb-daemon and run it manually from a terminal?

Segflow, 2017-05-27:
Actually after `sudo systemctl start ckb-daemon` the keyboard stop working. (nothing happens when i type something)

Segflow, 2017-05-27:
Even with manual start:
```
 ~/ckb-next/bin   master ●  sudo ckb-daemon 
    ckb: Corsair RGB driver v0.2.8
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming K63 Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Setup finished for /dev/input/ckb1

```

tatokis, 2017-05-27:
Try this 
```
diff --git a/src/ckb-daemon/usb.h b/src/ckb-daemon/usb.h
index d2a577b..8ddbbdb 100644
--- a/src/ckb-daemon/usb.h
+++ b/src/ckb-daemon/usb.h
@@ -38,6 +38,9 @@
 #define V_CORSAIR       0x1b1c
 #define V_CORSAIR_STR   "1b1c"
 
+#define P_K63           0x1b40
+#define P_K63_STR       "1b40"
+
 #define P_K65           0x1b17
 #define P_K65_STR       "1b17"
 #define P_K65_NRGB      0x1b07
@@ -46,7 +49,7 @@
 #define P_K65_LUX_STR   "1b37"
 #define P_K65_RFIRE     0x1b39
 #define P_K65_RFIRE_STR "1b39"
-#define IS_K65(kb)      ((kb)->vendor == V_CORSAIR && ((kb)->product == P_K65 || (kb)->product == P_K65_NRGB || (kb)->product == P_K65_LUX || (kb)->product == P_K65_RFIRE))
+#define IS_K65(kb)      ((kb)->vendor == V_CORSAIR && ((kb)->product == P_K65 || (kb)->product == P_K65_NRGB || (kb)->product == P_K65_LUX || (kb)->product == P_K65_RFIRE || (kb)->product == P_K63))
 
 #define P_K70                0x1b13
 #define P_K70_STR            "1b13"
@@ -122,7 +125,7 @@ const char* product_str(short product);
 /// The difference between non RGB and monochrome is, that monochrome has lights, but just in one color.
 /// nonRGB has no lights.
 /// Change this if new \b monochrome devices are added
-#define IS_MONOCHROME(vendor, product)  ((vendor) == (V_CORSAIR) && (product) == (P_STRAFE_NRGB))
+#define IS_MONOCHROME(vendor, product)  ((vendor) == (V_CORSAIR) && ((product) == (P_STRAFE_NRGB) || (product) == (P_K63)))
 
 /// For calling with a usbdevice*, vendor and product are extracted and IS_RGB() is returned.
 #define IS_RGB_DEV(kb)                  IS_RGB((kb)->vendor, (kb)->product)
```

Segflow, 2017-05-28:
Now the ckb tool detect a device with the name k63 but:
- Keyboard not working (no response from any key)
- nothing happen when i interact with ckb (set lighting mode etc..)

i close the deamon and i can type again using the keyboard

```
    ckb: Corsair RGB driver v0.2.8
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming K63 Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Setup finished for /dev/input/ckb1

```

tatokis, 2017-06-09:
Looks like I never posted about this in here, @tonyfreelance other than a few media keys, the rest should be working with code from this branch.
https://github.com/mattanger/ckb-next/tree/k63-nrgb

sonikbhoom, 2017-08-27:
@tatokis I used the src.rpm from johanh and added your patch to the spec file, rebuilt the RPM and installed.  I can use the keyboard (as before: with , but it does not get its own tab in the qt gui.  Is this expected behaviour?  The keyboard works alongside the SABRE RGB I have, but no color mode changes.  

`Dmesg on current boot:
Aug 27 18:14:41.980588 darkdog3 kernel: usb 1-1.3: New USB device found, idVendor=1b1c, idProduct=1b40
Aug 27 18:14:41.980659 darkdog3 kernel: usb 1-1.3: Product: Corsair Gaming K63 Keyboard
Aug 27 18:14:41.980694 darkdog3 kernel: usb 1-1.3: Manufacturer: Corsair
Aug 27 18:14:41.981034 darkdog3 kernel: input: Corsair Corsair Gaming K63 Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:1.0/input/input2
Aug 27 18:14:41.981446 darkdog3 kernel: usb 1-1.4: New USB device found, idVendor=1b1c, idProduct=1b2f
Aug 27 18:14:41.981517 darkdog3 kernel: usb 1-1.4: Product: Corsair Gaming SABRE RGB Mouse
Aug 27 18:14:41.981552 darkdog3 kernel: usb 1-1.4: Manufacturer: Corsair
Aug 27 18:14:41.981600 darkdog3 kernel: input: Corsair Corsair Gaming SABRE RGB Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4:1.0/input/input3
Aug 27 18:14:41.981645 darkdog3 kernel: hid-generic 0003:1B1C:1B2F.0003: input,hidraw1: USB HID v1.11 Mouse [Corsair Corsair Gaming SABRE RGB Mouse] on usb-0000:00:1a.0-1.4/input0
Aug 27 18:14:41.982264 darkdog3 kernel: hid-generic 0003:1B1C:1B40.0001: input,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair Gaming K63 Keyboard] on usb-0000:00:1a.0-1.3/input0
Aug 27 18:14:41.982308 darkdog3 kernel: hid-generic 0003:1B1C:1B2F.0004: usb_submit_urb(ctrl) failed: -1
Aug 27 18:14:41.982349 darkdog3 kernel: hid-generic 0003:1B1C:1B2F.0004: timeout initializing reports
Aug 27 18:14:41.982359 darkdog3 kernel: input: Corsair Corsair Gaming SABRE RGB Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4:1.1/input/input5
Aug 27 18:14:41.982395 darkdog3 kernel: hid-generic 0003:1B1C:1B2F.0004: input,hiddev0,hidraw4: USB HID v1.11 Mouse [Corsair Corsair Gaming SABRE RGB Mouse] on usb-0000:00:1a.0-1.4/input1
Aug 27 18:14:41.982485 darkdog3 kernel: hid-generic 0003:1B1C:1B40.0006: usb_submit_urb(ctrl) failed: -1
Aug 27 18:14:41.982523 darkdog3 kernel: hid-generic 0003:1B1C:1B40.0006: timeout initializing reports
Aug 27 18:14:41.982535 darkdog3 kernel: input: Corsair Corsair Gaming K63 Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:1.1/input/input6
Aug 27 18:14:41.982571 darkdog3 kernel: hid-generic 0003:1B1C:1B40.0006: input,hiddev0,hidraw5: USB HID v1.11 Keyboard [Corsair Corsair Gaming K63 Keyboard] on usb-0000:00:1a.0-1.3/input1
Aug 27 18:14:46.570693 darkdog3 kernel: ata5.00: ATA-8: Corsair Force GS, 5.05A, max UDMA/133
Aug 27 18:14:46.622827 darkdog3 kernel: scsi 4:0:0:0: Direct-Access     ATA      Corsair Force GS A    PQ: 0 ANSI: 5
Aug 27 18:14:58.936950 darkdog3 kernel: input: ckb1: Corsair Gaming SABRE RGB Mouse as /devices/virtual/input/input21
Aug 27 18:14:58.936969 darkdog3 kernel: input: ckb1: Corsair Gaming SABRE RGB Mouse as /devices/virtual/input/input22

Dmesg on last boot:

Daemon on current boot:
-- Logs begin at Sun 2017-08-27 18:14:41 EDT, end at Sun 2017-08-27 18:17:21 EDT. --
Aug 27 18:14:58.572145 darkdog3 systemd[1]: Started Corsair Keyboard Daemon.
Aug 27 18:14:58.572327 darkdog3 systemd[1]: Starting Corsair Keyboard Daemon...
Aug 27 18:14:58.576309 darkdog3 ckb-daemon[1017]: ckb: Corsair RGB driver v0.2.8
Aug 27 18:14:58.587384 darkdog3 ckb-daemon[1017]: [I] Root controller ready at /dev/input/ckb0
Aug 27 18:14:58.597094 darkdog3 ckb-daemon[1017]: [I] >>>vendor = 0x1b1c, product = 0x1b2f, path = /dev/bus/usb/001/004, syspath = /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4
Aug 27 18:14:58.718742 darkdog3 ckb-daemon[1017]: [I] Connecting Corsair Gaming SABRE RGB Mouse at /dev/input/ckb1
Aug 27 18:14:58.718854 darkdog3 ckb-daemon[1017]: [I] claiming interfaces. name=Corsair Gaming SABRE RGB Mouse, firmware=0204; Got >> 3<< as ep_str
Aug 27 18:14:58.718952 darkdog3 ckb-daemon[1017]: [I] claiming 3 endpoints
Aug 27 18:14:58.889809 darkdog3 ckb-daemon[1017]: [I] Starting input thread for /dev/input/ckb1
Aug 27 18:15:01.091782 darkdog3 ckb-daemon[1017]: [I] Setup finished for /dev/input/ckb1`

tatokis, 2017-08-27:
@sonikbhoom that patch above was nowhere near enough what was needed to get this working.
The few issues I mentioned above remain, but I would suggest not using the package for now and compiling directly from newdev.

There have been a lot of changes in newdev, and I think it wouldn't be worth generating a patch and applying it manually.

sonikbhoom, 2017-08-28:
I downloaded the newdev branch and applied it to the SPEC file from the other src.rpm, removing the patches.  (note: I like clean, removable, installs and rpms can be made to clean up after themselves)
The keyboard is detected and I'm running beta-0.2.8.

sonikbhoom, 2017-08-28:
I downloaded the newdev branch and applied it to the SPEC file from the other src.rpm, removing the patches.  (note: I like clean, removable, installs and rpms can be made to clean up after themselves)
The keyboard is detected and I'm running beta-0.2.8. <edit> rebooting and the mouse doesn't work at all.  I have reverted and will try again later.

tatokis, 2017-10-14:
K63 is now in master. If anyone is willing to help me map out some of the broken media keys, it would be appreciated.

triyanarief, 2017-10-18:
ok, i just bought k63 and gonna try to install in sierra. hopefully it works.

acwest, 2017-11-05:
I am working on adding the Strafe MX Silent right now (1B44), and have been unable to determine where the firmware files are downloaded from. It should be indistinguishable from the Strafe, but I don't want to risk trying to flash the wrong firmware. How do we normally determine where to download firmwares from?

light2yellow, 2017-11-05:
@acwest we tip and try if it works. For some keyboards we weren't able to determine the addresses. However, the owner of the keyboard can always determine the address by monitoring the CUE application on Windows.

tatokis, 2018-01-21:
Should be fully supported in https://github.com/ckb-next/ckb-next

closed 2018-01-21 by tatokis

xipher1, 2018-04-09:
Not to dig up an old horse, but ;

> This script will collect information about the Corsair devices in your system.
> Please make sure they are plugged in and press Enter.
> 
> 2018-04-09 19:26:42.803 system_profiler[50325:8092645] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
> Done! Please send the following report to the developers.
> Location: [redacted]

and then

> $ cat ckb-next-dev-detect-report
> ckb-next-dev-detect - Mon Apr  9 16:09:36 HKT 2018
> 
> Kernel: Darwin blinky 17.5.0 Darwin Kernel Version 17.5.0: Mon Mar  5 22:24:32 PST 2018; root:xnu-4570.51.1~1/RELEASE_X86_64 x86_64
> 
> OS Version:
> ProductName:	Mac OS X
> ProductVersion:	10.13.4
> BuildVersion:	17E199
> 
> USB Devices:
>                 CORSAIR K63 Wireless Mechanical Gaming Keyboard:
> 
>                   Product ID: 0x1b45
>                   Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
>                   Version: 1.01
>                   Serial Number: 0202C023AF081CE7589194B3F5001BC7
>                   Speed: Up to 12 Mb/sec
>                   Manufacturer: CORSAIR
>                   Location ID: 0x14414000 / 32
>                   Current Available (mA): 500
>                   Current Required (mA): 500
>                   Extra Operating Current (mA): 0
> 
> Daemon log:
>     ckb-next: Corsair RGB driver 0.2.9-160-gf5802bf
> [I] Root controller ready at /var/run/ckb0
> [I] Registering EventTap for modifier keys.
> 
> GUI log:

This is obv in High Sierra, connected via USB, through a usb-c -> usb-a dongle. Keyboard works fine, but the GUI says that no device is connected.

tatokis, 2018-04-09:
@xipher1 Your keyboard is a wireless one, which is not supported at the moment.
That aside, open an issue in ckb-next/ckb-next. Not here.

xipher1, 2018-04-09:
oops, my bad, I had a bunch of tabs opened, thought I was left on the correct one. Sorry, I will do so in the morning! Thanks

ZirconiumX, 2018-04-09:
Actually, ckb-next/ckb-next has an issue, which is my wireless branch; though I suspect we might need to add the K63 and Dark Core simultaneously because of how much common protocol there is between them.

tatokis, 2018-04-09:
@ZirconiumX Indeed, but keep in mind, there are still differences such as the inability to control per key lighting (compared to per zone on the dark core).

xipher1, 2018-04-09:
Thanks for the prompt replies guys, I'll take a look at the issue tmrw, happy to help test anything, very comfortable with everything involved.

songliqiangcn, 2018-09-14:
I just bought a Corsair K63 Wireless Mechanical Gaming Keyboard. But looks it doesn't support Mac OS to select the light. Did search found this post. 
Does it support my Mac OS 10.13.6 ? 
Thanks,