mattanger/ckb-next #163: Scimitar Pro issues

mooop12, 2017-04-25:
I'm having some weird behavior with my Scimitar Pro mouse:

- CKB detects a firmware update when there is none. I believe it thinks the mouse is a standard Scimitar and tries to upload its firmware. Luckily it said it couldn't download the update or else it could have bricked my mouse (possibly).
- Macros don't work. Even inputting them manually they don't seem to do anything. This is the case even for the hardware profile, with CKB gui turned off. Binding to regular keys works fine.

tatokis, 2017-04-25:
1. #97 #155 
2. #121

mooop12, 2017-04-25:
Ok, the first part checks out. But if understand correctly, the macro issue you linked is related to recording only. My issue is different. Inputting the macro manually should work (for example "+f5, -f5"), and it does nothing for me. Even when the macro was saved to a hardware profile using Corsair software on Windows it does not work on Linux, only regular key bindings work.

tatokis, 2017-04-25:
Hardware profiles are unsupported, and there's no ETA as to when they will be supported.
Ckb switches the mouse to software mode.

Try running `xinput list`, see if your device is listed under keyboard too.

mooop12, 2017-04-25:
It is listed under keyboard twice as `ckb1: Corsair Gaming SCIMITAR PRO RGB Mo`.

LivingWithHippos, 2017-04-25:
@mooop12  If you wrote it as you pasted it it's wrong, you need to remove the space between the comma and the -f5.
With xinput list I only get it listed once under keyboard and once under Pointer, maybe that's your problem. I suppose you didn't actually use quotes and pressed apply as you said normal keys are working. Do ctrl or shift work?

mooop12, 2017-04-26:
Well, I fell stupid. I was actually trying it with each button on a separate line. That's how it appeared in the box when I clicked record and started pressing mouse buttons, I guess it's a part of the bug mentioned above. It's working fine now, thanks.

closed 2017-04-26 by mooop12