mattanger/ckb-next #164: macros and remapped keys

lsentkiewicz, 2017-04-25:
macOS 10.11

- Ctrl and Command keys are swapped.
- G9 is a simple macro: Command + Tab (`+lctrl,+tab,-tab,-lctrl`).

Steps:
a) I have two spaces and each space contains an instance of WebStorm IDE.
b) When I press `ctrl+~` it switches between WebStorm IDEs.
c) G9 (command + tab) should switch between tabs in a single IDE.

Bug:
When pressing c) the IDE windows are switches.
I think it invokes `ctrl+tab` and then `cmd+tab`. When I press G9 multiple times it starts working correctly.

Notes:
- It always works correctly if I press `cmd+tab` instead of G9.
- It occurs only with b) step.

frickler24, 2017-04-25:
I have no macOS myself, so I speculate with my answer: Your first info is 
> Ctrl and Command keys are swapped.

Ckb daemon does not know anything about this.
So if your macro is` +lctrl, +tab, -tab, -lctrl` - is it possible for your macOS that it does interpret it as `+cmd, +tab, -tab, -cmd`? Did you try a macro with this "reverse" definition?

lsentkiewicz, 2017-04-25:
No, the macro keys are correct.
My macro works correctly, but if I make b) step the behavior is weird.
It switches spaces and also tabs in the IDE, so it makes two actions.

lsentkiewicz, 2017-04-25:
Another use case:

a) I have two monitors
b) Monitor 1 has two spaces with two IDEs (the same situation from 1st post)
b) press `ctrl+~` in the Monitor 1 (switch between spaces)
c) focus Monitor 2 and go back to Monitor 1 (click with the mouse)
d) press G9 macro (ctrl+tab)
It works correctly.
If I skip the c) point, there will be a weird bug.
Is it possible that there is something wrong with tilde character?
Why does work correctly without a macro?

tatokis, 2017-04-25:
Do you have the right keyboard layout set in ckb? Settings tab -> keyboard layout

lsentkiewicz, 2017-04-26:
Polish layout

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/15

closed 2018-01-22 by tatokis