mattanger/ckb-next #165: Support pt-BR layout

rainerborene, 2017-04-25:
Continue https://github.com/ccMSC/ckb/issues/589

rainerborene, 2017-04-25:
@tatokis why one are you referring?
[image]

tatokis, 2017-04-25:
[image]

adrianosantanna, 2017-07-03:
Hi there, I'm arriving now and have one pt-BR K70 lux rgb keyboard.

Maybe I can help with some information about the keyboard, hoping someone can improve ckb-next to support pt-BR layout.

@tatokis, regarding your question (at https://github.com/ccMSC/ckb/issues/589) about the key next to the right shift, it appears that key is not mapped under ckb-next. Its light doesn't turn on and it doesn't work when ckb-next is installed and running.

I would be happy if I can provide additional information, just let me know.

PR #331 (2017-11-11): Add pt-br keyboard layout
Fix #165
closed 2017-11-11 by tatokis