mattanger/ckb-next #166: CKB Uses ~30% of CPU

Maxattax97, 2017-04-26:
I tweaked around in the CKB configuration for awhile and determined that the largest factor by far was the hardware FPS.

I had a single pinwheel-searchlight animation with rainbow colors which is now disabled. While deleting the animation had no effect on CPU usage, reducing FPS definitely did. After decreasing my FPS from 30 to 5, my CPU usage now lingers around 2%. However, such high CPU usage does not seem normal to me for a mere LED animation...

I'm using a Scimitar RGB.

tatokis, 2017-04-26:
Disable the animation preview

frickler24, 2017-04-26:
Had you switched on the "Show animated" check on the lighting tab? I had additional load between 15% and 25%-points when checked.

Maxattax97, 2017-04-27:
This helps when the configuration GUI is open, which disabling the animation preview _does_ help CPU usage. My issue was occurring when the GUI was closed. Today my mouse is running ~5% CPU with a pinwheel-searchlight animation at a reduced 15 FPS and GUI closed.

axettone, 2017-10-02:
I also have high CPU usage, even when GUI is closed. Reducing framerate from 30 to 10, CPU dropped from 20-22% to 7% (ckb process only), but I think it's still higher than what it should be.

pjcv, 2017-10-03:
I did some experimenting and I found that getMuteState() is taking up quite a lot of time when running in linux. I see ckb taking about 3.5% CPU with it running, and 0.5% with it commented out. Can someone else try this out and confirm?

closed 2018-01-21 by tatokis