mattanger/ckb-next #167: My K70 lux Rapidfire isn't being detected

dragonpo, 2017-04-29:


dragonpo, 2017-04-29:
Basically I have the app downloaded and everything, but it doesn't give me the screen that lets me mess with my RGB lighting just the screen w/ button setups.

light2yellow, 2017-04-29:
1. What is the platform?
2. Did you start and enable the `ckb-daemon`?
3. Try `testing` instead of `master`.

michael-mikkichm, 2017-05-03:
Mac Os X 10.12.1 (16B2555) Sierra. 
Corsair k70 Rapidfire issues with master and with testing branch. Testing branch seems to have moments when it discovers the device but no changes are applied to the keyboard itself. 
Firmware updated in windows to 2.05 (As far as I remember.)

`$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
Password:
    ckb: Corsair RGB driver beta-v0.2.7
[F] ckb-daemon is already running (PID 112). Try `killall ckb-daemon`.`

[image]

michael-mikkichm, 2017-05-03:
I updated the Xcode; launched it, updated all it wanted. 
Removed all the previous ckb -- clean my mac deinstall. Recompiled the ckb from test branch.
AND = WORKS! Corsair rapidfire k70 is shining!!!

closed 2017-06-09 by tatokis