mattanger/ckb-next #168: K95 RGB random macro/remapped key reset

RickCody, 2017-04-29:
I'm using ckb-next beta-v0.2.7 on macOS Sierra 10.12.4. I have most of the additional G buttons setup to open various programs and some of the keys which aren't as useful in macOS remapped to ones that are. I've got 3 keylighting profiles setup, one of which is a rainbow gradient and two have static keylighting.

About once a week, for no apparent reason, a bunch of stuff resets. Under Binding all of my macros and remapped keys are reset, under Lighting the keys which used to be rainbow gradient (stored in profile M1) are reset to white and there are 3 additional profiles called M Unnamed, and under Performance, all of my changes from the default are reset as well. My first sign is generally when I see my keys are all white (I usually have the rainbow profile active). Below is a screenshot from the Lighting panel when I pull it up and the issue has happened again.

[image]

I've searched macOS's console for "ckb" and found nothing, and I didn't see any other issues that seemed similar to mine. I compiled ckb myself using the quickinstall script, then after the issue appeared I recompiled and reinstalled ckb but to no avail.

Other than that one bug, I love ckb and it works wonderfully. I'd love to provide any additional information you need, just ask :)

Thanks

tatokis, 2017-04-29:
I think this is a duplicate of #83.

I have no idea where the settings are stored on macOS, but if you could figure out, once something resets/breaks it would be very helpful if you could compress the ckb profile and upload it.

frickler24, 2017-04-29:
Have a look at[ QSettings Documentation ](http://doc.qt.io/qt-5/qsettings.html) or [this thread](http://stackoverflow.com/questions/31994646/where-does-qt-c-store-system-scope-qsettings-on-mac-osx-10).

RickCody, 2017-05-03:
I'm not sure if it's a duplicate. However, I did figure out my "issue" in case anyone else is in a similar boat.

Turns out the issue was operator error; I don't know why, but I was under the impression that the "Save to Hardware" button saved everything to the hardware. So when ckb would reload the profiles from hardware, they would only load the lighting layer I did save, not the binding layer I did not save. 

I don't know if that's how things work on Linux, but at least for the macOS section of the documentation, maybe spell that out so that people like myself don't misunderstand what's saved when you click "Save to Hardware".

Thanks

closed 2017-05-03 by RickCody