mattanger/ckb-next #169: K70 lux RGB: CBK stopped detecting the KB without any reason

ethanstal, 2017-04-29:
I was playing on my computer, I left it for 30 minutes, and when I returned, my keyboard was not detected anymore. 
 I restarted CBK, I unplugged my KB, I tried all USB ports, I installed CBK again, I tried all KB modes and nothing worked!
Can you help me ?

ethanstal, 2017-04-30:
ps I'm using a mac

ethanstal, 2017-04-30:
i will answer myself: for those who does have  the same problem: uninstall ckb, use the sos disk 
nction, turn off your computer, unplug everything ( even the power cable), turn on  your computer, plug everything, install ckb, DON'T NEVER EVER SWITCH TO BIOS MODE, and it should be ok.

light2yellow, 2017-04-30:
> DON'T EVER SWITCH TO BIOS MODE

Yep, exactly. Glad you figured the problem out. By now we (at least me) have decent troubleshooting feedback experience and I guess we will update "Troubleshooting" and "Known Issues" to aim most common problems better. Both sections talk about bios mode but don't discourage it's usage completely. Bios mode should be used as a last resort.

closed 2017-04-30 by light2yellow