mattanger/ckb-next #170: Tried quickinstall and BUILD.md (Didn't work)

DesperateKidNeedsHelp, 2017-04-29:
Ok, I had an issue before stating I needed help to setup ckb. Apparently all of the contributors who were helping were showing all the steps I already did (but I thought I did it wrong..) Ok, ignore that non-important statement I gave. (Sorry) quickinstall does it's thing in Terminal just fine, but.. When it asks me to start the application it quickly shows me the painful "no devices connected" windoww then switches to the window I want to see in less than a second, then quits. BUILD.md.. I start ckb.pro in Qt5, perform Cmd+B.. And yea. That's it.. ckb.app is supposed to end up in my directory but it's not there. BUILD.md also informed me to use this command in Terminal "./qmake-auto && make" which is supposed to inform where I built the file to, but all I get is "No such file or directory." Please help me. Also kudos to contributors                             Ya'll doing fine

DesperateKidNeedsHelp, 2017-04-30:
Anybody..? ._.

NOSDuco, 2017-04-30:
Does `lsusb` output that the keyboard or device is connected?

DesperateKidNeedsHelp, 2017-04-30:
Ok, what's `lsub` output? And my keyboard is never on bios mode.

tdis8629, 2017-04-30:
Since you have macOS, please run `ckb-dev-detect`, then post the resulting output. Also, which keyboard do you own? I searched the prior post, but you never stated a model.

DesperateKidNeedsHelp, 2017-04-30:
@tdis8629 I have K70 RGB LUX and I tried quickinstall and BUILD.md after using ckb-dev-detect. Same result as last time, but when I start ckb it doesn't show the RGB lighting editor after it shows the "_no devices connected_" window.

DesperateKidNeedsHelp, 2017-04-30:
@tdis8629 Oh, Am I supposed send you the _ckb-dev-detect-report_?

DesperateKidNeedsHelp, 2017-04-30:
I tried opening ckb with Terminal and it shows the whole process of opening it. Will it be useful to find the problem??

tdis8629, 2017-04-30:
Correct. The output in the terminal could also help identify why your keyboard isn't being detected.

DesperateKidNeedsHelp, 2017-04-30:
ckb-dev-detect - Sun Apr 30 16:48:03 EDT 2017

Kernel: Darwin TiMaKoVs-iMac.local 16.4.0 Darwin Kernel Version 16.4.0: Thu Dec 22 22:53:21 PST 2016; root:xnu-3789.41.3~3/RELEASE_X86_64 x86_64

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.12.3
BuildVersion:	16D32

USB Devices:
        Corsair Gaming K70 LUX RGB Keyboard :

          Product ID: 0x1b33
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 2.05
          Serial Number: 10037019AEBD18445619FD42F5001946
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14200000 / 7
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0

DesperateKidNeedsHelp, 2017-05-01:
@frickler24 @tatokis @light2yellow @ruler501 @mattanger Help, Please?

light2yellow, 2017-05-01:
1. Stop the daemon: `sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist`
2. Launch the daemon manually: `sudo /Applications/ckb.app/Contents/Resources/ckb-daemon`
3. Post the output.

What branch are you running?

DesperateKidNeedsHelp, 2017-05-02:
@light2yellow OK.. HELP Computer is going WILD!!!! After I entered the stop daemon it's going BEZERK!!!!

DesperateKidNeedsHelp, 2017-05-02:
@light2yellow At first it was pressing Enter REALLY FAST AND non stop (I unplugged computer but now...) it's making me stop typing and being freaky ;-; My computer is dead please please help

DesperateKidNeedsHelp, 2017-05-02:
@light2yellow Oh gosh.. I fixed it. That was terrifying. I saw ckb keep popping up in the top menu bar, so I quickly deleted it and it fixed the problem.

DesperateKidNeedsHelp, 2017-05-02:
@light2yellow Do you think I should try your process again??

light2yellow, 2017-05-02:
`README`, section `Known Issues`:

> When starting the driver manually, the Terminal window sometimes gets spammed with enter keys. You can stop it by unplugging and replugging the keyboard or by moving the poll rate switch.

I gave you 3 instructions. In order to help yourself you need to evaluate them.
I asked 1 question. In order to help yourself you need to answer it.

In case you are more willing to help yourself now, before even answering the aforementioned question I encourage you to install the [`testing` branch](https://github.com/mattanger/ckb-next/tree/testing).

DesperateKidNeedsHelp, 2017-05-03:
@light2yellow Ok, unplugging or moving the polling rate doesn't work.. I have to delete ckb. Aside that, I went to the store today and when I came back the keyboard was glowing a beautiful rainbow color! I was so confused.. The Brightness Button was glowing instead of white so I pressed it. The keyboard went back to it's original red state. -_-

DesperateKidNeedsHelp, 2017-05-03:
@light2yellow 
Last login: Tue May  2 21:22:25 on ttys000
TiMaKoVs-iMac:~ TiMaKoV$ sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
Password:
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
TiMaKoVs-iMac:~ TiMaKoV$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo: /Applications/ckb.app/Contents/Resources/ckb-daemon: command not found
TiMaKoVs-iMac:~ TiMaKoV$ 

I think it's because I don't have ckb.app

light2yellow, 2017-05-03:
Ok, all the best.

closed 2017-05-03 by light2yellow