mattanger/ckb-next #171: Linux Mint GTK Widget Crash

embluk, 2017-05-01:
I've installed Linux Mint 18.1 64 Bit with the kernel 4.10 and installed CKB following the readme with the quickinstall file. When it comes to the part asking do you want to start CKB now? I enter 'y' and nothing comes up, so I search for CKB in the menu of Linux Mint and open it from there.

It opened and everything seems to work, however, when I make a new profile, select all keys and click on change colour to basically set a solid colour the app crashes, I ran CKB in QT in debug mode and got this:

```
(ckb:18458): GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'

(ckb:18458): GLib-GObject-WARNING **: cannot add class private field to invalid type '<invalid>'

(ckb:18458): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(ckb:18458): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:18458): GLib-GObject-WARNING **: cannot register existing type 'GtkBuildable'

(ckb:18458): GLib-GObject-CRITICAL **: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(ckb:18458): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed

(ckb:18458): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:18458): GLib-GObject-CRITICAL **: g_type_register_static: assertion 'parent_type > 0' failed

(ckb:18458): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
```

tatokis, 2017-05-01:
#160 and also read README.md, it's under troubleshooting.

closed 2017-05-01 by tatokis

embluk, 2017-05-01:
Where do I find the Qt configuration files? And what config files for QT do I delete?