mattanger/ckb-next #175: QT Application Freezes When Selecting Change Color

pkkid, 2017-05-05:
OS: Ubuntu Gnome 17.04
Keyboard: K70 Rapidfire (non-RGB)
Ckb Branch: testing

When trying to select "Change Color" the application consistently freezes with this output on the command line:

```
(ckb:3439): GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'
(ckb:3439): GLib-GObject-WARNING **: cannot add class private field to invalid type '<invalid>'
(ckb:3439): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
(ckb:3439): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed
(ckb:3439): GLib-GObject-WARNING **: cannot register existing type 'GtkBuildable'
(ckb:3439): GLib-GObject-CRITICAL **: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed
(ckb:3439): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed
(ckb:3439): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed
(ckb:3439): GLib-GObject-CRITICAL **: g_type_register_static: assertion 'parent_type > 0' failed
(ckb:3439): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
```

tatokis, 2017-05-05:
#160

closed 2017-05-05 by tatokis

light2yellow, 2017-05-05:
I guess a quote from README would be useful:

> If you see GLib critical errors like GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed and you are using:
> * Qt 5.8 and newer, remove your Qt configuration files and restart the ckb GUI. Also watch out for different style overridings in dotfiles under ~/ generated by Qt automatically, remove them as well.
> * Qt 5.7 and lower, install qt5ct package on Arch Linux (find a similar one for your distribution). That's all. This is a known Qt bug. It happened because Qt did not ship required GTK files.
> If the above didn't work out for you for some reason, see #70, #51, ccMSC/ckb#500, ccMSC/ckb#461.

pkkid, 2017-05-05:
Thanks, sorry about the noise. :+1:

benhur10cp, 2017-11-29:
I tried, and it appeared "E: Unable to locate package qt5ct"

feutro, 2017-12-22:
I just tackled this problem on my Ubuntu 17.10 install. The problem is still present with Qt 5.9. Here's what I did. 

I installed `qt5ct` from [this PPA](https://launchpad.net/~mati75/+archive/ubuntu/qt5ct). Now with `qt5ct` installed, the `QT_QPA_PLATFORMTHEME=qt5ct ckb` line works to launch the CKB app. 

However, if you want to launch CKB from the Activities menu or from a launcher, this won't work. Putting `QT_QPA_PLATFORMTHEME=qt5ct` in `/etc/environment` did not work either as there is something on my system that sets this environment variable to another value afterwards during either boot-up or login. 

So I simply made a bash script called `launchckb.sh` in my home directory with the following content:

> `#!/bin/bash`
> `QT_QPA_PLATFORMTHEME=qt5ct ckb`

and changed the launcher in `/usr/bin/applications/ckb.desktop` so that the `Exec` line is commented and a new one is added:

> `#Exec=ckb`
> `Exec=path_to_launchkb.sh`

So now the launcher for ckb executes the script (don't forget to set the appropriate permissions if necessary) and the environment variable is set only for CKB, which can be a good idea anyway.

Hope this helps!