mattanger/ckb-next #179: Tutorial please?

DesperateKidNeedsHelp, 2017-05-06:
Can anyone show me a quick tutorial on how to build ckb.app with BUILD.md? In BUILD.md it asks me to build ckb.pro in Qt5, but it doesn't work.

frickler24, 2017-05-06:
On what system? Mac? Linux? What linux distro? What does not work? Did you install all prereqs? ...
Always the same questions. If you want an answer which is helpful for you, please learn to ask.

DesperateKidNeedsHelp, 2017-05-06:
Mac |  It said to build it with Cmd + B and it will put ckb.app in the directory. I didn't see any changes at all and it came with a command that would search the app if it the process went well.  |  Prereqs???  |

tatokis, 2017-05-06:
Try this 
https://github.com/mattanger/ckb-next/releases/download/v0.2.8/ckb-v0.2.8.pkg

fleischie, 2017-05-06:
I guess we can also add an Install/Build from Source section to the Wiki, if we restructure the docs.

I think the pre-packaged binary is a good step in the right direction, though.

DesperateKidNeedsHelp, 2017-05-06:
@frickler24 Same result when I try quickinstall. When I open it shows ckb for a second and quits leaving me to deal with the _Enter_ key spamming me.

frickler24, 2017-05-07:
@DesperateKidNeedsHelp I think you have more than one installation on your system.
To Prereqs: Did you proper install ALL prerequisites mentioned in the README.md for macOS?
Are there updates available for Xcode or QT5?
Have there been errors (no warnings) in your last run of quickinstall?
Did you install it via launchctl (answered two times "y" in install procedure)?

What do you get if you run `find / -name quickinstall -print 2>/dev/null`?
And what is shown with `pgrep ckb`?
What services are running `sudo launchctl list`?

What timestamp has your daemon `ls -l /Library/LaunchDaemons/com.ckb.daemon.plist`? 
Is this date congruent to your last run of quickinstall?

Did you try the package mentioned by Tatokis above? 
Did you deinstall ckb before you installed the link from Tatokis?
What results did you get?

Because we do not have access to your system, you should create a picture with us, what is wrong with your system. This is the only way we can help and that is the only way to make the time for it.

DesperateKidNeedsHelp, 2017-05-07:
1. It was only Qt5 and Xcode right?
2. No
3. No
4. No I did not install via launchctl, nor know what that is.
5. 6. 7. I didn't have anything at all using those 3 commands.
8.-rw-r--r--@ 1 root  wheel  516 May  7 13:06 /Library/LaunchDaemons/com.ckb.daemon.plist          9. Yes
10. Yes
11. Yes
12. Opening ckb for a second and closing it after, then ckb is popping up in the top bar and then it goes away activating the Enter key every time it goes away from the top bar.
13. Is this a question?

frickler24, 2017-05-07:
I wonder why you do not get info from the lauchctl command, because it should show at least your daemon (which exists with normal timestamp). But I do not have a mac, so it is a bit fuzzy for me.
Please show me the listing of your directory, where ckb-next starts `cd <somewhere>/ckb-next; ls -l`.
In this ckb-next start directory do a recompile with  `yes | quickinstall > /tmp/logfile 2>&1` 
and send me /tmp/logfile you created.

DesperateKidNeedsHelp, 2017-05-07:
@frickler24 (Quick note) Ok, tatokis before sent me a new file called, ckb-next-testing, which was supposed to maybe would work instead of the original ckb-next. I know very little about the technical things about computers. So if you could bear with me and explain the steps a little more clear for me to understand that would be so helpful. Thanks.

DesperateKidNeedsHelp, 2017-05-08:
```
TiMaKoVs-iMac:~ TiMaKoV$ pwd ; whoami
/Users/TiMaKoV
TiMaKoV
```

I'm not getting ANY error messages, only "Permission denied."
But do you mean this? `/Users/TiMaKoV/Desktop/ckb-next-testing/quickinstall`

This? `8 lrwxr-xr-x@ 1 root  wheel  11 Jan 27 19:01 /tmp -> private/tmp`

@frickler24

frickler24, 2017-05-08:
OK, then go in this directory and let's have a look what is in there:
`cd /Users/TiMaKoV/Desktop/ckb-next-testing ; ls -l ; cat VERSION`
Then please compile everything:
`yes | quickinstall`
This will ask you for your password (it contains a sudo-command); please answer it
Then paste all what happened.
What is your ckb client doing now?

frickler24, 2017-05-08:
@fleischie or @tatokis A question to our mac users: How should he install the 0.2.8.pkg file? Is it like a .deb-file starting automatically after download? Put it into /tmp and open it with some tool? Open the link with a tool?

tatokis, 2017-05-08:
I don't use Mac. Only thing I know is that you just double click it and it gets installed.

light2yellow, 2017-05-08:
@frickler24 
macOS package is like an NSIS installer.

frickler24, 2017-05-08:
OK, thanks, We will see what @DesperateKidNeedsHelp brings out now

DesperateKidNeedsHelp, 2017-05-08:
@frickler24 Sorry, how do you compile `cd /Users/TiMaKoV/Desktop/ckb-next-testing ; ls -l ; cat VERSION`? Do you use `yes | quickinstall`?

frickler24, 2017-05-08:
Yes, complete compilation and installation is done with 
`cd /Users/TiMaKoV/Desktop/ckb-next-testing ; yes | quickinstall`.

After compilation and installation, please post the output from quickinstall.

Currently you have Version v0.2.7, so here we do not see the content of the link, tatokis sent you.
Because 28th of April is a bit too old, you really should download the [package](https://github.com/mattanger/ckb-next/releases/download/v0.2.8/ckb-v0.2.8.pkg). If it asks you for a place to save, say `/Users/TiMaKoV/Desktop` or short "~".
Then double click on the package at your desktop.
What's going on? Different behavior of your ckb?

DesperateKidNeedsHelp, 2017-05-09:
@frickler24 Wow, what bad timing! I can't continue trying to fix ckb-next anymore. My mac started getting incredibly slow and laggy these past few days. Now, it doesn't even start. Sorry, I'm going to have to resume this later until my mac is fixed or snaps out of it. (This is posted on a other device.. Just so you know.)      

Sorry :(

DesperateKidNeedsHelp, 2017-05-13:
@frickler24 @tatokis @light2yellow @fleischie Ey, I fixed my computer, but.... I will no longer try to get ckb on my Mac. When I finally booted the computer back up I instantly went to check viruses or threats. I found ckb stuff. Sorry I don't want to take another risk. I'm so so so so sorry for wasting your time. After all it is just a keyboard, lighting doesn't make it better.

closed 2017-05-13 by DesperateKidNeedsHelp