mattanger/ckb-next #180: OS X K95 RGB Platinum

norganna, 2017-05-07:
I have implemented the changes as per #53 under the OS X version on ckb-daemon, and it appears to work, however the connection process seems to error out several times with the message:

```
iterate_devices_hid (usb_mac.c:829): Failed to create device plugin: e00002be
```

and then later fails with:

```
[E] os_usbsend (via firmware.c:15): Got return value 0xe000404f
[E] _usbrecv (via firmware.c:15): Too many send/recv failures. Dropping.
[W] _start_dev (device.c:25): Unable to load firmware version/poll rate
[I] Setup finished for /var/run/ckb1
[I] Disconnecting /var/run/ckb1
```

The same changes from #53 work fine with this keyboard under linux, however this is for my wife's Mac, so I need to get it working for her.

Any suggestions?

tatokis, 2017-05-15:
Unfortunately I can not help much with macOS, however it is my understanding that `Failed to create device plugin: e00002be` is the equivalent of broken pipe on Linux.

The real issue is that the daemon can not talk to the keyboard over its control endpoint, so when it queries it for things like firmware version, it receives no reply.
After that, the keyboard resets and the daemon fails with broken pipe on Linux (or e00002be in this case).

#156 works around this, by faking the initialisation response packets, so that the daemon starts sending the RGB light data to the keyboard without actually waiting for a reply from it (or receiving an invalid/null one, as it happens sometimes).

Maybe @fleischie can help more.

tatokis, 2017-06-07:
A long shot, but try this https://github.com/mattanger/ckb-next/tree/k95p

light2yellow, 2017-06-28:
The branch was merged into `newdev`, use it instead: https://github.com/mattanger/ckb-next/tree/newdev

closed 2018-01-21 by tatokis