mattanger/ckb-next #181: [Regression] qInfo not defined when Qt < 5.5

tatokis, 2017-05-07:
From the IRC room.

With the introduction of https://github.com/mattanger/ckb-next/commit/3f02027d8e9d74834160db0ad457e7437bf96d5c the application fails to compile with Qt < 5.5.

Relevant messages
```
<Boomer>	ckbsettings.cpp: In function ‘QSettings* globalSettings()’:
<Boomer>	ckbsettings.cpp:28:19: error: ‘qInfo’ was not declared in this scope
<Boomer>	qInfo() << "Path to settings is" << _globalSettings->fileName();
<Boomer>	QMake version 3.0 Using Qt version 5.3.2 in /usr/lib/x86_64-linux-gnu
```
on Debian using master.

I propose we either wrap qInfo (and other similar debug functions) around a preprocessor if for Qt ≥ 5.5, or do a simple function implementation in our code that just prints the message to stderr for when these functions are not defined.

Suggestions?

frickler24, 2017-05-07:
What about this, unless we have a better logging structure (yes it is a hack):
```
#if QT_VERSION < QT_VERSION_CHECK(5, 5, 0)
#define qInfo       qDebug
#define qWarning    qDebug
#define qFatal      qDebug
#define qCritical   qDebug
#endif
```
Currently, I run it through travis with Qt4.8 .. 5.7 to check if it gets problems.
If it runs, we should agree where to put this (new header file "qDebug" which includes <QDebug>?).

First results: 
Qt4.8: compiles w/o errror
Qt5.1: `main.cpp:5:30: fatal error: QCommandLineParser: No such file or directory`
Qt5.2 - 5.5: compiles w/o errror
rest is just working.

light2yellow, 2017-05-07:
@frickler24 
The docs say parser was introduced in 5.2, we should bump to 5.2 here: https://github.com/mattanger/ckb-next/blob/master/ckb.pro#L1

Actually, using CI for such things is a very good idea!

frickler24, 2017-05-07:
@light2yellow I agree with 5.2 as minimum.
Here is the screen shot with the results:
[image]
The first red line was 5.1 as desribed above, the other one is 5.5 with clang. The reason for this is the packet of Qt5.5 to install. I do not know why we have a special treatment here, but for short I have disabled this config.
The error message was as follows:
```
 if [ "$QT_BASE" = "55" ]; then if [ "$TRAVIS_OS_NAME" = "linux" ]; then sudo apt-get install -qq qt55base; source /opt/qt55/bin/qt55-env.sh; else brew install qt55; brew link --force qt55; fi fi
Updating Homebrew...
==> Downloading https://homebrew.bintray.com/bottles/qt@5.5-5.5.1_1.el_capitan.b
==> Pouring qt@5.5-5.5.1_1.el_capitan.bottle.tar.gz
==> Caveats
We agreed to the Qt opensource license for you.
If this is unacceptable you should uninstall.
This formula is keg-only, which means it was not symlinked into /usr/local.
This is an alternate version of another formula.
If you need to have this software first in your PATH run:
  echo 'export PATH="/usr/local/opt/qt@5.5/bin:$PATH"' >> ~/.bash_profile
For compilers to find this software you may need to set:
    LDFLAGS:  -L/usr/local/opt/qt@5.5/lib
    CPPFLAGS: -I/usr/local/opt/qt@5.5/include
For pkg-config to find this software you may need to set:
    PKG_CONFIG_PATH: /usr/local/opt/qt@5.5/lib/pkgconfig
==> Summary
🍺  /usr/local/Cellar/qt@5.5/5.5.1_1: 7,330 files, 330.5MB
Linking /usr/local/Cellar/qt@5.5/5.5.1_1... 305 symlinks created
If you need to have this software first in your PATH instead consider running:
  echo 'export PATH="/usr/local/opt/qt@5.5/bin:$PATH"' >> ~/.bash_profile
...
```
BTW: The very long 11 minutes are for compiling + doxygen generation

frickler24, 2017-05-07:
I've opened #182 If we want to handle it this way and in the first step for this file only.

light2yellow, 2017-05-07:
> The error message was as follows:

I don't see any error here.

frickler24, 2017-05-07:
You're right, my fault, I should have shown the next lines also:
```
The command "cd $TRAVIS_BUILD_DIR" exited with 0.
1.84s$ echo "n" | ./quickinstall
Command line tools are already installed.
Preparing build files...
No suitable qmake found.
Please make sure you have Qt5 installed.
Re-run this program with QMAKE=/path/to/qmake to select qmake manually.
Press enter to exit.
The command "echo "n" | ./quickinstall" exited with 1.
Done. Your build exited with 1.
```
It seems that the incomplete installation leads to the missing qmake.

light2yellow, 2017-05-07:
As `brew` suggests, you need to execute this line on macOS too if force-linking:
```
echo 'export PATH="/usr/local/opt/qt@5.5/bin:$PATH"' >> ~/.bash_profile
```

I guess it should fix this particular problem.

Also, not really regarding this issue, but when you'll be ready to automate the runners to rebuild master on every push for example, let us now somewhere so that we add a cute badge to the README :D

light2yellow, 2017-05-08:
Well, actually you would need to log out before running anything if you do the above which inappropriate for CI. So instead of appending to `bash_profile` I suggest you launch `quickinstall` on macOS like this:
```sh
PATH="/usr/local/opt/qt@5.5/bin:$PATH" ./quickinstall
```

light2yellow, 2017-05-08:
Also, closing this as a the problem in the topic is solved.

closed 2017-05-08 by light2yellow