mattanger/ckb-next #184: Caps-Lock Indicator does not remember settings

alex-hempel, 2017-05-09:
ckb-next version 0.2.8
Corsair M65 Pro RGB
Mac OS Sierra 10.12.4

In my profile, I have set up a Caps-Lock indicator under Performance - Lock Lights. The setting is 'RGB Indicator' with colours red for On and green for Off.

This setting never works after a restart of ckb-next, even though it is still set. The Caps-Lock key will remain green, even when Caps Lock is active. I can change it to something else and then back to make it work, but after the next restart it's defunct again.

frickler24, 2017-05-09:
@alex-hempel I've done the following test:
1. copy ckb.conf to ckb.conf.save
2. Set caps lock indicater as you want
3. diff ckb.conf ckb.conf.save

If you do this test: Are there lines like the following (I changed the colors for this test as you mentioned to green/red)
```
< 0F022014AE3B9406537275B6F5001945\%7B8850F984-8E7D-43B4-869C-1D97F605DFEA%7D\0\Performance\Indicators\1\RGB0=#ff00ff00
< 0F022014AE3B9406537275B6F5001945\%7B8850F984-8E7D-43B4-869C-1D97F605DFEA%7D\0\Performance\Indicators\1\RGB1=#ffff0000
---
> 0F022014AE3B9406537275B6F5001945\%7B8850F984-8E7D-43B4-869C-1D97F605DFEA%7D\0\Performance\Indicators\1\RGB0=#ff0000ff
> 0F022014AE3B9406537275B6F5001945\%7B8850F984-8E7D-43B4-869C-1D97F605DFEA%7D\0\Performance\Indicators\1\RGB1=#ffffff00
```

alex-hempel, 2017-05-09:
Hi @frickler24 

on MacOS, I don't seem to have a ckb.conf.

According to https://github.com/ccMSC/ckb/issues/227, the preferences are stored in ~/Library/Preferences/com.ckb.ckb.plist, however that's a binary file.

frickler24, 2017-05-10:
OK, is one of [these hints](https://ilostmynotes.blogspot.de/2011/09/reading-and-modifying-os-x-plist-files.html) helping you reading and comparing the content?
Please be careful, because I don't have a mac and can't test it.

alex-hempel, 2017-05-10:
Okay, that helps.

I have created a copy of the plist file, then changed the settings.

This is what I see in the old file:

    "Devices.14035039AEBA042655DC4FA4F5001944.{BA7FC152-2D51-4C26-A7A6-A036CC93D924}.0.Performance.Indicators.1.Enable" = 1;
    "Devices.14035039AEBA042655DC4FA4F5001944.{BA7FC152-2D51-4C26-A7A6-A036CC93D924}.0.Performance.Indicators.1.Hardware" = 0;
    "Devices.14035039AEBA042655DC4FA4F5001944.{BA7FC152-2D51-4C26-A7A6-A036CC93D924}.0.Performance.Indicators.1.RGB0" = "#ffff0000";
    "Devices.14035039AEBA042655DC4FA4F5001944.{BA7FC152-2D51-4C26-A7A6-A036CC93D924}.0.Performance.Indicators.1.RGB1" = "#ff000000";

After changing the settings, I can see that the file has been changed, but the above settings look exactly the same. I switched it on and off, changed the colours around, but nothing changes for those lines.

frickler24, 2017-05-15:
On my K95RGB I have a different behior:
The Caps lock key is managed as a normal key for the colors. 
So if I set it to some color - say white - and set a color change when hit, the color changes with each keystroke, but comes back after the time programmed.
Additionally I have a caps lock indicator, but this is a white micro-LED on top of the keyboard. If I program a color for caps-lock in the performance-tab, the LED keeps off.
If I set it to standard, the LED turn on/off on each status change. But it remains white (I assume, this is a while LED).
So my question: Do you have a separate caps lock indicator, or do you want to set the color of the caps lock key itself?

alex-hempel, 2017-05-16:
@frickler24 

I am wanting to set the colour of the caps lock key. CKB has the Performance - Lock Lights settings for each keyboard, where lock lights can be set for Caps Lock, Num Lock and Scroll Lock. This option worked fine in older versions IIRC, but hasn't been working since 0.2.7.

See screenshot:

[image]

With this setting, the Caps Lock key should be red when active, and green when not active. This works immediately after I change the setting, but it doesn't work after a few hours or a CKB restart.

frickler24, 2017-05-17:
OK, @alex-hempel, I have at least the same effect on linux mint 18.1 when restarting the machine. I never have used this before, so I son't know if the effect after some hours usage occurs also as you describe.
At weekend I'll try to debug the reason for it.

frickler24, 2017-05-21:
Hi @alex-hempel, please try the PR #200 and give some feedback.
Others may try it also, the behavior is independent from the OS 
but may be dependent on keyboard-hardware / firmware.
I've tested with K95RGB.

alex-hempel, 2017-05-21:
Hi @frickler, that looks good so far. The setting survives an application restart. Will see if it keeps working throughout the day.

Thanks for adding this fix, much appreciated!

alex-hempel, 2017-05-22:
24 hours later the setting is still working fine.

I'd say the fix is solid.

frickler24, 2017-05-24:
Merged into testing and master.
@alex-hempel thank you for testing and feedback!

closed 2017-05-24 by frickler24